micro-ROS/micro_ros_setup #25: Compilation errors in Dashing version

KevinHwangAi, 2019-10-10:
I have installed the dashing version of micro-ros-build.
After configuring the MCU (tools/configure.sh configs/olimex-stm32-e407/drive_base) and enter the build_firmware command (ros2 run micro_ros_setup build_firmware.sh) ,
I got the following  error :

----------------------------------------------------------------------------------------------------------------
kevin@kevin:~/uros_ws$ ros2 run micro_ros_setup build_firmware.sh
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> ament_cppcheck
Starting >>> gtest_vendor
Starting >>> ament_pep8
Finished <<< ament_package [1.13s]                           
Finished <<< ament_lint [1.13s]
Starting >>> ament_cmake_core                                 
Starting >>> ament_flake8
Finished <<< ament_cppcheck [1.45s]
Finished <<< ament_pep8 [1.80s]                                   
Starting >>> ament_pyflakes
Finished <<< ament_flake8 [0.80s]                                   
Finished <<< gtest_vendor [2.24s]
Starting >>> ament_pep257
Starting >>> gmock_vendor
Finished <<< ament_pyflakes [0.77s]                                     
Finished <<< ament_cmake_core [1.18s]
Starting >>> ament_cmake_test
Starting >>> ament_cmake_libraries
Starting >>> ament_cmake_export_libraries
Starting >>> ament_cmake_include_directories
Starting >>> ament_cmake_export_definitions
Starting >>> ament_cmake_export_include_directories
Starting >>> ament_cmake_export_link_flags
Starting >>> ament_cmake_python
Finished <<< gmock_vendor [0.64s]                            
Finished <<< ament_cmake_test [0.39s]                       
Starting >>> ament_cmake_gtest
Starting >>> ament_lint_auto
Starting >>> ament_cmake_nose
Starting >>> ament_cmake_pytest                               
Finished <<< ament_cmake_libraries [0.55s]
Finished <<< ament_cmake_export_definitions [0.56s]           
Finished <<< ament_cmake_include_directories [0.57s]
Finished <<< ament_cmake_export_libraries [0.58s]
Finished <<< ament_cmake_export_include_directories [0.56s]
Starting >>> ament_cmake_export_dependencies
Starting >>> ament_cmake_export_interfaces
Starting >>> ament_cmake_target_dependencies
Finished <<< ament_cmake_export_link_flags [0.56s]
Finished <<< ament_cmake_python [0.55s]
Finished <<< ament_pep257 [0.98s]
Starting >>> ament_copyright
Starting >>> ament_index_python
Finished <<< ament_copyright [0.96s]                                      
Starting >>> ament_lint_cmake
Starting >>> ament_cpplint
Starting >>> ament_xmllint                                                 
Starting >>> ament_clang_format
Starting >>> ament_clang_tidy                                              
Starting >>> ament_pclint
Finished <<< ament_cmake_gtest [1.32s]
Finished <<< ament_lint_auto [1.32s]
Finished <<< ament_cmake_nose [1.24s]
Finished <<< ament_cmake_pytest [1.22s]
Finished <<< ament_cmake_export_dependencies [1.11s]
Finished <<< ament_cmake_target_dependencies [1.11s]
Finished <<< ament_cmake_export_interfaces [1.11s]
Starting >>> ament_cmake
Starting >>> ament_cmake_gmock
Finished <<< ament_index_python [3.28s]                            
Finished <<< ament_lint_cmake [2.32s]
Finished <<< ament_xmllint [2.28s]
[... 137 lines cut ...]
  ADD_LIBRARY called with SHARED option but the target platform does not
  support dynamic linking.  Building a STATIC library instead.  This may lead
  to problems.
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Aborted  <<< libyaml_vendor

Summary: 6 packages finished [12.3s]
  1 package failed: rcutils
  3 packages aborted: libyaml_vendor microxrcedds_client rosidl_cmake
  2 packages had stderr output: libyaml_vendor rcutils
  37 packages not processed
Makefile:31: recipe for target 'colcon_compile' failed
make[3]: *** [colcon_compile] Error 2
make[3]: Leaving directory '/home/kevin/uros_ws/firmware/apps/uros'
Makefile:106: recipe for target 'uros_context' failed
make[2]: *** [uros_context] Error 2
make[2]: Leaving directory '/home/kevin/uros_ws/firmware/apps'
Makefile:179: recipe for target 'context_serialize' failed
make[1]: *** [context_serialize] Error 2
make[1]: Leaving directory '/home/kevin/uros_ws/firmware/apps'
tools/Makefile.unix:345: recipe for target 'context' failed
make: *** [context] Error 2
----------------------------------------------------------------------------------------------------------------------------

BorjaOuterelo, 2019-10-10:
Hi, @KevinHwangAi I will have a look at it and come back to you.
Thanks for reporting.

KevinHwangAi, 2019-10-10:
I found that in this file: micro-ros-build/micro_ros_setup/config/uros_packages.repos , 
the version  of  apps is still crystal :
----------------------------------------------------------------------------------------------------------------------------
repositories:
  NuttX:
    type: git
    url: https://github.com/micro-ROS/NuttX.git
    version: drive_base
  apps:
    type: git
    url: https://github.com/micro-ROS/apps.git
    version: crystal
  uclibc:
    type: git
    url:  https://github.com/micro-ROS/uclibc.git
    version: master
--------------------------------------------------------------------------------------------------------------------------------

Does this cause compilation errors?

BorjaOuterelo, 2019-10-10:
Those need to be updated to support Dashing...

KevinHwangAi, 2019-10-10:
@BorjaOuterelo Is that mean it still need to wait for the apps to be updated to Dashing ?

iluetkeb, 2019-10-10:
We haven't released the dashing version, yet, but we're working on it. That's why there are "dashing" branches in the repos.

See https://micro-ros.github.io//docs/tutorials/basic/getting_started/ for the current status. Dashing is still "TODO"...

KevinHwangAi, 2019-10-10:
@iluetkeb Thank you for your efforts. Because of the Dashing version of Ros2 used in our robotics project, we very much hope to use Mirco_Ros in MCU communications.

Could you tell me when the Dashing version will be updated?

iluetkeb, 2019-10-10:
I can't really say, there are several people, and I don't know their time-frame. Right now, we are also a bit busy with ROSCon preparations.

That said, you can compile the firmware using ROS 2 Crystal, and it can communicate with an agent running on ROS 2 Dashing. I think the agent is a bit further ahead than the rest, so this might be a faster solution.

KevinHwangAi, 2019-10-10:
@iluetkeb 
That sounds like a very good proposal !!!

KevinHwangAi, 2019-10-10:
Hi @iluetkeb ,I try run the agent in the ROS2 Dashing and compile the firmware with Crystal as what you said.
And now, In the NSH input 'publisher', I got this error:

publisher
modified publisher
Serial mode => dev: /dev/ttyS0

>>> [rcutils|error_handling.c:106] rcutils_set_error_state()
This error state is being overwritten:

  'failed to create node session on Micro ROS Agent., at /home/gemini/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_nod'

with this new error message:

  'rcl node's rmw handle is invalid, at /home/gemini/uros_ws/firmware/mcu_ws/ros2/rcl/rcl/src/rcl/node.c:535'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [rcl]: Failed to fini publisher for node: 1
[rclc_create_node] error in rcl_node_init: rcl node's rmw handle is invalid, at /home/gemini/uros_ws/firmware/mcu_ws/ros2/rcl/rcl/src/rcl/node.c:55
Issues creating node

KevinHwangAi, 2019-10-11:
Hi@iluetkeb @BorjaOuterelo,
I thing I have solved the problem.  
The MCU(Crtstal) can run the publisher now, and the agent(Dashing) got this :

kevin@kevin:~$ ~/uros_ws/install/micro_ros_agent/lib/micro_ros_agent/micro_ros_agent serial --dev /dev/ttyUSB0 
Enter 'q' for exit
[1570771546.421357] info     | SerialServerLinux.cpp | init                     | running...             | fd: 3
[1570771560.434907] info     | Root.cpp           | create_client            | create                 | client_key: 0x75E1799B, session_id: 0x81
[1570771560.434983] info     | SerialServerBase.cpp | on_create_client         | session established    | client_key: 0x75E1799B, address: 1

iluetkeb, 2019-10-11:
@KevinHwangAi what did you do to fix the issue? I have not encountered the "invalid cookie" problem before -- it would be good to know what to do to fix it.

KevinHwangAi, 2019-10-11:
@iluetkeb The problem seems to arise from the different definitions of "xrce_cookie" in xrce_protocol_internal.h between Crystal and Dashing.

Crystal:
typedef struct XrceCookie
{
    uint8_t data[4];

} XrceCookie;
#define XRCE_COOKIE (XrceCookie){{0x58, 0x52, 0x43, 0x45}}

typedef struct XrceVersion
{
    uint8_t data[2];

} XrceVersion;
#define XRCE_VERSION_MAJOR 0x01
#define XRCE_VERSION_MINOR 0x00
#define XRCE_VERSION (XrceVersion){{XRCE_VERSION_MAJOR, XRCE_VERSION_MINOR}}

Dashing:
typedef struct XrceCookie
{
    uint8_t data[4];

} XrceCookie;
#define XRCE_COOKIE COMPOUND_LITERAL(XrceCookie){{0x58, 0x52, 0x43, 0x45}}

typedef struct XrceVersion
{
    uint8_t data[2];

} XrceVersion;
#define XRCE_VERSION_MAJOR 0x01
#define XRCE_VERSION_MINOR 0x00
#define XRCE_VERSION COMPOUND_LITERAL(XrceVersion){{XRCE_VERSION_MAJOR, XRCE_VERSION_MINOR}}

iluetkeb, 2019-10-26:
@KevinHwangAi If this issue has been resolved successfully for you, please close it.

closed 2019-10-28 by KevinHwangAi