micro-ROS/micro_ros_setup #26: Try installing Mirco_Ros to another MCU(stm32f4dis)

KevinHwangAi, 2019-10-11:
@Viplav04 I want to try another MCU(stm32f4dis) to install the Mirco ros.
But it failed at configuration and compilation and  still trying to figure out ...

_Originally posted by @KevinHwangAi in https://github.com/micro-ROS/micro-ros-build/issues/18#issuecomment-541038688_

iluetkeb, 2019-10-11:
Which discovery board are you using and which configuration did you try?

KevinHwangAi, 2019-10-11:
@iluetkeb Please tell me what information I need to provide for the board(stm32f4discovery）

iluetkeb, 2019-10-11:
The exact name of the board, including the MCU, e.g. "STM32F746-Discovery" or the like.

You can look in NuttX/configs/ for the names of the boards NuttX has support for.

KevinHwangAi, 2019-10-11:
@iluetkeb Can you tell me what should I do now？

iluetkeb, 2019-10-11:
A little patience ;-) I'm currently looking at the setup.

KevinHwangAi, 2019-10-11:
@iluetkeb That's fine. Thank you!

iluetkeb, 2019-10-11:
If you go your firmware/NuttX directory and do 
```
git fetch -a
git checkout -b feature/stm32f4discovery
make distclean
tools/configure.sh configs/stm32f4discovery/uros
```
and then compile as usual (i.e., with build_firmware.sh), it should work for you.

Please note that the "uros" configuration provided is not optimal. There are many optimization options, e.g., using the FPU, set for the Olimex board which I did not enable for the STM32F4Discovery, because I do not have such a board and cannot test it.

If you test those things, I would appreciate a PR to improve the default configuration.

KevinHwangAi, 2019-10-11:
@iluetkeb Thanks a lot! I will test the configure and feedback relevant information.

KevinHwangAi, 2019-10-11:
@iluetkeb ，After input commend：ros2 run micro_ros_setup build_firmware.sh

I got this error：

Starting >>> rmw_implementation
--- stderr: rmw_implementation                             
CMake Warning at CMakeLists.txt:19 (find_package):
  By not providing "Findpoco_vendor.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "poco_vendor", but CMake did not find one.

  Could not find a package configuration file provided by "poco_vendor" with
  any of the following names:

    poco_vendorConfig.cmake
    poco_vendor-config.cmake

  Add the installation prefix of "poco_vendor" to CMAKE_PREFIX_PATH or set
  "poco_vendor_DIR" to a directory containing one of the above files.  If
  "poco_vendor" provides a separate development package or SDK, be sure it
  has been installed.

CMake Warning at CMakeLists.txt:20 (find_package):
  Could not find a configuration file for package "Poco" that is compatible
  with requested version "".

  The following configuration files were considered but not accepted:

    /usr/lib/cmake/Poco/PocoConfig.cmake, version: 1.8.0 (64bit)

CMake Warning at /home/gemini/uros_ws/firmware/mcu_ws/install/rmw_microxrcedds/share/rmw_microxrcedds/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rmw_microxrcedds' exports library 'pthread' which couldn't be
  found
Call Stack (most recent call first):
  /home/gemini/uros_ws/firmware/mcu_ws/install/rmw_microxrcedds/share/rmw_microxrcedds/cmake/rmw_microxrceddsConfig.cmake:38 (include)
  /home/gemini/uros_ws/firmware/mcu_ws/install/rmw_implementation_cmake/share/rmw_implementation_cmake/cmake/get_default_rmw_implementation.cmake:62 (find_package)
  CMakeLists.txt:28 (get_default_rmw_implementation)

---
Finished <<< rmw_implementation [0.84s]
Starting >>> rosidl_typesupport_c
--- stderr: rosidl_typesupport_c                               
In file included from /usr/include/newlib/c++/6.3.1/bits/char_traits.h:420:0,
                 from /usr/include/newlib/c++/6.3.1/string:40,
                 from /home/gemini/uros_ws/firmware/mcu_ws/ros2/rosidl_typesupport/rosidl_typesupport_c/src/type_support_dispatch.hpp:23,
                 from /home/gemini/uros_ws/firmware/mcu_ws/ros2/rosidl_typesupport/rosidl_typesupport_c/src/service_type_support_dispatch.cpp:17:
/home/gemini/uros_ws/firmware/NuttX/include/cxx/cstdint:75:11: error: 'uint_least32_t' is already declared in this scope
   using ::uint_least32_t;
           ^~~~~~~~~~~~~~
In file included from /usr/include/newlib/c++/6.3.1/bits/char_traits.h:420:0,
                 from /usr/include/newlib/c++/6.3.1/string:40,
                 from /home/gemini/uros_ws/firmware/mcu_ws/ros2/rosidl_typesupport/rosidl_typesupport_c/src/type_support_dispatch.hpp:23,
                 from /home/gemini/uros_ws/firmware/mcu_ws/ros2/rosidl_typesupport/rosidl_typesupport_c/src/message_type_support_dispatch.cpp:17:
/home/gemini/uros_ws/firmware/NuttX/include/cxx/cstdint:75:11: error: 'uint_least32_t' is already declared in this scope
   using ::uint_least32_t;
           ^~~~~~~~~~~~~~
In file included from /usr/include/newlib/c++/6.3.1/bits/char_traits.h:420:0,
                 from /usr/include/newlib/c++/6.3.1/string:40,
                 from /home/gemini/uros_ws/firmware/mcu_ws/ros2/rosidl_typesupport/rosidl_typesupport_c/src/type_support_dispatch.hpp:23,
                 from /home/gemini/uros_ws/firmware/mcu_ws/ros2/rosidl_typesupport/rosidl_typesupport_c/src/type_support_dispatch.cpp:15:
/home/gemini/uros_ws/firmware/NuttX/include/cxx/cstdint:75:11: error: 'uint_least32_t' is already declared in this scope
   using ::uint_least32_t;
           ^~~~~~~~~~~~~~
In file included from /usr/include/newlib/c++/6.3.1/string:52:0,
                 from /home/gemini/uros_ws/firmware/mcu_ws/ros2/rosidl_typesupport/rosidl_typesupport_c/src/type_support_dispatch.hpp:23,
                 from /home/gemini/uros_ws/firmware/mcu_ws/ros2/rosidl_typesupport/rosidl_typesupport_c/src/service_type_support_dispatch.cpp:17:
/usr/include/newlib/c++/6.3.1/bits/basic_string.h: In function 'int std::__cxx11::stoi(const wstring&, std::size_t*, int)':
/usr/include/newlib/c++/6.3.1/bits/basic_string.h:5534:42: error: 'wcstol' is not a member of 'std'
   { return __gnu_cxx::__stoa<long, int>(&std::wcstol, "stoi", __str.c_str(),
                                          ^~~
[... 932 lines cut ...]
                                     ^~
/usr/include/newlib/c++/6.3.1/arm-none-eabi/thumb/v7e-m/bits/ctype_base.h:54:42: error: '_N' was not declared in this scope
     static const mask alnum  = _U | _L | _N;
                                          ^~
make[6]: *** [CMakeFiles/rosidl_typesupport_c.dir/src/type_support_dispatch.cpp.obj] Error 1
make[5]: *** [CMakeFiles/rosidl_typesupport_c.dir/all] Error 2
make[4]: *** [all] Error 2
---
Failed   <<< rosidl_typesupport_c	[ Exited with code 2 ]

Summary: 20 packages finished [5.38s]
  1 package failed: rosidl_typesupport_c
  2 packages had stderr output: rmw_implementation rosidl_typesupport_c
  27 packages not processed
Makefile:31: recipe for target 'colcon_compile' failed
make[3]: *** [colcon_compile] Error 2
make[3]: Leaving directory '/home/gemini/uros_ws/firmware/apps/uros'
Makefile:106: recipe for target 'uros_context' failed
make[2]: *** [uros_context] Error 2
make[2]: Leaving directory '/home/gemini/uros_ws/firmware/apps'
Makefile:179: recipe for target 'context_serialize' failed
make[1]: *** [context_serialize] Error 2
make[1]: Leaving directory '/home/gemini/uros_ws/firmware/apps'
tools/Makefile.unix:345: recipe for target 'context' failed
make: *** [context] Error 2

iluetkeb, 2019-10-12:
Hmm. This looks like the include paths are not correctly propagated.

I would suggest cleaning your workspace, it might be that the oldconfiguration is still present.

 1. Remove `firmware/mcu_ws/build` and `firmware/mcu_ws/install`
 2. call `make clean` and then `make distclean` in your NuttX directory
 3. configure again and then run `build_firmware.sh` again

KevinHwangAi, 2019-10-14:
> Hmm. This looks like the include paths are not correctly propagated.
>
> I would suggest cleaning your workspace, it might be that the oldconfiguration is still present.
>
> 1. Remove [...]

I tried to follow the steps here, but got the same error as above.
I think the main reason is that "poco_vendor" has not been found?

Starting >>> rosidl_typesupport_c
stderr: rosidl_typesupport_c                                
CMake Warning at CMakeLists.txt:20 (find_package):
  By not providing "Findpoco_vendor.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "poco_vendor", but CMake did not find one.

iluetkeb, 2019-10-14:
The poco_vendor is not important, it's just for plugin-loading, and we don't need that in the firmware (however, entirely removing the package caused dependency issues, so now it's there but doesn't do anything).

What's happening here is that the build is using all sorts of includes from /usr/newlib/c++/, and you're getting errors in those files.

This should not happen -- NuttX comes with its c++ library (uClibc++), and the modifications I made to the build should make it use that one, instead of newlib.

Please have a look at the `Make.defs` in the NuttX directory. It should have a block saying
```
# Linux/Cygwin-native toolchain
  MKDEP = $(TOPDIR)/tools/mkdeps$(HOSTEXEEXT)
  ARCHINCLUDES = -I. -isystem $(TOPDIR)/include
  ARCHXXINCLUDES = -I. -isystem $(TOPDIR)/include -isystem $(TOPDIR)/include/cxx -isystem $(TOPDIR)/include/uClibc++
  ARCHSCRIPT = -T$(TOPDIR)/configs/$(CONFIG_ARCH_BOARD)/scripts/$(LDSCRIPT)
```

And the important part in that is the `$(TOPDIR)/include/uClibc++`.

Viplav04, 2019-10-14:
@iluetkeb  I had the same issue when installing the crystal version and you created a new branch to modify the build_firmware.sh file.
[build_firmware.txt](https://github.com/micro-ROS/micro-ros-build/files/3723893/build_firmware.txt)

KevinHwangAi, 2019-10-14:
@iluetkeb 
I replaced the contents of the NuttX (feature/stm32f4discovery)/configs/stm32f4discovery folder with  the NuttX (drive_base)/configs/stm32f4discovery folder.
Now it can skip the above error,but got this new error:

AR:   stm32_boot.o stm32_bringup.o stm32_spi.o libatomic.o stm32_autoleds.o stm32_buttons.o stm32_usb.o stm32_appinit.o 
make[2]: Leaving directory '/home/gemini/uros_ws/firmware/NuttX/configs/stm32f4discovery/src'
LD: nuttx
arm-none-eabi-ld --entry=__start -nostartfiles -nodefaultlibs -T/home/gemini/uros_ws/firmware/NuttX/configs/stm32f4discovery/scripts/ld.script -L"/home/gemini/uros_ws/firmware/NuttX/staging" -L"/home/gemini/uros_ws/firmware/NuttX/arch/arm/src/board" -L "/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7e-m" \
	-o "/home/gemini/uros_ws/firmware/NuttX/nuttx"   \
	--start-group -lsched -ldrivers -lconfigs -lc -lmm -larch -lxx -lapps -lnet -lfs -lbinfmt -lxx -lboard -lsupc++ "/usr/lib/gcc/arm-none-eabi/6.3.1/thumb/v7e-m/libgcc.a" --end-group
/home/gemini/uros_ws/firmware/NuttX/staging/libarch.a(up_initialize.o): In function `up_initialize':
up_initialize.c:(.text+0x26): undefined reference to `up_netinitialize'
/home/gemini/uros_ws/firmware/NuttX/staging/libapps.a(librmw_microxrcedds_rmw_node.c.obj): In function `create_node':
/home/gemini/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_node.c:138: undefined reference to `tcgetattr'
/home/gemini/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_node.c:173: undefined reference to `cfsetspeed'
/home/gemini/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_node.c:174: undefined reference to `cfsetspeed'
/home/gemini/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_node.c:176: undefined reference to `tcsetattr'
Makefile:184: recipe for target 'nuttx' failed
make[1]: *** [nuttx] Error 1
make[1]: Leaving directory '/home/gemini/uros_ws/firmware/NuttX/arch/arm/src'
tools/Makefile.unix:422: recipe for target 'pass2' failed
make: *** [pass2] Error 2

Viplav04, 2019-10-14:
@KevinHwangAi it's a config error, for tcgetattr, maybe try some changes in the RTOS features, I did got this error a few times, but don't exactly remember why, i was just playing around with configs

> @iluetkeb
> I replaced the contents of the NuttX (feature/stm32f4discovery)/configs/stm32f4discovery folder with the NuttX (drive_base)/configs/stm32f4discovery folder.
> Now it can skip the above [...]

KevinHwangAi, 2019-10-14:
Thanks @Viplav04.
I'll try to change the configuration file.

Viplav04, 2019-10-14:
you can also check the config file I sent you in the other thread. @KevinHwangAi

KevinHwangAi, 2019-10-14:
The program was compiled successfully and burned into the motherboard.
There are new problems.
After connecting the interfaces of mirco USB and mini usb to Linux, Linux can only display the mirco USB interface (st-link) and identify / dev / ttyACM0.
Entry the minicom and press "enter" three times,but there is no response.

Viplav04, 2019-10-14:
Can you send a photo of the hardware and config file?
Also type dmesg in Linux after resetting the board to see which ACM is used

Kind regards
Viplav

iluetkeb, 2019-10-14:
This is getting a bit chaotic, guys ;-)

@Viplav04 The changes to build_firmware.sh have been incorporated into the main drive_base branch.

@KevinHwangAi If you got it to run after changing the folder, maybe I pushed to the wrong branch. Odd, though. Could you please verify the Make.defs file, as I requested earlier?

KevinHwangAi, 2019-10-15:
@iluetkeb The Make.defs file in Nuttx(feature/stm32f4discovery) is like this:

`ifeq ($(WINTOOL),y)
  # Windows-native toolchains
  DIRLINK = $(TOPDIR)/tools/copydir.sh
  DIRUNLINK = $(TOPDIR)/tools/unlink.sh
  MKDEP = $(TOPDIR)/tools/mkwindeps.sh
  ARCHINCLUDES = -I. -isystem "${shell cygpath -w $(TOPDIR)/include}"
  ARCHXXINCLUDES = -I. -isystem "${shell cygpath -w $(TOPDIR)/include}" -isystem "${shell cygpath -w $(TOPDIR)/include/cxx}"
  ARCHSCRIPT = -T "${shell cygpath -w $(TOPDIR)/configs/$(CONFIG_ARCH_BOARD)/scripts/$(LDSCRIPT)}"
else
  # Linux/Cygwin-native toolchain
  MKDEP = $(TOPDIR)/tools/mkdeps$(HOSTEXEEXT)
  ARCHINCLUDES = -I. -isystem $(TOPDIR)/include
  ARCHXXINCLUDES = -I. -isystem $(TOPDIR)/include -isystem $(TOPDIR)/include/cxx -isystem $(TOPDIR)/include/uClibc++
  ARCHSCRIPT = -T$(TOPDIR)/configs/$(CONFIG_ARCH_BOARD)/scripts/$(LDSCRIPT)
endif`

KevinHwangAi, 2019-10-15:
@iluetkeb 
dmesg | grep tty*  got this:

[    0.000000] console [tty0] enabled
[    0.000000] DMAR-IR: Use 'intremap=no_x2apic_optout' to override the BIOS setting.
[    0.081322] MDS CPU bug present and SMT on, data leak possible. See https://www.kernel.org/doc/html/latest/admin-guide/hw-vuln/mds.html for more details.
[    0.312021] pci 0000:03:00.0: vgaarb: setting as boot VGA device
[    0.312126] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.991777] input: Power Button as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input0
[    0.991784] ACPI: Power Button [PWRB]
[    0.991817] input: Power Button as /devices/LNXSYSTM:00/LNXPWRBN:00/input/input1
[    0.991829] ACPI: Power Button [PWRF]
[    1.016203] 00:03: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    1.036702] 00:04: ttyS1 at I/O 0x2f8 (irq = 3, base_baud = 115200) is a 16550A
[    1.166037] evm: Initialising EVM extended attributes:
[    1.166041] evm: HMAC attrs: 0x1
[    1.166679] rtc_cmos 00:00: setting system clock to 2019-10-15 06:00:43 UTC (1571119243)
[    1.996886] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    1.997621] sd 0:0:0:0: [sda] Attached SCSI disk
[    1.998958] sd 3:0:0:0: Attached scsi generic sg1 type 0
[    1.999837] sd 3:0:0:0: [sdb] Attached SCSI disk
[    3.615684] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  430.26  Tue Jun  4 17:45:09 CDT 2019
[ 4692.362354] cdc_acm 3-4:1.2: ttyACM0: USB ACM device
[ 4693.395000] sd 12:0:0:0: Attached scsi generic sg2 type 0
[ 4693.402688] sd 12:0:0:0: [sdc] Attached SCSI removable disk
[ 5281.095424] cdc_acm 3-4:1.2: ttyACM0: USB ACM device
[ 5282.126028] sd 12:0:0:0: Attached scsi generic sg2 type 0
[ 5282.133947] sd 12:0:0:0: [sdc] Attached SCSI removable disk
[ 5491.155855] cdc_acm 3-4:1.2: ttyACM0: USB ACM device
[ 5492.173598] sd 12:0:0:0: Attached scsi generic sg2 type 0
[ 5492.181338] sd 12:0:0:0: [sdc] Attached SCSI removable disk

the hardware photo：
[image]

the config flie of Nuttx/configs/stm32f4discovery : 
[defconfig.txt](https://github.com/micro-ROS/micro-ros-build/files/3727865/defconfig.txt)

Viplav04, 2019-10-15:
@KevinHwangAi  You will have to use GPIO pins I guess to connect to the agent like with the olimex board using UART : ttyS0

KevinHwangAi, 2019-10-15:
@Viplav04  I try that way ,but still not working.

iluetkeb, 2019-10-15:
Your configuration does not have NSH configured to use the USB device as a console.

Viplav04, 2019-10-15:
@KevinHwangAi  watch this https://www.youtube.com/watch?v=rgZJPZh_po8

iluetkeb, 2019-10-15:
My experience with the USB device missing has been that there is a console problem in the configuraiton.

In your configuration, you don't have CONFIG_NSH_CONSOLE set. It should be set to yes.

btw, you can also enable USB console for NSH, but since your USB configuration claims /dev/console, that should not be neccessary.

KevinHwangAi, 2019-10-15:
@iluetkeb @Viplav04  Thanks guys.I'll try.

KevinHwangAi, 2019-10-15:
The board's mirco usb port can recognized by linux now.
[ 1319.204722] usb 3-9.2: cp210x converter now attached to ttyUSB0
[ 1336.964639] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 1338.285895] usb 3-9.2: cp210x converter now attached to ttyUSB0
[ 1338.756632] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 1339.158353] usb 3-9.2: cp210x converter now attached to ttyUSB0
[ 1339.414330] cdc_acm 3-9.4:1.2: ttyACM0: USB ACM device
[ 1339.414626] cdc_acm 3-4:1.0: ttyACM1: USB ACM device
[ 1465.743071] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 1465.955405] cdc_acm 3-9.4:1.2: ttyACM0: USB ACM device
[ 1505.556388] usb 3-9.2: cp210x converter now attached to ttyUSB0
[ 1511.044851] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 1511.445354] usb 3-9.2: cp210x converter now attached to ttyUSB0
[ 2424.197780] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 2446.982990] usb 3-9.2: cp210x converter now attached to ttyUSB0
[ 2447.237742] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 2447.638659] usb 3-9.2: cp210x converter now attached to ttyUSB0
[ 3943.134828] cdc_acm 3-4:1.0: ttyACM1: USB ACM device
[ 3945.190790] cdc_acm 3-4:1.0: ttyACM1: USB ACM device
[ 3952.687866] cdc_acm 3-9.4:1.2: ttyACM0: USB ACM device
[ 3953.214794] cdc_acm 3-4:1.0: ttyACM1: USB ACM device

I entry the NutShell,and input "publisher",and got this:

nsh> publisher
modified publisher
Serial mode => dev: /dev/ttyS0
>>> [rcutils|error_handling.c:106] rcutils_set_error_state()
This error state is being overwritten:
  'failed to create node session on Micro ROS Agent., at /home/gemini/uros_ws/firmware/mcu_ws/uros/rmw_m'

with this new error message:

  'rcl node's rmw handle is invalid, at /home/gemini/uros_ws/firmware/mcu_ws/ros2/rcl/rcl/src/rcl/node.c'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [rcl]: Failed to fini publisher for node: 1
[rclc_create_node] error in rcl_node_init: rcl node's rmw handle is invalid, at /home/gemini/uros_ws/fir5
Issues creating node

The agent info: 
gemini@kevin:~$ ~/uros_ws/install/micro_ros_agent/lib/micro_ros_agent/micro_ros_agent serial /dev/ttyUSB0Serial agent initialization... OK
Enter 'q' for exit
deserialize eprosima::fastcdr::exception::NotEnoughMemoryException
Error processing CREATE_CLIENT submessage.

but the stm32f4dis is not out of memory:
nsh> free
             total       used       free    largest
Umem:       174576      12000     162576      65008

The config flie of Nuttx/configs/stm32f4discovery :+1: 
[defconfig.txt](https://github.com/micro-ROS/micro-ros-build/files/3729384/defconfig.txt)

Viplav04, 2019-10-15:
@KevinHwangAi 
How are you contacting the agent?
through USART - ttyS0?

KevinHwangAi, 2019-10-15:
@Viplav04,I used the mirco usb to connect the  NuttShell and the USART to communication,
but I don't know why the mirco usb prot is working now...
and it change to ttyACM1.

iluetkeb, 2019-10-15:
You cannot use NSH with the USB port in CDC/ACM mode and use Micro-ROS over it at the same time. So you *either* have to use an UART for Micro-ROS instead (looks as if you have that wired up), *or* you could configure the USB port to be a composite device that provides both CDC/ACM and Ethernet-over-USB at the same time, and then use IP communication for Micro-ROS.

Viplav04, 2019-10-15:
@KevinHwangAi Then it looks like maybe the UART connections or config is wrong. Now you should be able to access NSH via both ttyACM0 and 1.

iluetkeb, 2019-10-15:
btw, guys, for this kind of interactive troubleshooting, you might want to join our Slack channel

iluetkeb, 2019-10-15:
Get your invite here: https://micro-ros.github.io/

KevinHwangAi, 2019-10-17:
The  board is finally up and running, and communicate with the Ros2_Dashing agent. 
The problem is that it appears on the permissions of the usb port,
The usb port can be used by adding permissions using the "chmod" command .

Now I want to write publisher, subscriber, PID & PWM control, and other functions on this board at the same time, and take stm32f4discovery as the main controller of the robot.

Is this possible?

iluetkeb, 2019-10-17:
> The usb port can be used by adding permissions using the "chmod" command .

Do I understand this correctly, that the device was always present, but you couldn't access it?

On my system, `/dev/ttyACM0` is in the `dialout` group, so the permanent solution (better than chmod'ing every time) is to put your user into that group as well (than logout and login, as usual).

KevinHwangAi, 2019-10-17:
The previous problem is  in  /dev/ttyusb0,This interface can't be monitored by the agent. It needs to be given higher permissions before it can be monitored by the agent.

KevinHwangAi, 2019-10-17:
@Viplav04 OK! I will try,thanks.

iluetkeb, 2019-10-26:
@KevinHwangAi Is this issue solved? If yes, please close.

closed 2019-10-28 by KevinHwangAi