micro-ROS/micro_ros_setup #29: compile error

chinahuangyong, 2019-10-25:
chip/stm32_usbhost.c:80:52: error: 'TRACE1_NSTRINGS' undeclared here (not in a function)
 static const struct stm32_usbhost_trace_s g_trace1[TRACE1_NSTRINGS] =
                                                    ^~~~~~~~~~~~~~~
chip/stm32_usbhost.c:150:52: error: 'TRACE2_NSTRINGS' undeclared here (not in a function)
 static const struct stm32_usbhost_trace_s g_trace2[TRACE2_NSTRINGS] =
                                                    ^~~~~~~~~~~~~~~
chip/stm32_usbhost.c: In function 'usbhost_trformat1':
chip/stm32_usbhost.c:235:13: warning: implicit declaration of function 'TRACE1_INDEX' [-Wimplicit-function-declaration]
   int ndx = TRACE1_INDEX(id);
             ^~~~~~~~~~~~
chip/stm32_usbhost.c: In function 'usbhost_trformat2':
chip/stm32_usbhost.c:247:13: warning: implicit declaration of function 'TRACE2_INDEX' [-Wimplicit-function-declaration]
   int ndx = TRACE2_INDEX(id);
             ^~~~~~~~~~~~

iluetkeb, 2019-10-25:
We unfortunately left the USBHOST support enabled, which isn't fully working in nuttx.

You can either do menuconfig and unset it yourself, or, if you don't know how to do that, yet, I've also fixed it in master. To use that,

 1. go to `firmware NuttX`
 1. call `git pull`
 1. call `make distclean`
 1. Configure again

chinahuangyong, 2019-10-25:
sorry,i didnot show the hardware platform,and my hardware platform is:
STM32F746g-disco,i want to run uros on it.

closed 2020-01-10 by iluetkeb