micro-ROS/micro_ros_setup #31: Deserializa error for subscription the msg (sensor_msgs_msg_jointstate)

KevinHwangAi, 2019-11-08:
Hi @iluetkeb 
I used stm32f4discovery to install Mirco_ROS. 
When testing the subscription function of uros, I publish a message of "sensor_msgs_msg_jointstate" type from linux, 
and output the following error in NSH:

[rclc_spin_node] error in rcl_take: Typesupport desserialize error., at /home/gemini/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_microxrcedds.c:186, at /home/gemini/uros_ws/firmware/mcu_ws/ros2/rcl/rcl/src/rcl/subscription.c:254

Here is the subscriber_main.c file:
[subscriber_main.txt](https://github.com/micro-ROS/micro-ros-build/files/3823338/subscriber_main.txt)

iluetkeb, 2019-11-08:
Messages containing arrays are not supported in the crystal version of Micro-ROS. The dashing version should be released soon.

Custom messages can be used by adding the message package to the firmware/mcu_ws manually.

closed 2019-11-08 by iluetkeb

KevinHwangAi, 2019-11-08:
@iluetkeb Thanks～