micro-ROS/micro_ros_setup #33: Building firmware CPP typesupport generation is triggered.

BorjaOuterelo, 2019-11-11:
Running `ros2 run micro_ros_setup build_firmware.sh` ends up in interface generation issues:

```shell
Starting >>> std_srvs
--- stderr: builtin_interfaces                                                                                                                                    
CMake Error at /uros_ws/firmware/mcu_ws/install/rosidl_typesupport_c/share/rosidl_typesupport_c/cmake/get_used_typesupports.cmake:34 (message):
  No 'rosidl_typesupport_cpp' found
Call Stack (most recent call first):
  /opt/ros/dashing/share/rosidl_typesupport_cpp/cmake/rosidl_typesupport_cpp_generate_interfaces.cmake:65 (get_used_typesupports)
  /uros_ws/firmware/dev_ws/install/ament_cmake_core/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /uros_ws/firmware/mcu_ws/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:16 (rosidl_generate_interfaces)
```

It does not happen running build_firmware.sh in a shell with no ROS 2 sourced. See
https://github.com/micro-ROS/docker/blob/37e28cfaee944f1549ad605fa434cc6f4561ba18/micro-ROS-Olimex-NuttX/Dockerfile#L18

iluetkeb, 2019-11-12:
Okay, I've found one issue at least. `build_firmware.sh` uses `COLCON_PREFIX_PATH` to determine which prefixes to clean. This variable is only set for crystal, however.

Fortunately, the variable `AMENT_PREFIX_PATH` is set on both versions, and it contains the same data, so we can use that.

iluetkeb, 2019-11-12:
8de7fad45ae1bf288e1c2357ec3ca280b5a603e3 should have fixed this. Can you try again, please?

iluetkeb, 2020-01-10:
Fixes it.

closed 2020-01-10 by iluetkeb