micro-ROS/micro_ros_setup #35: add stm32f7 to it

chinahuangyong, 2019-11-26:


iluetkeb, 2019-11-26:
Which board exactly?

chinahuangyong, 2019-11-28:
stm32f746g-disco

iluetkeb, 2020-01-10:
We cannot support all boards ourselves, but I've written up a tutorial on what is necessary to do.

Look at https://micro-ros.github.io/docs/tutorials/advanced/microros_nuttx_bsp/

Does that help?

gramss, 2020-05-21:
There exists a basic Zephyr Port for this board: https://docs.zephyrproject.org/1.13.0/boards/arm/stm32f746g_disco/doc/stm32f746g_disco.html

So you should be able to port it.
Have a look at the latest commits / PRs at this repo regarding adding the nucleo_f401re board to the scripts under config/zephyr/generic and look at the int32_serial_publisher example under the micro-ROS/zephyr_apps repo. The underlying scripts should work for zephyr boards. The tricky part will be to adjust the prj.conf to your needs. But it has UART supported by the zephyr base dts, so theoretically you should get basic things working.

closed 2020-06-03 by pablogs9