micro-ROS/micro_ros_setup #37: micro-ros-setup 'stm32_usbhost.c' target fails

dodsonmg, 2020-01-06:
I'm trying to follow the setup instructions:

)
```
chip/stm32_usbhost.c:80:52: error: 'TRACE1_NSTRINGS' undeclared here (not in a function)
 static const struct stm32_usbhost_trace_s g_trace1[TRACE1_NSTRINGS] =
                                                    ^~~~~~~~~~~~~~~
chip/stm32_usbhost.c:150:52: error: 'TRACE2_NSTRINGS' undeclared here (not in a function)
 static const struct stm32_usbhost_trace_s g_trace2[TRACE2_NSTRINGS] =
                                                    ^~~~~~~~~~~~~~~
chip/stm32_usbhost.c: In function 'usbhost_trformat1':
chip/stm32_usbhost.c:235:13: warning: implicit declaration of function 'TRACE1_INDEX' [-Wimplicit-function-declaration]
   int ndx = TRACE1_INDEX(id);
             ^~~~~~~~~~~~
chip/stm32_usbhost.c: In function 'usbhost_trformat2':
chip/stm32_usbhost.c:247:13: warning: implicit declaration of function 'TRACE2_INDEX' [-Wimplicit-function-declaration]
   int ndx = TRACE2_INDEX(id);
             ^~~~~~~~~~~~
At top level:
chip/stm32_usbhost.c:150:43: warning: 'g_trace2' defined but not used [-Wunused-variable]
 static const struct stm32_usbhost_trace_s g_trace2[TRACE2_NSTRINGS] =
                                           ^~~~~~~~
chip/stm32_usbhost.c:80:43: warning: 'g_trace1' defined but not used [-Wunused-variable]
 static const struct stm32_usbhost_trace_s g_trace1[TRACE1_NSTRINGS] =
                                           ^~~~~~~~
Makefile:172: recipe for target 'stm32_usbhost.o' failed
make[1]: *** [stm32_usbhost.o] Error 1
make[1]: Leaving directory '/home/broomstick/uros_ws/firmware/NuttX/arch/arm/src'
tools/LibTargets.mk:134: recipe for target 'arch/arm/src/libarch.a' failed
make: *** [arch/arm/src/libarch.a] Error 2
```

pankhurivanjani, 2020-01-07:
Hello !

I also faced the same issue, but I closed the stuff and ran it all over again the very next day and it worked fine. 
I read in the docs that users are facing issue in installing it for the first time but it works well the next time.

dodsonmg, 2020-01-07:
How far did you have to back out?

I re-ran everything starting from step 1 (setting up the workspace).

I tried step 4 without running `make distclean` and it failed, then tried after running `make distclean` and it still failed (with the same error messages.

pankhurivanjani, 2020-01-07:
I had done until the step 4/4. It failed in this one. 

Next day I started with 2/4 again. It worked this time.

iluetkeb, 2020-01-10:
Please update your nuttx checkout (`cd firmware/NuttX ; git pull`), do a distclean and configure again. It should be gone, then.

The root cause is that we had enabled USBHOST-support in the drive_base default config. This caused errors and is not really necessary anyway.

closed 2020-02-07 by iluetkeb