micro-ROS/micro_ros_setup #38: Error in flashing the firmware in Olimex-stm32-e407

pankhurivanjani, 2020-01-07:
Hello! 
I am trying to experiment with micro-ros and stm32 e407. I was able to build and compile the micro-ros and my board is also getting detected, I checked via lsusb. 
```
pankhuri@pankhuri-iMac:~/uros_ws/firmware/NuttX$ lsusb
Bus 002 Device 014: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
```

While flashing the firmware I am getting the error:

```
pankhuri@pankhuri-iMac:~/uros_ws/firmware/NuttX$ sudo scripts/flash.sh olimex-stm32-e407 
Open On-Chip Debugger 0.10.0+dev-01000-gdb23c13d (2020-01-06-14:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Error: no device found
Error: unable to open ftdi device with vid 0483, pid df11, description 'STMicroelectronics STM Device in DFU Mode', serial '338132673536' at bus location '*'

```

I checked the vid, pid and device description of my device and changed it in the olimex-arm-usb-ocd-h.cfg file.

Currently I am using usb cable for flashing the firmware. 

Do I need to use JTAG for this purpose?

Or, can we change the trasport select to enable usb? I tried changing it to libusb in commands but it was in a wrong way perhaps. 

Thanks,

Viplav04, 2020-01-07:
Why are you using sudo script/flash.sh? It should work without sudo.

Yes, you require a JTAG always while flashing. It won't flash without it.

Check this : https://micro-ros.github.io/docs/tutorials/advanced/nuttx_getting_started/
There are some photos and basic access settings which may help during setup.

pankhurivanjani, 2020-01-07:
Yeah it would work without sudo. 

I will check tha page. Thank you for the confirmation!

Viplav04, 2020-01-07:
@pankhurivanjani You can check this after you are able to do some simple settings on the board and get a JTAG. It took me a little while to get the hang of things since I had just begun with embedded and software stuff.

https://github.com/micro-ROS/micro-ros-build/issues/18

pankhurivanjani, 2020-01-07:
Sure.. Thanks a lot!

pankhurivanjani, 2020-01-20:
Hello! I got JTAG cable and got back to this, but I am getting the same error again, 

```
pankhuri@pankhuri-iMac:~/uros_ws/firmware/NuttX$ scripts/flash.sh olimex-stm32-e407
Open On-Chip Debugger 0.10.0+dev-01000-gdb23c13d (2020-01-06-14:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Error: no device found
Error: unable to open ftdi device with vid 15ba, pid 002b, description 'Olimex Ltd. ARM-USB-OCD-H JTAG+RS232', serial '*' at bus location '*'
```

Am I doing something wrong in deployment?

iluetkeb, 2020-01-20:
The flash-script always tries to use the Olimex Tiny-H JTAG adapter, but from your `lsusb` output, you're using something else.

As long as your device is supported by OpenOCD, you should be able to use it. You just need to change the script. It's the `interface` argument to OpenOCD.

pankhurivanjani, 2020-01-21:
Hello @iluetkeb !

I checked my flash script and I believe it is using olimex-arm-usb-ocd adaptor. lsusb also gave the same with vid/pid 15ba/002b

flash.sh

```
#!/bin/bash

if [ $1 = "olimex-stm32-e407" ]; then
  openocd -f interface/ftdi/olimex-arm-usb-ocd-h.cfg -f target/stm32f4x.cfg -c init -c "reset halt" -c "flash write_image erase nuttx.bin 0x08000000"
elif [ $1 = "stm32l1" ]; then
  openocd -f interface/stlink-v2.cfg -f target/stm32l1.cfg -c init -c "reset halt" -c "flash write_image erase nuttx.bin 0x08000000"
else
  echo "Error. Try: ./flash.sh olimex-stm32-e407 or ./flash.sh stm32l1"
fi
```

in openocd my config file is:

```
#
# Olimex ARM-USB-OCD-H
#
# http://www.olimex.com/dev/arm-usb-ocd-h.html
#

interface ftdi
#ftdi_device_desc "Olimex OpenOCD JTAG ARM-USB-OCD-H"

ftdi_device_desc "Olimex Ltd. ARM-USB-OCD-H JTAG+RS232"
#ftdi_vid_pid 0x15ba 0x002b
ftdi_vid_pid 0x15ba 0x002b
#ftdi_serial 338132673536

ftdi_layout_init 0x0908 0x0b1b
ftdi_layout_signal nSRST -oe 0x0200
ftdi_layout_signal nTRST -data 0x0100
ftdi_layout_signal LED -data 0x0800
```

iluetkeb, 2020-01-21:
Hmm. Maybe there is something wrong with the permissions? Check `/dev/bus/usb/<busid>/<deviceid>`

pankhurivanjani, 2020-01-22:
Hello @iluetkeb! 
I checked the permission thing, it doesn't change anything. I am still getting the same error. 

The error says the device not found, so I am wondering about the hardware schematic.

1. I have connected usb to my board and PC.
2. I have connected the JTAG cable (Olimex ARM-USB-OCD-H) to the board and other end to PC

When I connect JTAG to the board the green flashing light disappears. Is this the right configuration?

iluetkeb, 2020-01-22:
What's your "lsusb" output?

> When I connect JTAG to the board the green flashing light disappears. Is this the right configuration?

That's normal.

pankhurivanjani, 2020-01-23:
My lsusb output is:

Bus 002 Device 006: ID 15ba:002b Olimex Ltd. ARM-USB-OCD-H JTAG+RS232

> That's normal. Once you have restarted the board after flashing, it will come on again.

alright!

iluetkeb, 2020-01-23:
okay, that's the right device. so...
 * you have the device specified in your configuration
 * lsusb lists it
 * you have checked device permissions to make sure the device is writeable by your user
 * However, OpenOCD doesn't find it and reports "unable to open".

I have no idea what's going on here, sorry.

What I can say is that none of the flashing process is Micro-ROS specific. This is all just OpenOCD, Olimex JTAG and STM32. Maybe you can google a bit for more information on that combo, e.g. you could try asking Olimex directly. They sold you the device, they should be able to support it.

closed 2020-01-24 by iluetkeb