micro-ROS/micro_ros_setup #39: Error of subscribe to sequences message type

KevinHwangAi, 2020-01-08:
I try subscribe Joy's message or other message which include the sequence element,but all the message fail to subscribe and arise error :

` error in rcl_take: Typesupport desserialize error.`

I checked the joint_state__type_support_c.c file and found some parts of the file are incomplete,
 here is the deserialize function, and the name field is empty:

```
static bool _JointState__cdr_deserialize(
  ucdrBuffer * cdr,
  void * untyped_ros_message)
{
  (void) cdr;
  bool rv = false;
  if (!untyped_ros_message) {
    fprintf(stderr, "ros message handle is null\n");
    return false;
  }
  _JointState__ros_msg_type * ros_message = (_JointState__ros_msg_type *)(untyped_ros_message);
  (void)ros_message;
  // Field name: header
  rv = ((const message_type_support_callbacks_t *)(
      ROSIDL_TYPESUPPORT_INTERFACE__MESSAGE_SYMBOL_NAME(rosidl_typesupport_microxrcedds_c, std_msgs, msg, Header
      )()->data))->cdr_deserialize(cdr, &ros_message->header);
  // Field name: name // Is it necessary?
  {
  }
  // Field name: position
  {
    uint32_t size;
    const size_t capacity = ros_message->position.capacity;
    rv = ucdr_deserialize_sequence_double(cdr, ros_message->position.data, capacity, &size);
  }
  // Field name: velocity
  {
    uint32_t size;
    const size_t capacity = ros_message->velocity.capacity;
    rv = ucdr_deserialize_sequence_double(cdr, ros_message->velocity.data, capacity, &size);
  }
  // Field name: effort
  {
    uint32_t size;
    const size_t capacity = ros_message->effort.capacity;
    rv = ucdr_deserialize_sequence_double(cdr, ros_message->effort.data, capacity, &size);
  }
  return rv;
}
```

I add the code to “Field name: name” :

```
   const size_t capacity = ros_message->name.capacity;
    rv = ucdr_deserialize_string(cdr, ros_message->name.data, capacity);
     if (rv) {
      ros_message->name.size = strlen(ros_message->name.data);
    }
```

But still not working. 

What can I do now?

iluetkeb, 2020-01-10:
This is not a build-system issue. I suggest to move it to https://github.com/micro-ROS/rmw-microxrcedds

closed 2020-01-10 by iluetkeb

KevinHwangAi, 2020-01-10:
@iluetkeb OK.