micro-ROS/micro_ros_setup #54: Leger generator

julibert, 2020-02-03:
I'm following the build instructions at [dashin](https://github.com/micro-ROS/micro-ros-build/tree/dashing/micro_ros_setup) on a clean [docker](https://hub.docker.com/layers/ros/library/ros/dashing/images/sha256-0bd7b270fe8e014d811b929887e892418a2575f0f3a929b27651db09aefeb111) with ROS 2 Dashing, and I find the following error:
```                       
configure: error: can not find a lexer generator (such as lex or flex)
make[2]: *** [kconfig-prefix/src/kconfig-stamp/kconfig-configure] Error 1
make[1]: *** [CMakeFiles/kconfig.dir/all] Error 2
make: *** [all] Error 2
```

I think it should go back to the previous approach and replace
```
colcon build
```
by
```
colcon build --packages-select micro_ros_setup
```

iluetkeb, 2020-02-03:
No, nuttx_kconfig_vendor is necessary, you cannot just ignore it.

Regarding your error: Have you run `rosdep install` before invoking colcon? If that is not in the docs, it should be.

julibert, 2020-02-03:
Hi @iluetkeb, do I need `nuttx_kconfig_vendor` if I want to compile in my host?
I confirm that `rosdep install --from-path src --ignore-src -y` works for me.

iluetkeb, 2020-02-03:
If you only want to build the agent, you do not need `nuttx_kconfig_vendor`.

If you want to build NuttX, regardless of wether for the host or for the microcontroller, you *do* need it.

That said, the presence of `nuttx_kconfig_vendor` in the `micro-ros-build` repository is more or less a historical accident and not really important. We could put it somewhere else if you want. For NuttX builds, we could easily compile it as part of the `dev_ws` and everything would still work just fine.

julibert, 2020-02-07:
Hi @iluetkeb, I added the `rosdep` step in this PR #68.
I propose to move `micro_ros_cmake` and `nuutx_kconfig_vendor` to a separate branch until it is fully implemented.
What do you think?

iluetkeb, 2020-02-07:
OK. It's in progress.

closed 2020-02-07 by iluetkeb