micro-ROS/micro_ros_setup #74: Configuration/Build errors on STM32F4Discovery

takasehideki, 2020-02-16:
Hi, all:
I'm trying to build micro-ROS for Nuttx/STM32F4Discovery with Dashing. But following errors occurred.
Does anyone know helpful information or solutions?

- Configuring firmware
```
$ ros2 run micro_ros_setup configure_firmware.sh uros
~~~[skip]~~~
--- stderr: rcutils                                                         
CMake Warning:
  Manually-specified variables were not used by the project:

    ENABLE_TESTING

---
~~~[skip]~~~
--- stderr: rosidl_typesupport_microxrcedds_cpp
CMake Error at /home/takase/ros2/uros_ws/firmware/mcu_ws/install/rosidl_typesupport_microxrcedds_c/share/rosidl_typesupport_microxrcedds_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package):
  Could not find a configuration file for package "microcdr" that is
  compatible with requested version "".

  The following configuration files were considered but not accepted:

/home/takase/ros2/uros_ws/firmware/mcu_ws/install/microcdr/share/microcdr/cmake/microcdrConfig.cmake, version: 1.1.2 (32bit)

Call Stack (most recent call first):
/home/takase/ros2/uros_ws/firmware/mcu_ws/install/rosidl_typesupport_microxrcedds_c/share/rosidl_typesupport_microxrcedds_c/cmake/rosidl_typesupport_microxrcedds_cConfig.cmake:38 (include)
  CMakeLists.txt:6 (find_package)

---
Failed   <<< rosidl_typesupport_microxrcedds_cpp    [ Exited with code 1 ]
Aborted  <<< rosidl_typesupport_introspection_cpp

~~~[skip]~~~
Summary: 18 packages finished [6.31s]
  1 package failed: rosidl_typesupport_microxrcedds_cpp
  1 package aborted: rosidl_typesupport_introspection_cpp
  2 packages had stderr output: rcutils rosidl_typesupport_microxrcedds_cpp
  35 packages not processed
  Copy files
  Refreshing...
```

- Building firmware
```
$ ros2 run micro_ros_setup build_firmware.sh
~~~[skip]~~~
LD: nuttx
arm-none-eabi-ld --entry=__start -nostartfiles -nodefaultlibs -T/home/takase/ros2/uros_ws/firmware/NuttX/configs/stm32f4discovery/scripts/ld.script -L"/home/takase/ros2/uros_ws/firmware/NuttX/staging" -L"/home/takase/ros2/uros_ws/firmware/NuttX/arch/arm/src/board" -L "/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7e-m" \
    -o "/home/takase/ros2/uros_ws/firmware/NuttX/nuttx"   \
    --start-group -lsched -ldrivers -lconfigs -lc -lmm -larch -lxx -lapps -lnet -lfs -lbinfmt -lxx -lboard -lsupc++ "/usr/lib/gcc/arm-none-eabi/6.3.1/thumb/v7e-m/libgcc.a" --end-group
/home/takase/ros2/uros_ws/firmware/NuttX/staging/libarch.a(up_initialize.o): In function `up_initialize':
up_initialize.c:(.text+0x26): undefined reference to `up_netinitialize'
/home/takase/ros2/uros_ws/firmware/NuttX/staging/libapps.a(librcutils_logging.c.obj): In function `rcutils_logging_console_output_handler':
/home/takase/ros2/uros_ws/firmware/mcu_ws/uros/rcutils/src/logging.c:817: undefined reference to `isatty'
Makefile:184: recipe for target 'nuttx' failed
make[1]: *** [nuttx] Error 1
make[1]: Leaving directory '/home/takase/ros2/uros_ws/firmware/NuttX/arch/arm/src'
tools/Makefile.unix:422: recipe for target 'pass2' failed
make: *** [pass2] Error 2
```

jfm92, 2020-03-04:
Hi @takasehideki 

Unfortunately, at this moment the board that you're using is not supported. The issue building the firmware is a bad configuration that is not set properly the ethernet and the serial peripheral.
On this file ([uROS config profile](https://github.com/micro-ROS/NuttX/blob/drive_base/configs/olimex-stm32-e407/uros/defconfig) ), you can find the configuration that works fine for our NuttX default board. You can adapt it to your specific board.

takasehideki, 2020-03-04:
@jfm92 Thanks for reply.

The reasons why I wanna use Nuttx/STM32F4Discovery are that it was listed on `[Supported platforms](https://github.com/micro-ROS/micro-ros-build/tree/dashing/micro_ros_setup#supported-platforms)` and I only have its board. Of course, I understand its board is not suitable because it does not have Ethernet port. I wanna suggest its board might be eliminated as Supported platforms.

I will buy Olimex STM32-E407 board and try it again.

pablogs9, 2020-06-03:
Closing since STM32F4Discovery is not currently supported.

closed 2020-06-03 by pablogs9