micro-ROS/micro_ros_setup #75: create_agent_ws.sh fails with rosdep keys for fastrtps not resolvable

lukmi15, 2020-02-17:
- [x] I have searched open and closed issues for duplicates
  

---

### Steps to Reproduce

Using this guide: https://github.com/micro-ROS/micro-ros-build/blob/crystal/micro_ros_setup/README.md
```
student@ce-pc3:~/ba/micro-ros$ ls -A
tutorial.txt
student@ce-pc3:~/ba/micro-ros$ source /opt/ros/crystal/setup.bash
student@ce-pc3:~/ba/micro-ros$ sudo apt install python-rosdep
[sudo] password for student: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-rosdep is already the newest version (0.18.0-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
student@ce-pc3:~/ba/micro-ros$ mkdir -p uros_ws/src
student@ce-pc3:~/ba/micro-ros$ cd uros_ws
student@ce-pc3:~/ba/micro-ros/uros_ws$ git clone --recursive https://github.com/micro-ROS/micro-ros-build.git src/micro-ros-build
Cloning into 'src/micro-ros-build'...
remote: Enumerating objects: 202, done.
remote: Counting objects: 100% (202/202), done.
remote: Compressing objects: 100% (117/117), done.
remote: Total 1111 (delta 118), reused 132 (delta 65), pack-reused 909
Receiving objects: 100% (1111/1111), 263.35 KiB | 1.01 MiB/s, done.
Resolving deltas: 100% (682/682), done.
student@ce-pc3:~/ba/micro-ros/uros_ws$ colcon build --packages-select micro_ros_setup
Starting >>> micro_ros_setup
Finished <<< micro_ros_setup [1.66s]                  

Summary: 1 package finished [1.77s]
student@ce-pc3:~/ba/micro-ros/uros_ws$ source install/local_setup.bash
student@ce-pc3:~/ba/micro-ros/uros_ws$ ros2 run micro_ros_setup create_agent_ws.sh  # add agent packages
/home/student/ba/micro-ros/uros_ws/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:10: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_info = yaml.load(sys.stdin)
/home/student/ba/micro-ros/uros_ws/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:11: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_keep = yaml.load(open(sys.argv[1]))['keep'].split()
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
micro_ros_agent: Cannot locate rosdep definition for [fastrtps]
student@ce-pc3:~/ba/micro-ros/uros_ws$ 
```

### Operating System:

Ubuntu 18.04.4 LTS
Kernel release: 5.3.0-28-generic

---

Does anyone know how to resolve this?

FranFin, 2020-07-20:
@lukmi15 Hi! Thank you for your interest in micro-ROS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460). You’re invited to bring your micro-ROS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.

lukmi15, 2020-07-20:
Thank you for the invitation! Sadly, I will not have the time to participate.

closed 2022-01-28 by pablogs9