micro-ROS/micro_ros_setup #101: FreeRTOS build broken after micro-ROS#98 Commit

Kelly-Coffey, 2020-04-01:
I am seeing the following error below when performing the "FreeRTOS Getting Started Demo"
I reverted mirco-ROS#98 and the build completes without error.

In file included from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.h:20,
                 from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_client.c:15:
/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:60:3: error: unknown type name 'uxrSerialPlatform'
   uxrSerialPlatform serial_platform;
   ^~~~~~~~~~~~~~~~~
make[3]: *** [CMakeFiles/rmw_microxrcedds.dir/src/rmw_client.c.obj] Error 1
make[2]: *** [CMakeFiles/rmw_microxrcedds.dir/all] Error 2
make[1]: *** [all] Error 2

Kelly-Coffey, 2020-04-01:
I did not think a change in the Zephyr config would impact FreeRTOS so further investigation I found I can get it to work if I run.
$ sudo apt update
after 
# Create step
ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407

The same requirement to run "sudo apt update" to get the build to complete without error is needed before the micro-ROS#98 commit.
I will leave open for another day for others to comment if this is expected.

closed 2020-04-02 by Kelly-Coffey

genemerewether, 2020-04-20:
@Kelly-Coffey what specifically worked for you here? I'm running into the same (or a similar?) problem, and running apt update in between create_firmware_ws and build_firmware didn't fix things