micro-ROS/micro_ros_setup #108: Creating complex FreeRTOS apps

krm-mir, 2020-04-23:
Hi microROS

I'm working with the Olimex board running microROS with FreeRTOS.

I'm trying to create more complex apps that consist of more files than the app.c.
Specifically i have:
- app.c
- app-colcon.meta
- ros_handle.h
- sensor_handle.h

I don't want to touch the makefile for consistency, therefore I'm creating a symbolic link between my headerfiles and the <extension_dir>/Inc directory, so at least my own headers are included in the build. This works for now.

Ultimately I would like to have a ros_handle.c/cpp and sensor.c/cpp, but I can't figure out how to add them to the C_SOURCES.
Any solutions to this?

Also, being able the spin multible threads with microROS running would be awesome (maybe a bit into the future?).

pablogs9, 2020-04-23:
We are currently working on multiple threads with different micro-ROS nodes, they would be available soon.

Regarding your multiple files question, would it work modifying [this line](https://github.com/micro-ROS/freertos_apps/blob/00271ee291900f491be48bae19d8e3198a0ee6b3/microros_olimex_e407_extensions/Makefile#L164) to:

```
$(UROS_APP_FOLDER)/*.c
```

and adding your app folder to includes?

Please feel free to PR your changes to the repository, they are welcome. For my part when dealing with micro-ROS + FreeRTOS + Olimex board it is required that regenerating the project with STMCubeMX does not have a conflict with your Makefile changes.

krm-mir, 2020-04-24:
It should work adding:
```
$(wildcard $(UROS_APP_FOLDER)/*.c)
```
to C_SOURCES and
```
-I$(UROS_APP_FOLDER)
```
to C_INCLUDES.

I can create a pull-request soon.

pablogs9, 2020-04-24:
Willing to see that PR ;)

krm-mir, 2020-05-13:
[this PR](https://github.com/micro-ROS/freertos_apps/pull/3) solves the issue

closed 2020-05-13 by krm-mir