micro-ROS/micro_ros_setup #110: Branch clean

pablogs9, 2020-04-29:
Please consider removing the following branches if they are not required any more:

- [ ] https://github.com/micro-ROS/micro-ros-build/tree/feature/agent_lite
- [ ] https://github.com/micro-ROS/micro-ros-build/tree/feature/mempools_refactor
- [ ] https://github.com/micro-ROS/micro-ros-build/tree/feature/parameters
- [ ] https://github.com/micro-ROS/micro-ros-build/tree/feature/tof_sensor <- Is this used on the Malaga demo? Should it be a tag?

pablogs9, 2021-12-17:
@Acuadros95 can we remove the branch and close this?

Acuadros95, 2021-12-17:
Yes, and maybe we should also delete the rest of old branches:
- [https://github.com/micro-ROS/micro_ros_setup/tree/feature/freertos_host](https://github.com/micro-ROS/micro_ros_setup/tree/feature/freertos_host)
- [https://github.com/micro-ROS/micro_ros_setup/tree/feature/hardware-bridge](https://github.com/micro-ROS/micro_ros_setup/tree/feature/hardware-bridge)

pablogs9, 2021-12-17:
Not sure about the others. Closing this

closed 2021-12-17 by pablogs9