micro-ROS/micro_ros_setup #111: Potential Micro-CDR version mismatch

driftregion, 2020-04-30:
Compilation of Micro-XRCE-DDS-Client failed as follows:
```
/home/njames/uros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/core/session/stream/input_reliable_stream.c:217:11: error: 'ucdrBuffer' {aka 'struct ucdrBuffer'} has no member named 'offset'
         ub->offset);
           ^~
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:79: CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/input_reliable_stream.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:105: CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/output_reliable_stream.c.obj] Error 1
```
I've seen this [exact compilation error before](https://github.com/eProsima/Fast-RTPS/issues/1178#issuecomment-618889798), but that was due to my mistakenly installing the incorrect version of `Micro-CDR` as root.  I think the cause this time is different.

# `Micro-CDR`
I was able to find the missing struct in `Micro-CDR` with
`git grep 'ucdr_init_buffer_origin_offset' $(git rev-list --all)`
Checking out the matching commit:
```
commit 32e872511e9d5b5d8a2628f22aa93399a3a69369 (HEAD, origin/develop, origin/dashing)
Author: Julián Bermúdez Ortega <julianbermudez@eprosima.com>
Date:   Tue Apr 7 13:38:36 2020 +0200
```
allowed the client to compile successfully

The original commit checked out by CMake was:
```
commit b156463fbba8b670c904c9ba0727fd5454b0db7b (HEAD, tag: v1.1.2, origin/master, origin/HEAD, master)
Author: Julián Bermúdez Ortega <julianbermudez@eprosima.com>
Date:   Fri Dec 13 09:48:26 2019 +0100

    Fix dynamic library soversion name. (#46)
```

# `Micro-XRCE-DDS-Client`
```
commit 91f234a3216cfa501e48badc09990d376b587fa5 (HEAD -> dashing, origin/develop, origin/dashing)
Author: Julián Bermúdez Ortega <julianbermudez@eprosima.com>
Date:   Tue Apr 28 17:55:21 2020 +0200
```
# To Fix This
It looks like the version of `Micro-CDR` is set in a couple of places:
1.  `$RTOS/$PLATFORM/client_uros_packages.repos`
1.  `Micro-XRCE-DDS-Client/CMakeLists.txt`

I tried changing 2 first, but see that it's not respected when invoking `build_firmware.sh`
Removing `firmware` and changing `client_uros_packages.repos` to
```
  eProsima/Micro-CDR:
    type: git
    url: https://github.com/eProsima/micro-CDR.git
    version: dashing
```
doesn't seem to work either.

julibert, 2020-04-30:
Hi @driftregion,

This issue was fixed with #114.

driftregion, 2020-05-06:
Confirmed, thanks @julibert

closed 2020-05-06 by driftregion