micro-ROS/micro_ros_setup #112: fatal error: arpa/inet.h: No such file or directory

driftregion, 2020-04-30:
After bypassing #111, I find the following:
```
/home/njames/uros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/ip_posix.c:17:10: fatal error: arpa/inet.h: No such file or directory
 #include <arpa/inet.h>
          ^~~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:404: CMakeFiles/microxrcedds_client.dir/src/c/profile/transport/ip/ip_posix.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:76: CMakeFiles/microxrcedds_client.dir/all] Error 2
make[1]: *** [Makefile:133: all] Error 2
```
I think there must be some shim between this `#include` and `FreeRTOS-Plus-POSIX` or `LwIP` that I'm missing.

krm-mir, 2020-04-30:
I have the same issue.

julibert, 2020-04-30:
Hi @driftregion and @krm-mir, 

This issue was fixed with #114.

driftregion, 2020-05-06:
Thanks!  I see the client fix.
On my machine, now it fails a little later with the following error:
```
/home/njames/uros_ws/firmware/mcu_ws/install/rcutils/include/rcutils/types/array_list.h:23:10: error: 'string.h' file not found [clang-diagnostic-error]
#include <string.h>
         ^
```
This looks simple enough, but I'm not yet familiar enough with the sequence of build tool invocations that gets us to this point to figure out where the error is.

driftregion, 2020-05-07:
The error is tagged as a `[clang-diagnostic-error]`, and yet the compiler is decidedly not clang.  Those errors appear to be coming from `clang-tidy` a CMake plugin.
These posts appear related:
- https://stackoverflow.com/questions/57851621/clang-tidy-with-cmake-and-an-embedded-target
- https://stackoverflow.com/questions/57082124/how-to-integrate-clang-tidy-to-cmake-and-gcc

I'm able to get around this and compile `rmw_microxrcedds` by removing `clang-tidy` from
`uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/CMakeLists.txt`

driftregion, 2020-05-11:
closing because original issue is resolved

closed 2020-05-11 by driftregion