micro-ROS/micro_ros_setup #115: Zephyr failed build RMW with new board STM32F401 Nucleo-64 - nucleo_f401re

mrlowalowa, 2020-05-10:
Hello,

I wanted to build micro-ROS for a new board.
Similar to #74 the build process failed during setting up the rmw_implementation.

here is the error code:

```
--- stderr: rmw_microxrcedds                                                                                 
CMake Warning at CMakeLists.txt:153 (find_package):
  By not providing "Findrosidl_typesupport_microxrcedds_cpp.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "rosidl_typesupport_microxrcedds_cpp", but
  CMake did not find one.

  Could not find a package configuration file provided by
  "rosidl_typesupport_microxrcedds_cpp" with any of the following names:

    rosidl_typesupport_microxrcedds_cppConfig.cmake
    rosidl_typesupport_microxrcedds_cpp-config.cmake

  Add the installation prefix of "rosidl_typesupport_microxrcedds_cpp" to
  CMAKE_PREFIX_PATH or set "rosidl_typesupport_microxrcedds_cpp_DIR" to a
  directory containing one of the above files.  If
  "rosidl_typesupport_microxrcedds_cpp" provides a separate development
  package or SDK, be sure it has been installed.

---
Finished <<< rmw_microxrcedds [0.69s]
Starting >>> rmw_implementation
Finished <<< rcl_logging_noop [0.54s]                                                            
Starting >>> libyaml_vendor
--- stderr: rmw_implementation                                                                     
CMake Error at /opt/ros/dashing/share/rosidl_typesupport_fastrtps_cpp/cmake/rosidl_typesupport_fastrtps_cpp-extras.cmake:6 (find_package):
  Could not find a configuration file for package "fastcdr" that is
  compatible with requested version "".

  The following configuration files were considered but not accepted:

    /opt/ros/dashing/share/fastcdr/cmake/fastcdr-config.cmake, version: 1.0.10 (64bit)
    /usr/local/lib/cmake/fastcdr/fastcdr-config.cmake, version: 1.0.13 (64bit)

Call Stack (most recent call first):
  /opt/ros/dashing/share/rosidl_typesupport_fastrtps_cpp/cmake/rosidl_typesupport_fastrtps_cppConfig.cmake:38 (include)
  /opt/ros/dashing/share/rmw_fastrtps_cpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/dashing/share/rmw_fastrtps_cpp/cmake/rmw_fastrtps_cppConfig.cmake:38 (include)
  /home/flo/microros_ws/firmware/mcu_ws/install/share/rmw_implementation_cmake/cmake/get_default_rmw_implementation.cmake:62 (find_package)
  CMakeLists.txt:25 (get_default_rmw_implementation)

---
Failed   <<< rmw_implementation	[ Exited with code 1 ]
Aborted  <<< libyaml_vendor                                     
                                  
Summary: 15 packages finished [5.03s]
  1 package failed: rmw_implementation
  1 package aborted: libyaml_vendor
  2 packages had stderr output: rmw_implementation rmw_microxrcedds
  32 packages not processed
Makefile:57: recipe for target 'colcon_compile' failed
make[3]: *** [colcon_compile] Error 1
make[3]: Leaving directory '/home/flo/microros_ws/firmware/zephyr_apps/microros_extensions'
CMakeFiles/microroslib_project.dir/build.make:134: recipe for target 'microros_extensions/src/microroslib_project-stamp/microroslib_project-build' failed
make[2]: *** [microros_extensions/src/microroslib_project-stamp/microroslib_project-build] Error 2
make[2]: Leaving directory '/home/flo/microros_ws/firmware/zephyrproject/build'
CMakeFiles/Makefile2:2087: recipe for target 'CMakeFiles/microroslib_project.dir/all' failed
make[1]: *** [CMakeFiles/microroslib_project.dir/all] Error 2
make[1]: Leaving directory '/home/flo/microros_ws/firmware/zephyrproject/build'
Makefile:106: recipe for target 'all' failed
make: *** [all] Error 2
```

Can somebody give me a hint about how to add the necessary config/dtb/??? for the rmw to use the serial with this board? I tried to re-engineer your different scripts and examples but I'm still not sure where I must add this board.. This is more the fault of the complex build complex by zephyr, but I hope you can give me a few hints about it.. :smile: 

It would be nice to increase the documentation about setting up an own board.

Side fact: I needed to get a newer cmake version via pip3 in order to start building this. The scripts worked sort of straight forward after adding  `nucleo_f401re` as new board descriptor here and there. But a nice documentation about this would help a lot! I might even be willing to help writing it.

pablogs9, 2020-05-11:
Which RTOS are you using?

gramss, 2020-05-11:
This is the command I'm using to invoke the build process (including the current folder, that is handled as a active working space by west):
```~/microros_ws/firmware/zephyrproject$ west build -b nucleo_f401re -p auto /home/flo/microros_ws/firmware/zephyr_apps/apps/ping_pong/ -- -G'Unix Makefiles' -DCMAKE_VERBOSE_MAKEFILE=ON```

@pablogs9 , can you help me with this zephyr environment? :) We have 90-100 students per year that are currently implementing a own serial communication with ROS1 melodic and this board. I would like to propose to change that to use an underlying RTOS + microROS + ROS2

pablogs9, 2020-05-12:
Hello @gramss, these are the steps I have followed:

```
mkdir uros_ws; cd uros_ws;
source /opt/ros/dashing/setup.bash
git clone -b dashing https://github.com/micro-ROS/micro-ros-build.git src/micro-ros-build
rosdep update && rosdep install --from-path src --ignore-src -y
colcon build
source install/local_setup.bash

ros2 run micro_ros_setup create_firmware_ws.sh zephyr nucleo_f401re
ros2 run micro_ros_setup configure_firmware.sh int32_serial_publisher -t serial
ros2 run micro_ros_setup build_firmware.sh
```

This builds on my environment but I don't have the board to test the flash and the operation.

I see that this board only has the `UART_2` enabled so the serial transport for micro-ROS uses it and it is expected to have the Zephyr console disabled.

Please notice that I have made changes in the build system repo and in the `zephyr_apps` repo, here they are:

https://github.com/micro-ROS/zephyr_apps/pull/2
https://github.com/micro-ROS/micro-ros-build/pull/117

We are merging them, but it would be great if you test them on hardware and give us feedback.

gramss, 2020-05-12:
@pablogs9 thank you for the fast feature integration! I'm currently rebuilding the environment to check if everything works as aspected. Here are my notes: 

Minor "error" during create_firmware_ws.sh (was there probably before):
```
#All required rosdeps installed successfully
[sudo] password for flo: 

OK
/home/flo/uros_ws/install/micro_ros_setup/config/zephyr/generic/create.sh: line 5: /etc/apt/sources.list.d/kitware.list: Permission denied
Error while installing CMake version >= 3.13.1. Check with cmake --version
Please if not installed follow the instructions: https://docs.zephyrproject.org/latest/getting_started/index.html
```

I installed cmake via pip3.
 ``` 
$cmake --version
cmake version 3.17.2
```
I'm only sceptical about this because it just got root and then complains about denied permissions..

#117 was already merged and micro-ROS/zephyr_apps#2 was checkout out manually.

Build process also worked for me.

The flash script obviously failed: https://github.com/micro-ROS/micro-ros-build/blob/dashing/micro_ros_setup/config/zephyr/generic/flash.sh
besides `cd` into the firmware folder and setting up the env variables, west did the flashing as expected. I would suggest that you try to flash all unknown boards with the flash tool and after failing fail the rest/script.

I can confirm that my previous blink sketch is gone. (led stopped blinking)
Currently I'm building the agent from the NuttX documentation/tutorial. Should be the same..
ehm... can you provide a ubuntu package for this in the future..? I believe that students will get pretty frustrated when they all have to build 20-40 minutes or so.. depending on hardware..
Even had a TLS error during agent build (30 minutes in) and now I have to clean everything again.. damn..

```
colcon build --metas src
Starting >>> microxrcedds_agent
Starting >>> drive_base_msgs
Finished <<< drive_base_msgs [0.44s]                                       
Starting >>> micro_ros_setup
Finished <<< micro_ros_setup [0.17s]                                       
[Processing: microxrcedds_agent]                             
[Processing: microxrcedds_agent]                                     
--- stderr: microxrcedds_agent                                          
Cloning into 'cli11'...
CMake Warning (dev) at /usr/local/lib/python3.6/dist-packages/cmake/data/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  cmake/common/eprosima_libraries.cmake:96 (find_package)
  CMakeLists.txt:220 (eprosima_find_thirdparty)
This warning is for project developers.  Use -Wno-dev to suppress it.

Cloning into 'microxrcedds_client'...
Note: checking out 'v1.7.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 49ac989 Fixes to the readme
Submodule 'extern/googletest' (https://github.com/google/googletest.git) registered for path 'extern/googletest'
Submodule 'extern/json' (https://github.com/nlohmann/json.git) registered for path 'extern/json'
Submodule 'extern/sanitizers' (https://github.com/arsenm/sanitizers-cmake) registered for path 'extern/sanitizers'
[... 34 lines cut ...]
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /home/flo/uros_ws/build/microxrcedds_agent/temp_install/fastrtps-1.8.3/share/fastrtps/cmake/fastrtps-config.cmake:50 (find_package)
  CMakeLists.txt:126 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< microxrcedds_agent [1min 24s]
Starting >>> micro_ros_agent
Finished <<< micro_ros_agent [4.29s]                           

Summary: 4 packages finished [1min 28s]
  1 package had stderr output: microxrcedds_agent

```

missed a dependency.. `sudo apt install libtinyxml2-dev`

agent is now running:
`[1589289146.239714] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3`

ros2 topic list with sourced dashing env does not list any topics.
The board is connected via USB. Does Init indicate a valid connection to the Zephyr App or is this working as soon as I connect it with any serial device..? I do not have my FTDI breakout board around.. But it would be great if we can make the connection work over the onboard USB interface.

Thank you for your help!! I think we can get the last bit going as well! :smiley: 
But I still would love some more documentation about how to setup a new board like you did. I can learn a lot from your changes but a written tutorial would be of course better.. :+1:

pablogs9, 2020-05-12:
Hello @gramss, the transport (UART or UART over USB) is selected in the configure step as you can see. If you check the configure.sh script you will see that there are two of them: the recently added serial (not tested) and the usb-serial (this is the one we use with STM32L4 Discovery kit IoT board).

I don't know the serial interfaces of your specific board but I guess that there is only one enabled (UART_2). This serial port may be used to communicate between micro-ROS agent and the board, so you shouldn't be able to use it as a "print()" interface. I have set its configuration here: https://github.com/micro-ROS/zephyr_apps/blob/f420820774fbda0bf51cba53c37fb0eebfea2308/apps/int32_serial_publisher/prj.conf#L16.

Does your board still print messages through this interface?

Which command are you using to run the agent? It should be something like `ros2 run micro_ros_agent micro_ros_agent serial --dev [YOUR BOARD PORT] -v6`. The last parameter will enable the debug output, it is pretty useful when debugging a new board.

gramss, 2020-05-12:
so I get this here with the serial configured. UART2 is connected to the USB. The board it self does not have a own USB interface. That is handled by the on-board STLink Programmer that works as ann onboard FTDI Chip.

I just made a cat on my serial device with the firmare configured with serial:

```
cat /dev/ttyACM0 
*** Booting Zephyr OS build v2.2.0-rc1-115-g38b77be6c8ba  ***
~~~~~~~~~~~~~~~~~~~~Failed status on line 26: 1. Aborting.
Failed status on line 31: 101. Aborting.
Failed status on line 35: 200. Aborting.
```
This is the Zephyr shell I would suggest? So first it is failing and second the config for serial is not right?

serial-USB does not build for me as the board probably do not have a own usb port:
```
[ 95%] Linking C executable zephyr_prebuilt.elf
cd /home/flo/uros_ws/firmware/build/zephyr && /usr/local/lib/python3.6/dist-packages/cmake/data/bin/cmake -E cmake_link_script CMakeFiles/zephyr_prebuilt.dir/link.txt --verbose=1
ccache /home/flo/uros_ws/firmware/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc    CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr_prebuilt.elf  -Wl,-T linker.cmd -Wl,-Map=/home/flo/uros_ws/firmware/build/zephyr/zephyr_prebuilt.map -Wl,--whole-archive ../app/libapp.a libzephyr.a arch/common/libarch__common.a arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a arch/arch/arm/core/aarch32/cortex_m/mpu/libarch__arm__core__aarch32__cortex_m__mpu.a lib/libc/newlib/liblib__libc__newlib.a lib/posix/liblib__posix.a soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a boards/arm/nucleo_f401re/libboards__arm__nucleo_f401re.a drivers/gpio/libdrivers__gpio.a drivers/serial/libdrivers__serial.a ../modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a -Wl,--no-whole-archive kernel/libkernel.a CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -L"/home/flo/uros_ws/firmware/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/thumb/v7e-m/nofp" -L/home/flo/uros_ws/firmware/build/zephyr -lgcc -Wl,--print-memory-usage /home/flo/uros_ws/firmware/zephyr_apps/apps/int32_serial_publisher/../../../mcu_ws/install/libmicroros.a -mthumb -mcpu=cortex-m4 -Wl,--gc-sections -Wl,--build-id=none -Wl,--sort-common=descending -Wl,--sort-section=alignment -Wl,-u,_OffsetAbsSyms -Wl,-u,_ConfigAbsSyms -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--orphan-handling=warn -lstdc++ -mabi=aapcs -march=armv7e-m -lm -lc -lgcc -L"/home/flo/uros_ws/firmware/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi"/lib/thumb/v7e-m/nofp 
Memory region         Used Size  Region Size  %age Used
           FLASH:      115136 B       512 KB     21.96%
            SRAM:       45412 B        96 KB     46.20%
        IDT_LIST:         200 B         2 KB      9.77%
/home/flo/uros_ws/firmware/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/flo/uros_ws/firmware/zephyr_apps/apps/int32_serial_publisher/../../../mcu_ws/install/libmicroros.a(libmicroxrcedds_client-serial_transport_external.c.obj): in function `uxr_init_serial_platform':
/home/flo/uros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/serial/serial_transport_external.c:63: undefined reference to `usb_enable'
/home/flo/uros_ws/firmware/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/flo/uros_ws/firmware/zephyr_apps/apps/int32_serial_publisher/../../../mcu_ws/install/libmicroros.a(libmicroxrcedds_client-serial_transport_external.c.obj): in function `uart_line_ctrl_get':
/home/flo/uros_ws/firmware/build/zephyr/include/generated/syscalls/uart.h:274: undefined reference to `z_impl_uart_line_ctrl_get'
/home/flo/uros_ws/firmware/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/flo/uros_ws/firmware/zephyr_apps/apps/int32_serial_publisher/../../../mcu_ws/install/libmicroros.a(libmicroxrcedds_client-serial_transport_external.c.obj): in function `uart_line_ctrl_set':
/home/flo/uros_ws/firmware/build/zephyr/include/generated/syscalls/uart.h:261: undefined reference to `z_impl_uart_line_ctrl_set'
/home/flo/uros_ws/firmware/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/flo/uros_ws/firmware/build/zephyr/include/generated/syscalls/uart.h:261: undefined reference to `z_impl_uart_line_ctrl_set'
/home/flo/uros_ws/firmware/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/flo/uros_ws/firmware/zephyr_apps/apps/int32_serial_publisher/../../../mcu_ws/install/libmicroros.a(libmicroxrcedds_client-serial_transport_external.c.obj): in function `uart_line_ctrl_get':
/home/flo/uros_ws/firmware/build/zephyr/include/generated/syscalls/uart.h:274: undefined reference to `z_impl_uart_line_ctrl_get'
collect2: error: ld returned 1 exit status
zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:124: recipe for target 'zephyr/zephyr_prebuilt.elf' failed
make[2]: *** [zephyr/zephyr_prebuilt.elf] Error 1
make[2]: Leaving directory '/home/flo/uros_ws/firmware/build'
CMakeFiles/Makefile2:2264: recipe for target 'zephyr/CMakeFiles/zephyr_prebuilt.dir/all' failed
make[1]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make[1]: Leaving directory '/home/flo/uros_ws/firmware/build'
Makefile:106: recipe for target 'all' failed
make: *** [all] Error 2
FATAL ERROR: command exited with status 2: /usr/local/bin/cmake --build /home/flo/uros_ws/firmware/build
```

I try to add the west flash as a PR yes.

pablogs9, 2020-05-12:
Well, the micro-ROS code is running as we can see in these `Failed status on line 31: 101. Aborting.
` but we have to make sure that we are using the UART interface only for client-to-agent communication and the `printk()` function uses other UART (or none). You can check the micro-ROS transport here: https://github.com/micro-ROS/zephyr_apps/blob/feature/add_nucleo_f401re_support/microros_extensions/zephyr_serial_transport.c

pablogs9, 2020-05-12:
Can you test with `UART_CONSOLE=n` on the `prj.conf. Or setting `CONFIG_UART_CONSOLE_ON_DEV_NAME` to another UART?

gramss, 2020-05-12:
yes, I just found that these were not errors by zyphr OS but prinntk() by your main.c

These fatals are fatal because they do not connect to the agent I would suggest? I just want to make sure to process these error codes somehow.. :D 

maybe put printk into log: "CONFIG_LOG_PRINTK: Redirect printk calls to the logger." https://docs.zephyrproject.org/2.0.0/reference/kconfig/CONFIG_LOG_PRINTK.html#cmdoption-arg-config-log-printk and then turn off log? https://docs.zephyrproject.org/2.0.0/reference/kconfig/CONFIG_LOG.html#cmdoption-arg-config-log

But how can we test that we are writing to UART2 for the micro_ros part..? You are saying "only" but I'm not seeing anything from micro-DDS implementation. I belive that they would send out something when they try to configure the publisher?

gramss, 2020-05-12:
Building with firmare config on serial again:
added `UART_CONSOLE=n` on the last line. Didn't seem to like it.

warning: PRINTK (defined at subsys/debug/Kconfig:108) was assigned the value 'n' but got the value
'y'. See http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_PRINTK.html and/or look up
PRINTK in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

/home/flo/uros_ws/firmware/zephyr_apps/apps/int32_serial_publisher/prj.conf:22: warning: ignoring malformed line 'UART_CONSOLE=n'

error: Aborting due to Kconfig warnings

CMake Error at /home/flo/uros_ws/firmware/zephyrproject/zephyr/cmake/kconfig.cmake:216 (message):
  command failed with return code: 1

pablogs9, 2020-05-12:
In fact you have these `~` which does not come from any `printk()` in the main, so I guess that they are part of the binary data sent by the underlying Micro XRCE-DDS implementation of micro-ROS. 

Let me know if any of these config parameters disables correctly the `printk()` functionality. I also would use another serial tool such as miniterm or minicom instead of `cat` because they have options to watch raw binary data (to check if this micro-ROS client is effectively trying to talk to its agent).

pablogs9, 2020-05-12:
Regarding error management you can check the RCL API in order to implement a error prune logic. I have done something similar for other platform check this: https://github.com/micro-ROS/freertos_apps/blob/feature/dual_transports/apps/crazyflie_position_publisher_reconnection/app.c

gramss, 2020-05-12:
> In fact you have these `~` which does not come from any `printk()` in the main, so I guess that they are part of the binary data sent by the underlying Micro XRCE-DDS implementation of micro-ROS.

Yes, but they were animated and looked more like a fancy zephyr startup "logo". At least to me. Have you seen a Zephyr OS shell with 0.2.2?

pablogs9, 2020-05-12:
Let's try to disable that console output and then we can check if the transport is doing its job.

gramss, 2020-05-12:
> Let's try to disable that console output and then we can check if the transport is doing its job.

Yes, have you seen my two approaches here? https://github.com/micro-ROS/micro-ros-build/issues/115#issuecomment-627419177
I updated the comment for completeness. Maybe it got lost

gramss, 2020-05-12:
`CONFIG_UART_CONSOLE=n` is building currently

pablogs9, 2020-05-12:
Another question, how are you using Zephyr RTOS 0.2.2? It should be working with a local branch in one of our repos, check this: https://github.com/micro-ROS/micro-ros-build/blob/1dd94f43cbddd332ce9bcb609078f9b70eedce22/micro_ros_setup/config/zephyr/generic/create.sh#L38

It is not checked to work with newer version, and as they are updating fast, we have to be sure that any updagrade does not break any compatibility.

gramss, 2020-05-12:
excellent question..! :100: 
I did alter with zephyr OS a bit and managed to get the blinky sketch running. As mentioned earlier.. But the sdk was already removed from /opt/ and only in a local git folder..
I worked from a new clean shell.. maybe the .zephyrrc in my home folder got picked up by west..?

Now I cleaned every zephyr sdk I could find from my OS and rebuild it.

With `CONFIG_UART_CONSOLE=n` there was nothing on the serial line. I removed that now again. 
I got again the 0.2.2 Zephr OS.. I totally removed the firmware folder now and rebuild the firmware from scratch again.. Probably that helps

pablogs9, 2020-05-12:
Regarding the Zephyr version, can you check that you are using the eProsima fork at micro_ros branch?

gramss, 2020-05-12:
Here is the firmware setup output:

```
ros2 run micro_ros_setup create_firmware_ws.sh zephyr nucleo_f401re
Creating firmware for zephyr platform nucleo_f401re
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Add distro "dashing"
Add distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
updated cache in /home/flo/.ros/rosdep/sources.cache
#All required rosdeps installed successfully
/home/flo/uros_ws_new/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:10: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_info = yaml.load(sys.stdin)
/home/flo/uros_ws_new/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:11: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_keep = yaml.load(open(sys.argv[1]))['keep'].split()
List of repositories is empty
#All required rosdeps installed successfully
/home/flo/uros_ws_new/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:10: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_info = yaml.load(sys.stdin)
/home/flo/uros_ws_new/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:11: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_keep = yaml.load(open(sys.argv[1]))['keep'].split()
Crosscompiled environment: cleaning path
Starting >>> ament_package
Starting >>> ament_lint
Finished <<< ament_lint [0.78s]                                              
Starting >>> ament_flake8
Finished <<< ament_package [0.80s]                                             
Starting >>> ament_cmake_core
Finished <<< ament_flake8 [0.62s]                                                       
Starting >>> ament_pep257
Finished <<< ament_cmake_core [1.02s]                                                     
Starting >>> ament_cmake_libraries
Finished <<< ament_pep257 [0.62s]                                                            
Starting >>> ament_copyright
Finished <<< ament_cmake_libraries [0.62s]                                                        
Starting >>> ament_cmake_export_libraries
Finished <<< ament_copyright [0.63s]                                                                   
Starting >>> ament_lint_cmake
Finished <<< ament_cmake_export_libraries [0.64s]                                                       
Starting >>> ament_cmake_include_directories
Finished <<< ament_lint_cmake [0.63s]                                                                      
Starting >>> ament_cmake_python
Finished <<< ament_cmake_include_directories [0.53s]                                                                 
Starting >>> ament_cmake_export_dependencies
Finished <<< ament_cmake_python [0.53s]                                                                               
Starting >>> ament_cmake_test
Finished <<< ament_cmake_export_dependencies [0.55s]                                                                
Starting >>> ament_cmake_export_definitions
Finished <<< ament_cmake_test [0.61s]                                                                              
Starting >>> ament_cmake_lint_cmake
Finished <<< ament_cmake_export_definitions [0.52s]                                                                    
Starting >>> ament_cmake_export_include_directories
Finished <<< ament_cmake_lint_cmake [0.51s]                                                                                      
[... 1068 lines cut ...]
 * [new branch]            v1.12-branch                   -> eprosima/v1.12-branch
 * [new branch]            v1.13-branch                   -> eprosima/v1.13-branch
 * [new branch]            v1.14-branch                   -> eprosima/v1.14-branch
 * [new branch]            v1.5-branch                    -> eprosima/v1.5-branch
 * [new branch]            v1.6-branch                    -> eprosima/v1.6-branch
 * [new branch]            v1.7-branch                    -> eprosima/v1.7-branch
 * [new branch]            v1.8-branch                    -> eprosima/v1.8-branch
 * [new branch]            v1.9-branch                    -> eprosima/v1.9-branch
 * [new branch]            v2.0-branch                    -> eprosima/v2.0-branch
 * [new branch]            v2.1-branch                    -> eprosima/v2.1-branch
[... cut ...]
You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 38b77be6c8 Working sensor
#All required rosdeps installed successfully
Segmentation fault (core dumped)
```

gramss, 2020-05-12:
This is the head of the build process:

```
ros2 run micro_ros_setup build_firmware.sh
Crosscompiled environment: cleaning path
Building firmware for zephyr platform nucleo_f401re
Selected app: int32_serial_publisher
-- west build: build configuration:
       source directory: /home/flo/uros_ws_new/firmware/zephyr_apps/apps/int32_serial_publisher
       build directory: /home/flo/uros_ws_new/firmware/build (created)
       BOARD: nucleo_f401re (origin: command line)
-- west build: generating a build system
-- Zephyr version: 2.2.0
-- Found PythonInterp: /usr/bin/python3.6 (found suitable version "3.6.9", minimum required is "3.6") 
-- git describe stderr: 
-- Selected BOARD nucleo_f401re
-- Found west: /home/flo/.local/bin/west (found suitable version "0.7.2", minimum required is "0.6.0")
-- Loading /home/flo/uros_ws_new/firmware/zephyrproject/zephyr/boards/arm/nucleo_f401re/nucleo_f401re.dts as base
Devicetree header saved to '/home/flo/uros_ws_new/firmware/build/zephyr/include/generated/devicetree_unfixed.h'

warning: NATIVE_POSIX_TIMER (defined at drivers/timer/Kconfig:180) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: BOARD_NATIVE_POSIX (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_NATIVE_POSIX_TIMER.html and/or look up
NATIVE_POSIX_TIMER in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: PRINTK (defined at subsys/debug/Kconfig:108) was assigned the value 'n' but got the value
'y'. See http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_PRINTK.html and/or look up
PRINTK in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

Parsing /home/flo/uros_ws_new/firmware/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/flo/uros_ws_new/firmware/zephyrproject/zephyr/boards/arm/nucleo_f401re/nucleo_f401re_defconfig'
Merged configuration '/home/flo/uros_ws_new/firmware/zephyr_apps/apps/int32_serial_publisher/prj.conf'
Configuration saved to '/home/flo/uros_ws_new/firmware/build/zephyr/.config'
Kconfig header saved to '/home/flo/uros_ws_new/firmware/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/flo/uros_ws_new/firmware/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/flo/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/flo/uros_ws_new/firmware/build
-- west build: building application

```

the `.cache` seems to might cause some problem here...?
Inside the firmware/zephyr-sdk is a info file with 0.11 and definitly not 0.22 ... hm..

gramss, 2020-05-12:
> Another question, how are you using Zephyr RTOS 0.2.2? I

output is and during the build process is 2.2.0 !
The arm-sdk tools are taken form your branch. But the Zypherproject is cloned from Zypher directly (latest)
```
*** Booting Zephyr OS build v2.2.0-rc1-115-g38b77be6c8ba  ***
~~~~~~~~~~~~~~~~~~~~Failed status on line 26: 1. Aborting.
Failed status on line 31: 101. Aborting.
Failed status on line 35: 200. Aborting.
```

pablogs9, 2020-05-12:
Yes but in the create script it should checkout out branch: `Note: checking out 'remotes/eprosima/feature/micro_ros'.` in https://github.com/micro-ROS/micro-ros-build/issues/115#issuecomment-627480491

gramss, 2020-05-12:
how can I fix this..? zephyr-sdk in firmware is not a git when I type in git status

pablogs9, 2020-05-12:
Check your environment variables or .zephyrrc because the SDK version is defined here: https://github.com/micro-ROS/micro-ros-build/blob/1dd94f43cbddd332ce9bcb609078f9b70eedce22/micro_ros_setup/config/zephyr/generic/create.sh#L28

gramss, 2020-05-12:
echo $ZEPHYR_TOOLCHAIN_VARIANT 
zephyr
flo@ubuntu:~/uros_ws_new$ echo $ZEPHYR_SDK_INSTALL_DIR 
/home/flo/uros_ws_new/firmware/zephyr-sdk

they should be okay like this. the setup.run installes the 0.11.2 at the given installation folders like /firmware/zephyr-sdk

pablogs9, 2020-05-12:
which problems do you have with rosdep?

gramss, 2020-05-12:
> ``` 
> flo@ubuntu:~/uros_ws_new/firmware$ rosdep install -y --from-paths mcu_ws -i mcu_ws --rosdistro dashing --skip-keys="$SKIP"
> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> rmw_microxrcedds: Cannot locate rosdep definition for [microxrcedds_client]
> rosidl_typesupport_microxrcedds_c: Cannot locate rosdep definition for [microcdr]
> ```

but I don't know if they are problems...

I figured out the PRINTK stuff. When the image is compiled the config is merged with the defconf for this board. And here the console is activated which depends on PRINTK. Therefore, PRINTK was re-enabled:

https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nucleo_f401re/nucleo_f401re_defconfig

Also `CONFIG_UART_CONSOLE_ON_DEV_NAME` works now:
`CONFIG_UART_CONSOLE_ON_DEV_NAME="UART_1"`

But besides that UART_2 is dead.. Neither the agent nor minicom shows any life on the device..
Do you have any experience with `west debug` ? I have a debugger attached to my board but I don't know how to properly use it in this build env..

pablogs9, 2020-05-12:
I have no experience with `west debug`... let me check the transport tomorrow to see if can catch something

gramss, 2020-05-12:
I did a little rudimentary debugging with printk()..
So the number in the main is printed just fine.

I also added here a "flag":
```
    platform->uart_dev = device_get_binding("UART_2");
    if (!platform->uart_dev) {
        printk("Serial device not found\n"); // this doesn't get printed..
        return false;
    }
    printk("UART_2 should be init /n"); // <<<<<<-- this gets printed before the ~~~~~~ 
```
[Link to Line 62 in code](https://github.com/micro-ROS/zephyr_apps/blob/f420820774fbda0bf51cba53c37fb0eebfea2308/microros_extensions/zephyr_serial_transport.c#L62 )
I think I'm at the end of my debugging capabilities here.. Can you implement some test outputs on your uart_dev..? I'm at a point where I would need a peak in your implementation to go further..

Without Console and PrintK I'm currently not able to test if your code has the UART_2 device.. There comes no new bits over a larger time and your agent also gives me no new information..

Do I need to rebuild the agent in the new env..? I kept the old ws just for the agent..

pablogs9, 2020-05-13:
Hi again @gramss, please check my new commit here: https://github.com/micro-ROS/zephyr_apps/tree/feature/add_nucleo_f401re_support

I have changed the serial transport to a polling approach. Well, I have test it in the ST Disco (same bridged UART over USB approach than your board) and it works. Please test it a tell me your results.

gramss, 2020-05-13:
Hey @pablogs9, I just tested it and it worked with the updated branch! :trophy: 
Disabling `PRINTK=n` worked now as well with your disable banner config!

I'll close this issue now and add the PR about flashing the nucle_f401re board with `west flash` like the other ST board.
Thank you for your effort! :smiley:

closed 2020-05-13 by gramss

pablogs9, 2020-05-13:
One more thing @gramss, can you explain to us where are you using micro-ROS, can you give some details about your use case? You told it has educational purposes right?

gramss, 2020-05-17:
Hey @pablogs9 ,
sorry for the late response. Of course, I can give you some detail about my desired application.
I'm currently a student in the laboratory of mobile robotics at the Nuremberg Institute of Technology. There is a training for 80-90 people happening yearly. One task of this is working with the Nucleo STM32F401 and the ToF sensor extension board VL6180X. The students need to implement an own serial protocol and agent that republishes the data to the ROS1 melodic system.
I wanted to take the drive of this to update this part of the training to ROS2 with micro-ros.
My prof designed more trainings about localization, navigation and sensor manipulation with ROS and Octave. We have already like 50 boards with those two components. So I thought it would be worth a try to integrate them with micro-ros. I always wanted to learn more about zephyr as I happen to have some few more boards waiting for this RTOS.
I also happen to possess the ST IoT Discovery board.

My long-term goals for an application/exercise for this training are containing these steps:
* analyse micro-ros stability / performance / speed over UART
** change baud-rate to the max of this board / your communication stack
* check ease of use ( pub / sub ) of micro-ros
* (optional) implement/port VL6180X library to zephry and integrate it in the zephyrproject
* (optional) build a sample app in your zephyr_apps repo

Optional regarding my time. The rest would be the minimum to prepare the basics for additional future trainings with micro-ros and ROS2