micro-ROS/micro_ros_setup #116: Need help for porting to ESP32

MS71, 2020-05-10:
Hi,
i like to use micro ROS on my ESP32 board (esp32cam).
For that, I made temporary fork on https://github.com/MS71/micro-ros-build.git.

By the way: esp32 is using FreeRTOS by default. So porting should be easy.

After executing:
ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32cam
...  I get a folder with next structure.

My idea is to use this generated micro ROS folder as an ESP IDF component directory. Now I have to create ESP IDF build support files.

My Questions:
* Should this work?
* I need to compile everything inside for ESP32?

components/uros/
├── CMakeLists.txt
├── component.mk
├── colcon.meta
├── eProsima
│   ├── Micro-CDR
│   └── Micro-XRCE-DDS-Client
├── Kconfig
├── ros2
│   ├── common_interfaces
│   ├── example_interfaces
│   ├── libyaml_vendor
│   ├── rcl
│   ├── rcl_interfaces
│   ├── rcl_logging
│   ├── rmw
│   ├── rmw_implementation
│   ├── rosidl
│   ├── rosidl_dds
│   ├── rosidl_defaults
│   ├── rosidl_typesupport
│   ├── test_interface_files
│   ├── tinydir_vendor
│   └── unique_identifier_msgs
├── ros2.repos
└── uros
    ├── rcl
    ├── rcutils
    ├── rmw_microxrcedds
    └── rosidl_typesupport_microxrcedds

pablogs9, 2020-05-11:
Folder generated by the micro-ROS build system use colcon in order to compile. I think that it would be easier to integrate IDF inside the micro-ROS build system, check the way we are integrating Zephyr RTOS inside the build system for an example. Maybe it is not the cleanest option, but it is the preferred approach for this version of the build system.

By using the `create_firmware.sh` of your platform you should download all the required dependencies (such as the IDF).
In the `configure.sh` step you should execute colcon (with a toolchain file in order to provide a crosscopiler) to obtain a static library (usually microros.a) ready to be linked with your ESP32 code.

I would start preparing the environment in order to provide a toolchain with the crosscompilation options colcon.

ralph-lange, 2020-07-08:
@MS71, we recently brought the micro-ROS stack on FreeRTOS on a ESP32 Dev Kit C. (We have also tried it with Zephyr, but failed at https://github.com/zephyrproject-rtos/zephyr/issues/1495.) We will extend the build files in the next days to support the ESP32. Also, we will try out the ESP32-CAM board that you mentioned. I'll keep you updated via this issue.

MS71, 2020-07-08:
Hi, this sounds good.

I think, you could use any of the ESP32 modules.
The ESP32 CAM could be interesting because the 4 MByte external Flash and RAM and the embedded camera. You need a 3.3V FDTI Adapter for programming.

For development, a ESP32 module with embedded FTDI and USB interface could be useful.

Meanwhile, I got ROS2 running on my ESP32 with a similar approach.
I did some modification on ros2arduino. Now this is use able for ESP-IDF tool.
https://github.com/MS71/ROS2Mower/tree/master/esp32/ros2node/components/ros2

In my opinion, micro-ROS should should generate a  esp-idf component.
For the target project, this ROS2 component is compile by the ESP IDF environment.

Please let me, know if you need some help.

Maik

ralph-lange, 2020-07-10:
Draft PR now available at https://github.com/micro-ROS/micro-ros-build/pull/140.

FranFin, 2020-07-20:
@MS71 Hi! Thank you for your interest in micro-ROS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460). You’re invited to bring your micro-ROS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.

donRaphaco, 2020-07-22:
Hi, I created the PR #140 and https://github.com/micro-ROS/freertos_apps/pull/12 which adds support for ESP32.
Serial transport works fine, unfortunately I could not get it to work with UDP or TCP transport.
I tried it with WiFi, as I don't have Ethernet available on my devBoard and I get a runtime error after the WiFi is connected and the microros init functions are executed. I have no idea what exactly the problem is there, so help would be appreciated.

pablogs9, 2020-07-22:
Can we close this issue and move the discussion to the pull requests?

ralph-lange, 2020-07-23:
> Can we close this issue and move the discussion to the pull requests?
I'm fine to move the discussion to the PR. Are you also fine with that, @MS71 ?

pablogs9, 2020-08-04:
Hello, ESP32 support is ready both for serial and Wifi/UDP transports. You can check this https://github.com/micro-ROS/micro_ros_setup/pull/140 and you can find instructions in the main micro_ros_setup repository: https://github.com/micro-ROS/micro_ros_setup.

Please give us feedback if you try this port!
Thanks!

closed 2020-08-04 by pablogs9