micro-ROS/micro_ros_setup #118: Weird environment setup issue with clang-tidy

rpcme, 2020-05-16:
I followed the instruction to install ros2 `dashing`.

I am following the instruction here

https://micro-ros.github.io/docs/tutorials/core/first_application_linux/

And am at the very beginning, just trying to get the environment sorted.

```bash
rpcme@proteus:~/microros_ws$ rosdep install --from-path src --ignore-src -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
micro_ros_setup: No definition of [clang-tidy] for OS version [bionic]
rpcme@proteus:~/microros_ws$ uname -a
Linux proteus 5.3.0-51-generic #44~18.04.2-Ubuntu SMP Thu Apr 23 14:27:18 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
rpcme@proteus:~/microros_ws$ dpkg-query -l clang-tidy
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  clang-tidy     1:6.0-41~exp amd64        clang-based C++ linter tool

```

Any ideas?

pablogs9, 2020-05-18:
This system targets Ubuntu. Anyways, rosdep is no able to resolve this dependency but you have it installed. Try keeping with the procedure ignoring this rosdep warning.

rpcme, 2020-05-18:
I'm running Ubuntu 18.04, but old habits die hard when using the command line since I have been running Linux well before Ubuntu existed.

Anyways, I went through the ROS2 installation procedure again and it seems I missed some package install step there.  The command now completes successfully.

closed 2020-05-18 by rpcme