micro-ROS/micro_ros_setup #120: Error in Running the micro-ROS app

kashishdhal, 2020-05-19:
Hello,

I started installing the micro-ROS. I was on first application linux page. In that page I was particularly on Running the micro-ROS app. I got the following error. I don't know if it is normal. I thought of sharing this. I am pretty new here so don't know much.

Regards,
Kashish Dhal

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup build_firmware.sh
Compiling for host environment: not cleaning path
Building firmware for host platform generic
Starting &gt;&gt;&gt; rosidl_adapter
Starting &gt;&gt;&gt; rosidl_typesupport_interface
Starting &gt;&gt;&gt; microcdr
Starting &gt;&gt;&gt; rmw_implementation_cmake
Starting &gt;&gt;&gt; rcl_logging_log4cxx
Starting &gt;&gt;&gt; rcl_logging_noop                                 
Starting &gt;&gt;&gt; libyaml_vendor
Starting &gt;&gt;&gt; microxrcedds_agent
Finished &lt;&lt;&lt; rmw_implementation_cmake [3.76s]                       
Starting &gt;&gt;&gt; test_interface_files
Finished &lt;&lt;&lt; rosidl_typesupport_interface [3.89s]                   
Starting &gt;&gt;&gt; micro_ros_setup
Finished &lt;&lt;&lt; micro_ros_setup [0.85s]                                
Finished &lt;&lt;&lt; microcdr [5.97s]                                       
Starting &gt;&gt;&gt; microxrcedds_client
Finished &lt;&lt;&lt; rcl_logging_noop [6.33s]                                 
Finished &lt;&lt;&lt; test_interface_files [3.03s]                             
Finished &lt;&lt;&lt; rosidl_adapter [6.91s]
Starting &gt;&gt;&gt; rosidl_parser
Finished &lt;&lt;&lt; rcl_logging_log4cxx [7.03s]                                   
Finished &lt;&lt;&lt; libyaml_vendor [10.0s]                                          
Finished &lt;&lt;&lt; microxrcedds_client [4.40s]                                      
Finished &lt;&lt;&lt; rosidl_parser [4.97s]                                             
Starting &gt;&gt;&gt; rosidl_cmake
Finished &lt;&lt;&lt; rosidl_cmake [3.77s]                                              
Starting &gt;&gt;&gt; rosidl_generator_c
Starting &gt;&gt;&gt; rosidl_generator_dds_idl
Finished &lt;&lt;&lt; rosidl_generator_dds_idl [3.68s]                                  
Finished &lt;&lt;&lt; rosidl_generator_c [10.6s]                                        
Starting &gt;&gt;&gt; rosidl_generator_cpp
Starting &gt;&gt;&gt; rosidl_typesupport_microxrcedds_c
Starting &gt;&gt;&gt; rmw
Starting &gt;&gt;&gt; rosidl_typesupport_introspection_c                                
Finished &lt;&lt;&lt; rosidl_typesupport_microxrcedds_c [5.03s]                         
Finished &lt;&lt;&lt; rosidl_typesupport_introspection_c [6.14s]                        
Finished &lt;&lt;&lt; rmw [11.4s]                                                       
Finished &lt;&lt;&lt; rosidl_generator_cpp [17.5s]                                      
Starting &gt;&gt;&gt; rosidl_typesupport_microxrcedds_cpp
Starting &gt;&gt;&gt; rosidl_typesupport_introspection_cpp
Finished &lt;&lt;&lt; rosidl_typesupport_microxrcedds_cpp [11.2s]                       
Starting &gt;&gt;&gt; rmw_microxrcedds
Finished &lt;&lt;&lt; rosidl_typesupport_introspection_cpp [14.4s]                      
--- stderr: rmw_microxrcedds                                                   
/home/gnclab/microros_ws/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_count.c:15:1: warning: #includes are not sorted properly [llvm-include-order]
#include &lt;rmw/rmw.h&gt;
^        ~~~~~~~~~~~
         &lt;rmw/error_handling.h&gt;
/home/gnclab/microros_ws/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_count.c:22:12: warning: pointer parameter &apos;count&apos; can be pointer to const [readability-non-const-parameter]
  size_t * count)
  ~~~~~~   ^
  const 
/home/gnclab/microros_ws/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_count.c:35:12: warning: pointer parameter &apos;count&apos; can be pointer to const [readability-non-const-parameter]
  size_t * count)
  ~~~~~~   ^
  const 
/home/gnclab/microros_ws/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_compare_gids_equal.c:15:1: warning: #includes are not sorted properly [llvm-include-order]
#include &lt;rmw/rmw.h&gt;
^        ~~~~~~~~~~~
         &lt;rmw/error_handling.h&gt;
/home/gnclab/microros_ws/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_compare_gids_equal.c:30:5: warning: do not use &apos;else&apos; after &apos;return&apos; [readability-else-after-return]
[... 1033 lines cut ...]
     <font color="#AD7FA8"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/gnclab/microros_ws/src/uros/micro-ROS-demos/rcl/ping_pong/main.c:107:5:</b> wa<font color="#AD7FA8"><b>rning: </b></font>ignoring return value of ‘<b>rcl_wait</b>’, declared with attribute warn_unused_result [<font color="#AD7FA8"><b>-Wunused-result</b></font>]
     <font color="#AD7FA8"><b>rcl_wait(&amp;wait_set, RCL_MS_TO_NS(100))</b></font>;
     <font color="#AD7FA8"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/gnclab/microros_ws/src/uros/micro-ROS-demos/rcl/ping_pong/main.c:125:9:</b> <font color="#AD7FA8"><b>warning: </b></font>ignoring return value of ‘<b>rcl_publish</b>’, declared with attribute warn_unused_result [<font color="#AD7FA8"><b>-Wunused-result</b></font>]
         <font color="#AD7FA8"><b>rcl_publish(&amp;pong_publisher, (const void*)&amp;rcv_msg, NULL)</b></font>;
         <font color="#AD7FA8"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/gnclab/microros_ws/src/uros/micro-ROS-demos/rcl/ping_pong/main.c:143:7:</b> <font color="#AD7FA8"><b>warning: </b></font>ignoring return value of ‘<b>rcl_publish</b>’, declared with attribute warn_unused_result [<font color="#AD7FA8"><b>-Wunused-result</b></font>]
       <font color="#AD7FA8"><b>rcl_publish(&amp;ping_publisher, (const void*)&amp;msg, NULL)</b></font>;
       <font color="#AD7FA8"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/gnclab/microros_ws/src/uros/micro-ROS-demos/rcl/fibonacci_action_server/main.c:</b> In function ‘<b>main</b>’:
<b>/home/gnclab/microros_ws/src/uros/micro-ROS-demos/rcl/fibonacci_action_server/main.c:87:5:</b> <font color="#AD7FA8"><b>warning: </b></font>ignoring return value of ‘<b>rcl_wait</b>’, declared with attribute warn_unused_result [<font color="#AD7FA8"><b>-Wunused-result</b></font>]
     <font color="#AD7FA8"><b>rcl_wait(&amp;wait_set, RCL_MS_TO_NS(200))</b></font>;
     <font color="#AD7FA8"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
---
Finished &lt;&lt;&lt; micro_ros_demos_rcl [22.8s]
Finished &lt;&lt;&lt; sensor_msgs [2min 13s]                                  
Starting &gt;&gt;&gt; stereo_msgs
Finished &lt;&lt;&lt; stereo_msgs [13.6s]                                
Starting &gt;&gt;&gt; common_interfaces
Finished &lt;&lt;&lt; common_interfaces [1.95s]                          

Summary: 60 packages finished [8min 32s]
  4 packages had stderr output: micro_ros_demos_rcl micro_ros_demos_rclcpp microxrcedds_agent rmw_microxrcedds
</pre>

pablogs9, 2020-05-19:
I can't see any error here, colcon is saying that you have 60 packages built and 4 has some kind of stderr (but they successfully compile). Have you tried to continue with the tutorial?

kashishdhal, 2020-05-19:
Hello @pablogs9,

If you read through the output it will be clear that app.c file is not compiled that I created inside my_brand_new_app folder. That is throwing an error on screen.

Anyways, continuing to tutorials I get error Executable not found.

pablogs9, 2020-05-19:
Have you included your app in the demo repository CMakeList?

In `src/uros/micro-ROS-demos/rcl/CMakeLists.txt`, there should be a line with `export_executable(my_brand_new_app)`

Is this correct?

kashishdhal, 2020-05-19:
Yes, here is my CMakeList:

cmake_minimum_required(VERSION 3.5)

project(my_brand_new_app LANGUAGES C)

find_package(ament_cmake REQUIRED)
find_package(rcl REQUIRED)
find_package(std_msgs REQUIRED)
find_package(rmw_microxrcedds REQUIRED)

add_executable(${PROJECT_NAME} app.c)

export_executable(my_brand_new_app)

ament_target_dependencies(${PROJECT_NAME}
  rcl
  std_msgs
  rmw_microxrcedds
  )

install(TARGETS ${PROJECT_NAME}
  DESTINATION ${PROJECT_NAME}
  )

kashishdhal, 2020-05-19:
If I remove that line then I don't get any error but proceeding still shows no executables found

kashishdhal, 2020-05-19:
So interestingly, when I re-ran, this time it went through. But I am not sure if app.c got compiled as I don't see any output on screen saying compiling etc (see below). Also, it still giving error Executable not found.

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup build_firmware.sh
Compiling for host environment: not cleaning path
Building firmware for host platform generic
Starting &gt;&gt;&gt; rosidl_adapter
Starting &gt;&gt;&gt; rosidl_typesupport_interface
Starting &gt;&gt;&gt; microcdr
Starting &gt;&gt;&gt; rmw_implementation_cmake
Starting &gt;&gt;&gt; rcl_logging_log4cxx                              
Starting &gt;&gt;&gt; rcl_logging_noop
Starting &gt;&gt;&gt; libyaml_vendor
Starting &gt;&gt;&gt; microxrcedds_agent
Finished &lt;&lt;&lt; rosidl_typesupport_interface [0.74s]                     
Starting &gt;&gt;&gt; test_interface_files
Finished &lt;&lt;&lt; rmw_implementation_cmake [0.76s]                         
Starting &gt;&gt;&gt; micro_ros_setup
Finished &lt;&lt;&lt; rcl_logging_log4cxx [0.95s]                              
Finished &lt;&lt;&lt; microcdr [1.01s]
Finished &lt;&lt;&lt; rcl_logging_noop [0.95s]
Starting &gt;&gt;&gt; microxrcedds_client
Finished &lt;&lt;&lt; libyaml_vendor [0.97s]                                   
Finished &lt;&lt;&lt; test_interface_files [0.58s]                             
Finished &lt;&lt;&lt; micro_ros_setup [0.59s]                                  
Finished &lt;&lt;&lt; rosidl_adapter [1.46s]
Starting &gt;&gt;&gt; rosidl_parser
Finished &lt;&lt;&lt; microxrcedds_agent [1.63s]                                        
Starting &gt;&gt;&gt; micro_ros_agent
Finished &lt;&lt;&lt; microxrcedds_client [0.89s]
Finished &lt;&lt;&lt; rosidl_parser [0.81s]                                    
Starting &gt;&gt;&gt; rosidl_cmake
Finished &lt;&lt;&lt; micro_ros_agent [0.50s]                                         
Finished &lt;&lt;&lt; rosidl_cmake [0.71s]                    
Starting &gt;&gt;&gt; rosidl_generator_c
Starting &gt;&gt;&gt; rosidl_generator_dds_idl              
Finished &lt;&lt;&lt; rosidl_generator_dds_idl [0.75s]                                 
Finished &lt;&lt;&lt; rosidl_generator_c [1.51s]                         
Starting &gt;&gt;&gt; rosidl_generator_cpp
Starting &gt;&gt;&gt; rosidl_typesupport_microxrcedds_c       
Starting &gt;&gt;&gt; rmw
Starting &gt;&gt;&gt; rosidl_typesupport_introspection_c
Finished &lt;&lt;&lt; rmw [1.03s]                                                       
Finished &lt;&lt;&lt; rosidl_typesupport_microxrcedds_c [1.28s]                       
Finished &lt;&lt;&lt; rosidl_typesupport_introspection_c [1.35s]
Finished &lt;&lt;&lt; rosidl_generator_cpp [1.82s]                         
Starting &gt;&gt;&gt; rosidl_typesupport_microxrcedds_cpp
Starting &gt;&gt;&gt; rosidl_typesupport_introspection_cpp
Finished &lt;&lt;&lt; rosidl_typesupport_microxrcedds_cpp [0.89s]
Starting &gt;&gt;&gt; rmw_microxrcedds
Finished &lt;&lt;&lt; rosidl_typesupport_introspection_cpp [0.91s]
Finished &lt;&lt;&lt; rmw_microxrcedds [1.03s]                         
Starting &gt;&gt;&gt; rmw_implementation
Finished &lt;&lt;&lt; rmw_implementation [0.57s]                       
Starting &gt;&gt;&gt; rosidl_typesupport_c
Finished &lt;&lt;&lt; rosidl_typesupport_c [0.95s]                          
Starting &gt;&gt;&gt; rosidl_typesupport_cpp
Finished &lt;&lt;&lt; rosidl_typesupport_cpp [0.97s]                          
Starting &gt;&gt;&gt; rosidl_default_runtime
Starting &gt;&gt;&gt; rosidl_default_generators
Finished &lt;&lt;&lt; rosidl_default_runtime [0.50s]                                   
Finished &lt;&lt;&lt; rosidl_default_generators [0.49s]
Starting &gt;&gt;&gt; builtin_interfaces
Starting &gt;&gt;&gt; unique_identifier_msgs
Starting &gt;&gt;&gt; lifecycle_msgs
Starting &gt;&gt;&gt; std_srvs                                               
Finished &lt;&lt;&lt; unique_identifier_msgs [3.13s]                                    
Finished &lt;&lt;&lt; builtin_interfaces [3.17s]
Starting &gt;&gt;&gt; std_msgs
Starting &gt;&gt;&gt; rcl_interfaces                                                 
Starting &gt;&gt;&gt; action_msgs
[... 33 lines cut ...]
Finished &lt;&lt;&lt; rcl_yaml_param_parser [1.52s]                                  
Finished &lt;&lt;&lt; rcl_lifecycle [1.56s]                                          
Finished &lt;&lt;&lt; geometry_msgs [9.37s]                                          
Starting &gt;&gt;&gt; sensor_msgs
Starting &gt;&gt;&gt; diagnostic_msgs                                            
Starting &gt;&gt;&gt; nav_msgs
Starting &gt;&gt;&gt; shape_msgs                                                  
Starting &gt;&gt;&gt; trajectory_msgs
Starting &gt;&gt;&gt; visualization_msgs                                          
Finished &lt;&lt;&lt; rcl_action [2.26s]
Starting &gt;&gt;&gt; micro_ros_demos_rcl
Finished &lt;&lt;&lt; trajectory_msgs [7.36s]                                    
Finished &lt;&lt;&lt; shape_msgs [7.55s]                                         
Finished &lt;&lt;&lt; diagnostic_msgs [7.92s]                                    
Finished &lt;&lt;&lt; micro_ros_demos_rcl [8.56s]                                
Finished &lt;&lt;&lt; nav_msgs [8.92s]                                           
Finished &lt;&lt;&lt; visualization_msgs [9.12s]                                 
Finished &lt;&lt;&lt; sensor_msgs [11.1s]                           
Starting &gt;&gt;&gt; stereo_msgs
Finished &lt;&lt;&lt; stereo_msgs [1.96s]                          
Starting &gt;&gt;&gt; common_interfaces
Finished &lt;&lt;&lt; common_interfaces [0.62s]                     

Summary: 60 packages finished [47.4s]
</pre>

pablogs9, 2020-05-20:
Hello @kashishdhal I just have tried the tutorial step by step in a ROS 2 dashing docker container without any problems. I have skipped agent installation because I have an agent already running.

You have the session here:
https://asciinema.org/a/IQ1w29fEtOPTxnwfjVCtamH8P

Please check that your installation is not interfering with the tutorial.

kashishdhal, 2020-05-21:
Hello @pablogs9 ,

Firstly, I am sorry that I thought **stderr** means that app.c is not compiled. I see that you got the same error. So I checked the video that you sent. Almost, everything matches with my console outputs. One thing that I noticed is that you did everything as a super User (root or sudo -s mode) that should not be necessary, correct?

Secondly, you skipped agent installation so just to make that's correct I am pasting the output:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup create_agent_ws.sh
Repo-file ros2.repos already present, overwriting!
/home/gnclab/microros_ws/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:10: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_info = yaml.load(sys.stdin)
/home/gnclab/microros_ws/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:11: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_keep = yaml.load(open(sys.argv[1]))[&apos;keep&apos;].split()
#All required rosdeps installed successfully</pre>

Thirdly, I noticed that your CMakeList.txt looked a bit different, I see that you are also exporting the other packages in export_executable.

So, now assuming everything went well. Can you guess why am I getting the error? See below:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ source /opt/ros/dashing/setup.bash
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ source install/local_setup.bash
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ echo $ROS_DISTRO
dashing
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ ros2 run micro_ros_demos_rcl my_brand_new_app
No executable found
</pre>

So when I go the build directory, inside the micro_ros_demo_rcl, I don't see a folder named my_brand_new_app (see below). Does that mean it's not compiled or should it be located somewhere else?

[image]

Regards,
Kashish Dhal

pablogs9, 2020-05-21:
Where are you pasting the `export_executable(my_brand_new_app)` exactly? Please let me know the path of this CMakeLists.txt

kashishdhal, 2020-05-21:
This is my CMakeList.txt if you are asking that?

cmake_minimum_required(VERSION 3.5)

project(my_brand_new_app LANGUAGES C)

find_package(ament_cmake REQUIRED)
find_package(rcl REQUIRED)
find_package(std_msgs REQUIRED)
find_package(rmw_microxrcedds REQUIRED)

add_executable(${PROJECT_NAME} app.c)

export_executable(my_brand_new_app)

ament_target_dependencies(${PROJECT_NAME}
rcl
std_msgs
rmw_microxrcedds
)

install(TARGETS ${PROJECT_NAME}
DESTINATION ${PROJECT_NAME}
)

kashishdhal, 2020-05-21:
Path of CMakeList is as given in tutorial

pablogs9, 2020-05-21:
> Have you included your app in the demo repository CMakeList?
> 
> In `src/uros/micro-ROS-demos/rcl/CMakeLists.txt`, there should be a line with `export_executable(my_brand_new_app)`
> 
> Is this correct?

Please recheck the CMakeLists.txt path of this comment, you are putting  `export_executable` inside the app CMakeLists, but it need to be in the CMakeLists of the `micro-ROS-demos/rcl` folder.

kashishdhal, 2020-05-21:
Okay, got it, thanks. I will correct it tomorrow and let you know.

pablogs9, 2020-05-21:
Close the issue when this is working please.

kashishdhal, 2020-05-21:
Yes I will

kashishdhal, 2020-05-21:
It's pretty late here so I will do it tomorrow

kashishdhal, 2020-05-21:
Hello @pablogs9 ,

First of all thank you very much for the help!
It's working now but I would like to make a small suggestion. Since that CMakeList part is a little confusing. May be write the command lines here so that one is not confused between two CMakeList files.

[image]

Instead may be just write these commands there:
`cd src/uros/micro-ROS-demos/rcl`
`nano CMakeLists.txt`

Add the following line:
`export_executable(my_brand_new_app)`

Then:
`cd ~/microros_ws`

closed 2020-05-21 by kashishdhal