micro-ROS/micro_ros_setup #121: Error in

kashishdhal, 2020-05-19:
Hello, 

I was going through the tutorials as is given in, **First micro-ROS Application on an RTOS**. However, I got a few errors. I am pasting the whole procedure started from the scratch. Please have a look and if possible, please tell me what am I doing wrong.

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ source /opt/ros/dashing/setup.bash
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ echo $ROS_DISTRO
melodic
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ source ros2_dashing/install/setup.bash 
ROS_DISTRO was set to &apos;melodic&apos; before. Please make sure that the environment does not mix paths from different distributions.
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ echo $ROS_DISTRO
dashing
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ mkdir uros_ws &amp;&amp; cd uros_ws
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_ws</b></font>$ git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro-ros-build.git src/micro-ros-build
Cloning into &apos;src/micro-ros-build&apos;...
remote: Enumerating objects: 40, done.
remote: Counting objects: 100% (40/40), done.
remote: Compressing objects: 100% (31/31), done.
remote: Total 1717 (delta 16), reused 16 (delta 9), pack-reused 1677
Receiving objects: 100% (1717/1717), 364.50 KiB | 1.62 MiB/s, done.
Resolving deltas: 100% (1168/1168), done.
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_ws</b></font>$ sudo apt update &amp;&amp; rosdep update
[sudo] password for gnclab: 
Get:1 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Get:2 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic InRelease [4,256 B]
Hit:3 http://archive.ubuntu.com/ubuntu bionic InRelease                        
Hit:4 http://ppa.launchpad.net/system76-dev/stable/ubuntu bionic InRelease     
Hit:5 http://packages.ros.org/ros/ubuntu bionic InRelease                      
Get:6 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic/main amd64 Packages [63.8 kB]
Get:7 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]      
Hit:8 http://packages.ros.org/ros2/ubuntu bionic InRelease                     
Hit:9 http://linux.teamviewer.com/deb stable InRelease                         
Get:10 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [717 kB]
Get:11 http://security.ubuntu.com/ubuntu bionic-security/main Translation-en [227 kB]
Get:12 http://security.ubuntu.com/ubuntu bionic-security/main amd64 DEP-11 Metadata [40.6 kB]
Get:13 http://security.ubuntu.com/ubuntu bionic-security/restricted amd64 Packages [41.9 kB]
Get:14 http://security.ubuntu.com/ubuntu bionic-security/restricted Translation-en [10.5 kB]
Get:15 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 Packages [665 kB]
Get:16 http://security.ubuntu.com/ubuntu bionic-security/universe Translation-en [221 kB]
Get:17 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [941 kB]
Get:18 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 DEP-11 Metadata [42.0 kB]
Get:19 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 Packages [7,596 B]
Get:20 http://security.ubuntu.com/ubuntu bionic-security/multiverse Translation-en [2,824 B]
Get:21 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 DEP-11 Metadata [2,464 B]
Get:22 http://archive.ubuntu.com/ubuntu bionic-updates/main Translation-en [321 kB]
Get:23 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 DEP-11 Metadata [303 kB]
Get:24 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages [1,073 kB]
Get:25 http://archive.ubuntu.com/ubuntu bionic-updates/universe Translation-en [333 kB]
Get:26 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 DEP-11 Metadata [273 kB]
Get:27 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 DEP-11 Metadata [2,468 B]
Fetched 5,469 kB in 2s (2,903 kB/s)<font color="#C4A000">               </font>
Reading package lists... Done
Building dependency tree       
Reading state information... Done
441 packages can be upgraded. Run &apos;apt list --upgradable&apos; to see them.
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro &quot;ardent&quot;
Skip end-of-life distro &quot;bouncy&quot;
Skip end-of-life distro &quot;crystal&quot;
Add distro &quot;dashing&quot;
Add distro &quot;eloquent&quot;
Add distro &quot;foxy&quot;
Skip end-of-life distro &quot;groovy&quot;
Skip end-of-life distro &quot;hydro&quot;
Skip end-of-life distro &quot;indigo&quot;
[... 1491 lines cut ...]
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_ws</b></font>$ cd uros_ws/firmware/NuttX
bash: cd: uros_ws/firmware/NuttX: No such file or directory
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_ws</b></font>$ cd /firmware/NuttX
bash: cd: /firmware/NuttX: No such file or directory
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_ws</b></font>$ cd firmware/NuttX
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_ws/firmware/NuttX</b></font>$ make menuconfig
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/gnclab/uros_ws/firmware/NuttX/configs/olimex-stm32-e407/include
LN: include/arch/chip to arch/arm/include/stm32
LN: arch/arm/src/board to /home/gnclab/uros_ws/firmware/NuttX/configs/olimex-stm32-e407/src
LN: arch/arm/src/chip to arch/arm/src/stm32
make[1]: Entering directory &apos;/home/gnclab/uros_ws/firmware/NuttX/configs&apos;
make[1]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/NuttX/configs&apos;
make[1]: Entering directory &apos;/home/gnclab/uros_ws/firmware/apps&apos;
make[2]: Entering directory &apos;/home/gnclab/uros_ws/firmware/apps/platform&apos;
LN: platform/board to /home/gnclab/uros_ws/firmware/apps/platform/dummy
make[2]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/apps/platform&apos;
make[1]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/apps&apos;
make[1]: Entering directory &apos;/home/gnclab/uros_ws/firmware/apps&apos;
make[1]: Nothing to be done for &apos;preconfig&apos;.
make[1]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/apps&apos;
/bin/sh: 1: kconfig-mconf: not found
tools/Makefile.unix:510: recipe for target &apos;do_menuconfig&apos; failed
make: *** [do_menuconfig] Error 127
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_ws/firmware/NuttX</b></font>$ </pre>

pablogs9, 2020-05-20:
Please use a clean dashing installation, it seems that your melodic distro is messing up with micro-ROS...

We have updated the tutorials, please check the new ones: https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/

kashishdhal, 2020-05-21:
So do I need to uninstall the ROS dashing and install it back again? Then re-try this tutorial?

pablogs9, 2020-05-21:
I recommend you using a docker image of ROS2 dashing for this testing purpose if you have multiple local installations of ROS 2. Honestly, I don't know how this build system responds when you have multiple installations of ROS 2.

gramss, 2020-05-21:
It works fine with ROS2 dashing and eloquent installed simultaneously.
And by installed I mean physically present in /opt/ros/ .
But make sure that you `$ source` only **one** `/opt/ros/<distro>/setup.bash` at a time (in your ~/.bashrc or manually) 🙂 

It comes to a messed up setup when you mix source them.. But you can get rid of it by simply opening a new shell where you did not source any of those. Also be careful with setup.bash scripts from workingspaces build with other ROS distros.

kashishdhal, 2020-05-22:
Hello @pablogs9 ,

I do not have multiple ROS2. Melodic is ROS1 and Dashing is ROS2. I have never worked with docker before but I will try to remove `/opt/ros/<distro>/setup.bash` from ~/.bashrc as @gramss suggested.

If that still doesn't work then I will probably revisit the docker what you suggested.

Regards,
Kashish Dhal

kashishdhal, 2020-05-22:
Hello @pablogs9 ,

I see that you updated the tutorials. But you removed the command `mkdir uros_ws && cd uros_ws.` I believe that you can not run `ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407` in microros_ws. You will need a different folder for that, correct? Otherwise, it will complain that firmware folder already exists. Also, one would not want to mess-up with the local linux apps.

But anyways, coming back to the discussion, I removed `/opt/ros/melodic/setup.bash` from ~/.bashrc as suggested by @gramss but when you run `ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407` something is happening while the script is executed which changes the distribution again to melodic see below:

kashishdhal, 2020-05-22:
Hello @pablogs9 ,

I spent an hour to learn how docker works but I guess there is quite a bit of learning curve there. I don't think I will be able to continue with docker because of other commitments. I will install a new Ubuntu OS in raspberry pi and install dashing then micro-ros then continue from there.

But I guess a lot of people have already installed different versions of ROS in their computer. I think that it should support that feature. You may want to look into your package to see if it can be made compatible for the same. 

Also, I still wonder that how is it working for @gramss ?

gramss, 2020-05-22:
Have you opened a new terminal after removing all `source /opt/ROS/<distro>/setup.bash` from your `~/.bashrc` ? Maybe comment out all other env variables you typically set for melodic as well.

The error flag you are pointing out says that the script wants to use rosdistro=dashing but your env variable `ROSDISTRO` is already set with melodic before you run the script..

Make sure you do not source melodic anywhere else in your system.
You can check this by opening a new terminal and enter: `echo $ROS_DISTRO`. If it prints melodic or dashing you somehow failed and there is still an automatic sourcing of the setup happening.

Regarding where you run the commands: I need to run all of them in the Root of my top-level workspace.
microros-ws <— here
|—-src (in here this repo gets cloned)
|—-firmware (this folder get created by the script you mentioned)

If the firmware folder exists already, make sure to remove it. Otherwise you‘ll get an error.

kashishdhal, 2020-05-22:
Hello @gramss ,

I restarted the computer after modifying the ~/.bashrc and /etc/bash.bashrc (I could not think of any other place). Also, before I ran the script I checked that distro is set to dashing. 

So here is the output of env:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ env | grep melodic
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ echo $ROS_DISTRO

<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ env | grep ros
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ </pre>

I did not share the whole output before but since the command window is open, let me paste it here.

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ source /opt/ros/dashing/setup.bash 
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ echo $ROS_DISTRO

<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ source ~/ros2_dashing/install/setup.bash 
ROS_DISTRO was set to &apos;melodic&apos; before. Please make sure that the environment does not mix paths from different distributions.
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ echo $ROS_DISTRO
dashing
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ source install/local_setup.bash
bash: install/local_setup.bash: No such file or directory
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ cd microros_ws/
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ source install/local_setup.bash
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407
Firmware already created. Please delete /home/gnclab/microros_ws/firmware folder if you want a fresh installation.
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407
Creating firmware for nuttx platform olimex-stm32-e407
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro &quot;ardent&quot;
Skip end-of-life distro &quot;bouncy&quot;
Skip end-of-life distro &quot;crystal&quot;
Add distro &quot;dashing&quot;
Add distro &quot;eloquent&quot;
Add distro &quot;foxy&quot;
Skip end-of-life distro &quot;groovy&quot;
Skip end-of-life distro &quot;hydro&quot;
Skip end-of-life distro &quot;indigo&quot;
Skip end-of-life distro &quot;jade&quot;
Add distro &quot;kinetic&quot;
Skip end-of-life distro &quot;lunar&quot;
Add distro &quot;melodic&quot;
Add distro &quot;noetic&quot;
updated cache in /home/gnclab/.ros/rosdep/sources.cache
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
micro_ros_agent: Cannot locate rosdep definition for [microxrcedds_agent]
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup configure_firmware.sh uros_pingpong
Configuring firmware for nuttx platform olimex-stm32-e407
Configuration configs/olimex-stm32-e407/uros_pingpong (expanded from uros_pingpong) not found
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ cd firmware/apps/examples
bash: cd: firmware/apps/examples: No such file or directory
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ ls
 <font color="#729FCF"><b>build</b></font>   <font color="#729FCF"><b>firmware</b></font>  <font color="#729FCF"><b>&apos;firmware (copy)&apos;</b></font>   <font color="#729FCF"><b>install</b></font>   <font color="#729FCF"><b>log</b></font>   <font color="#729FCF"><b>src</b></font>
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ cd ..
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ ls | grep uros
<font color="#EF2929"><b>uros</b></font>_ws
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ cd uros_ws/
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_ws</b></font>$ ls
<font color="#729FCF"><b>build</b></font>  <font color="#729FCF"><b>firmware</b></font>  <font color="#729FCF"><b>install</b></font>  <font color="#729FCF"><b>log</b></font>  <font color="#729FCF"><b>src</b></font>
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_ws</b></font>$ ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407
[... cut ...] os2_dashing/install/gmock_vendor/lib/pkgconfig:/home/gnclab/ros2_dashing/install/gtest_vendor/lib/x86_64-linux-gnu/pkgconfig:/home/gnclab/ros2_dashing/install/gtest_vendor/lib/pkgconfig:/home/gnclab/ros2_dashing/install/cyclonedds/lib/x86_64-linux-gnu/pkgconfig:/home/gnclab/ros2_dashing/install/cyclonedds/lib/pkgconfig:/opt/ros/<font color="#EF2929"><b>melodic</b></font>/lib/pkgconfig
</pre>

This means that something was messed up in between which changes the environment variable. Do you agree?

gramss, 2020-05-22:
Something is wrong with your setup somehow..:

```
$ echo $ROS_DISTRO

$ source /opt/ros/dashing/setup.bash 
$ echo $ROS_DISTRO
dashing

```

have you checked out the right branch of this repo? dashing? It is the default now but I do not know how long already.

Also I do not know what you are doing here:
```
gnclab@system76-pc:~$ source ~/ros2_dashing/install/setup.bash 
ROS_DISTRO was set to 'melodic' before. Please make sure that the environment does not mix paths from different distributions.
gnclab@system76-pc:~$ echo $ROS_DISTRO
dashing
gnclab@system76-pc:~$ source install/local_setup.bash
bash: install/local_setup.bash: No such file or directory
```
I would not touch these files. I installed melodic / dashing / eloquent all through the availible packages via apt.

Maybe re setup your whole system with a clean Ubuntu 18.04 install. It is hard to debug from remote. Because your setup.bash from /opt/ros/dashing/ is not producing the same effects as on my machine.

kashishdhal, 2020-05-22:
Hello @gramss ,

Yes, I have installed the default version from apt. But my outputs looks like:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ source /opt/ros/dashing/setup.bash 
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ echo $ROS_DISTRO

<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ source ~/ros2_dashing/install/setup.bash 
ROS_DISTRO was set to &apos;melodic&apos; before. Please make sure that the environment does not mix paths from different distributions.
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ echo $ROS_DISTRO
dashing
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ 
</pre>

So for the new system can I use VM? I was planning to do it in raspberry pi. I don't want to mess-up with the current system as it has too many things installed in it.

gramss, 2020-05-22:
https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/

Are you building ROS2 Dashing from source..?

You need to install ROS2 Dashing like this:
https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/

Maybe remove everything ROS2 Dashing related before. That includes your ~/ros2_dashing/ folder and via apt remove ros-dashing-*

You should be able to install everything on a raspi, but I have never tested it. I'm unsure if the ubuntu packages for the armhf platform (raspi) would be up-to-date.

So better try fixing your main pc until you get the output I got with the tutorial I posted (second link)

kashishdhal, 2020-05-22:
Yes, from what I remember after visiting the link, I installed it from source. So I will go-ahead and delete everything as is suggested by you.

kashishdhal, 2020-05-23:
Hello @pablogs9 and @gramss ,

I reinstalled the dashing but the link given in updated tutorials for Sample app (nuttx) is broken. Please correct this link:
https://github.com/micro-ROS/apps/tree/dashing/examples/uros_pingpong
Also, which RTOS is most supported for micro_ros? I want to use the most supported one.

Regards,
Kashish Dhal

gramss, 2020-05-23:
Hi,

maybe they changed the name to this: https://github.com/micro-ROS/apps/tree/dashing/examples/uros_pong_server But I don't know. I only worked with zephyr on my Nucleo_f401RE so far. I'm not aware which other RTOS are supported with which level of intensity.

Glad that you got your setup working again! :) 
I don't know which MCU you have. But maybe this tutorial might be for you: https://micro-ros.github.io/docs/tutorials/core/first_application_linux/
I think you can run it only on a host if you'd like so. But I never tested it.

Have fun exploring! :) 

Cheers

kashishdhal, 2020-05-23:
Hello @gramss ,

I did host linux one yesterday but wanted to flash something on the board. I purchased the same board given in tutorial so that I can follow through and learn the methodology. 

Regards,
Kashish Dhal

gramss, 2020-05-23:
Maybe try to skip the NuttX and go to the zephyr section. The app is there at least. So you should be able to do the tutorial.

For NuttX you probably need to alter your command to this:
`$ ros2 run micro_ros_setup configure_firmware.sh uros_pong_server`

When you just enter
`$ ros2 run micro_ros_setup configure_firmware.sh`
it should list you all available apps for your board. That's at least how I know it for zephyr.. :)

You can always look up the scripts you are playing with in this repo. This might give you a better overview what is actually happening under the hood.

kashishdhal, 2020-05-23:
Hello @gramss ,

Thanks for the input. 

I was asking about different versions of RTOS because I see that there is a tutorial based on Crazyflie that uses FreeRTOS. I have a few of them and it will be nice if I will be able to run micro-ros on them. Currently, I am using crazyflie_ros stack to fly them around. 

But coming back to our discussion, I wanted to know that if I develop apps on different RTOS versions then will I need to created different workspace folders for each RTOS version? Because running the script will complain firmware already exists. I have already made one folder named micro_ros workspace for nuttx, so if I want to develop apps for zephyr then do I need delete firmware folder or create another worskpace folder for it.

And, if someone want to work on different RTOS versions on same computer then what will he/she do?

gramss, 2020-05-23:
Yeah, I think you can totally run micro-ROS on your crazyflys. Maybe hunt down the code and the maintainer of this code to ask for help setting it up. Besides that there exists already a tutorial regarding the flash process: https://micro-ros.github.io/docs/tutorials/demos/crazyflie_demo/
Try it out and open a new issue if you have problems with this demo in the corresponding repo.

You have multiple options:
* If you want to be able to build multiple RTOS firmwares at a fast pace, than setting up 3 different workspaces would be ideally
* the idea behind those scripts is, that you can rebuild the environment easily over and over again. So you would create a fork of the zephyr_apps repo for example and switch the config to download your fork (remember to rebuild the scripts if you change them! ;) ).
This way you have your code always in a repo and everybody can clone and build it.

Also keep in mind that all RTOS behave a little different in writing them. The core principals of micro-ROS on top of the c-libary rcl remains the same. But things like spinning new threads inside your RTOS are different for all of them.

As you resolved your issue with your first micro-ROS app, you might consider closing this issue now :)

kashishdhal, 2020-05-23:
Hello @gramss 

Please have a look what's wrong here:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup create_firmware_ws.sh zephyr olimex-stm32-e407
Creating firmware for zephyr platform olimex-stm32-e407
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro &quot;ardent&quot;
Skip end-of-life distro &quot;bouncy&quot;
Skip end-of-life distro &quot;crystal&quot;
Add distro &quot;dashing&quot;
Add distro &quot;eloquent&quot;
Add distro &quot;foxy&quot;
Skip end-of-life distro &quot;groovy&quot;
Skip end-of-life distro &quot;hydro&quot;
Skip end-of-life distro &quot;indigo&quot;
Skip end-of-life distro &quot;jade&quot;
Add distro &quot;kinetic&quot;
Skip end-of-life distro &quot;lunar&quot;
Add distro &quot;melodic&quot;
Add distro &quot;noetic&quot;
updated cache in /home/gnclab/.ros/rosdep/sources.cache
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
micro_ros_agent: Cannot locate rosdep definition for [microxrcedds_agent]</pre>

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport serial-usb
App ping_pong for Zephyr not available
Available apps for Zephyr and olimex-stm32-e407:
+-- my_brand_new_app
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup build_firmware.sh
Crosscompiled environment: cleaning path
/home/gnclab/microros_ws/install/micro_ros_setup/lib/micro_ros_setup/build_firmware.sh: line 36: /home/gnclab/microros_ws/firmware/dev_ws/install/setup.bash: No such file or directory
</pre>

gramss, 2020-05-23:
For the first:

probably build your agent in another workspace for now. They are working on a release version that can be just installed later.
That is my setup as well..

The second one:
It seems that the board is called a little different in the zephyr environment: https://github.com/micro-ROS/zephyr_apps/blob/3b05857e14dcf58f31d22329e8b4a66f33ad3f80/apps/ping_pong/CMakeLists.txt#L5
The naming is probably not from micro-ROS but from zephyr themself: https://docs.zephyrproject.org/latest/boards/arm/olimex_stm32_e407/doc/index.html
`olimex_stm32_e407`

kashishdhal, 2020-05-23:
Hello @gramss ,

Now I am getting something like this:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_zephyr_ws</b></font>$ ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport serial-usb
Configuring firmware for zephyr platform olimex_stm32_e407
Using USB serial device.
Configured serial-usb mode with agent at USB serial
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_zephyr_ws</b></font>$ ros2 run micro_ros_setup build_firmware.sh
Crosscompiled environment: cleaning path
Building firmware for zephyr platform olimex_stm32_e407
Selected app: ping_pong
Unrecognized board: olimex_stm32_e407</pre>

kashishdhal, 2020-05-23:
By the way, this time I used board name suggested by you when I build the firmware:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_zephyr_ws</b></font>$ ros2 run micro_ros_setup create_firmware_ws.sh zephyr olimex_stm32_e407
Creating firmware for zephyr platform olimex_stm32_e407
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro &quot;ardent&quot;
Skip end-of-life distro &quot;bouncy&quot;
Skip end-of-life distro &quot;crystal&quot;
Add distro &quot;dashing&quot;
Add distro &quot;eloquent&quot;
Add distro &quot;foxy&quot;
Skip end-of-life distro &quot;groovy&quot;
Skip end-of-life distro &quot;hydro&quot;
Skip end-of-life distro &quot;indigo&quot;
Skip end-of-life distro &quot;jade&quot;
Add distro &quot;kinetic&quot;
Skip end-of-life distro &quot;lunar&quot;
Add distro &quot;melodic&quot;
Add distro &quot;noetic&quot;
updated cache in /home/gnclab/.ros/rosdep/sources.cache
#All required rosdeps installed successfully
/home/gnclab/uros_zephyr_ws/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:10: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_info = yaml.load(sys.stdin)
/home/gnclab/uros_zephyr_ws/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:11: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_keep = yaml.load(open(sys.argv[1]))[&apos;keep&apos;].split()
List of repositories is empty
#All required rosdeps installed successfully
/home/gnclab/uros_zephyr_ws/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:10: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_info = yaml.load(sys.stdin)
/home/gnclab/uros_zephyr_ws/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:11: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_keep = yaml.load(open(sys.argv[1]))[&apos;keep&apos;].split()
Crosscompiled environment: cleaning path
Starting &gt;&gt;&gt; ament_package
Starting &gt;&gt;&gt; ament_lint
Starting &gt;&gt;&gt; ament_cppcheck
Starting &gt;&gt;&gt; gtest_vendor
Starting &gt;&gt;&gt; ament_pep8
Finished &lt;&lt;&lt; gtest_vendor [0.86s]                            
Starting &gt;&gt;&gt; gmock_vendor
Finished &lt;&lt;&lt; gmock_vendor [0.09s]                            
Finished &lt;&lt;&lt; ament_lint [1.07s]                              
Starting &gt;&gt;&gt; ament_flake8
Finished &lt;&lt;&lt; ament_cppcheck [1.28s]                          
Finished &lt;&lt;&lt; ament_package [1.29s]
Finished &lt;&lt;&lt; ament_pep8 [1.28s]
Starting &gt;&gt;&gt; ament_cmake_core               
Starting &gt;&gt;&gt; ament_pyflakes
Finished &lt;&lt;&lt; ament_flake8 [0.51s]                           
Starting &gt;&gt;&gt; ament_pep257
Finished &lt;&lt;&lt; ament_pyflakes [0.50s]                                     
Finished &lt;&lt;&lt; ament_cmake_core [0.51s]
Starting &gt;&gt;&gt; ament_cmake_libraries
Starting &gt;&gt;&gt; ament_cmake_export_libraries
Starting &gt;&gt;&gt; ament_cmake_include_directories
Starting &gt;&gt;&gt; ament_cmake_python
Starting &gt;&gt;&gt; ament_cmake_export_definitions
Starting &gt;&gt;&gt; ament_cmake_export_include_directories
Starting &gt;&gt;&gt; ament_cmake_export_link_flags
Finished &lt;&lt;&lt; ament_pep257 [0.74s]                           
Starting &gt;&gt;&gt; ament_copyright
Finished &lt;&lt;&lt; ament_cmake_libraries [0.55s]
Starting &gt;&gt;&gt; ament_cmake_export_dependencies
Finished &lt;&lt;&lt; ament_cmake_export_libraries [0.77s]
Starting &gt;&gt;&gt; ament_cmake_export_interfaces
[... 1066 lines cut ...]
 * [new branch]            v1.12-branch           -&gt; eprosima/v1.12-branch
 * [new branch]            v1.13-branch           -&gt; eprosima/v1.13-branch
 * [new branch]            v1.14-branch           -&gt; eprosima/v1.14-branch
 * [new branch]            v1.5-branch            -&gt; eprosima/v1.5-branch
 * [new branch]            v1.6-branch            -&gt; eprosima/v1.6-branch
 * [new branch]            v1.7-branch            -&gt; eprosima/v1.7-branch
 * [new branch]            v1.8-branch            -&gt; eprosima/v1.8-branch
 * [new branch]            v1.9-branch            -&gt; eprosima/v1.9-branch
 * [new branch]            v2.0-branch            -&gt; eprosima/v2.0-branch
 * [new branch]            v2.1-branch            -&gt; eprosima/v2.1-branch
Note: checking out &apos;remotes/eprosima/feature/micro_ros&apos;.

You are in &apos;detached HEAD&apos; state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b &lt;new-branch-name&gt;

HEAD is now at 38b77be6c8 Working sensor
#All required rosdeps installed successfully
Segmentation fault (core dumped)
</pre>

kashishdhal, 2020-05-24:
Hello @gramss ,

So I was able to build the firmware for FreeRTOS. However, I am not able to flash it. I actually purchased ST-LINK/V2 as debug adapter instead of ARM-USB-TINY-H. Could that be an issue? or do I need to configure the adapter?

I am getting the following error:
<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_freertos_ws</b></font>$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for freertos platform olimex-stm32-e407
Flashing firmware for freertos platform olimex-stm32-e407
Error. Unsuported OpenOCD USB programmer
</pre>

gramss, 2020-05-24:
never did flash this board. sorry. My Nucleo has a debugger on board and the flashing get handled by the zephyr tool `west`
Now that you mention it, the olimex has some other flashing process integrated in the zephyr RTOS. So I cannot say for sure if it works with your debugger. Check the flash script also under the FreeRTOS folder and look for something that does not match your debugger.

Referring to this page: https://docs.zephyrproject.org/latest/boards/arm/olimex_stm32_e407/doc/index.html#flashing-an-application-to-the-olimex-stm32-e407 you should also be able to flash your board with the standard west tool..

Have a look at my PR here: https://github.com/micro-ROS/micro-ros-build/pull/123/commits/03caa8066fb036f65239540dfa56b8384fc72a41

Maybe add your board also to this. Just for testing..? But remember to rebuild your micro-ros-build packages when you test them under workspace/src/micro..... . Also remember to clean your test steps if @pablogs9 helps you next week.. I think he can support you with the flashing of the olimex board. At least it is the official supported board from them, so it should be doable ;)

>By the way, this time I used board name suggested by you when I build the firmware:

 was this before or after your comment before that..?

kashishdhal, 2020-05-24:
> was this before or after your comment before that..?

I built with the board name before I sent you the output. But I was unable to build for zephyr as you can see in the output. Also, I didn't create ping_pong app for FreeRTOS, it was already there in the Firmware folder. But I was able to build everything. However, when I tried flashing then it gave error as shown above.

Talking about flashing, I can't understand what you wrote. But I will re-read it carefully tomorrow and get back to you if I need help.

kashishdhal, 2020-05-24:
Hello @gramss ,

> My Nucleo has a debugger on board and the flashing get handled by the zephyr tool west

Okay so you have debugger on-board and probably you are using micro-USB to access the debugger and program it from there? Correct?

> Referring to this page: https://docs.zephyrproject.org/latest/boards/arm/olimex_stm32_e407/doc/index.html#flashing-an-application-to-the-olimex-stm32-e407 you should also be able to flash your board with the standard west tool..

So this is basically for Zephyr but as I said earlier, I am not able to build for Zephyr. You saw in the output it is saying unrecognized board.

Also, I saw that there are few apps already developed for crazyflie on FreeRTOS version. So if I would like to stick to the FreeRTOS version instead of Zephyr for now then I would like to know how to flash it on both the Olimex and Crazyflie. 

So according to you, you have never used the debugger to flash, and maybe @pablogs9 will help me with the flashing process? Correct?

gramss, 2020-05-24:
> Okay so you have debugger on-board and probably you are using micro-USB to access the debugger and program it from there? Correct?

I'm using the now upstream micro-ROS scripts to create, configure (with serial and not usb-serial), build and flash my board. Yes.

You mean failed zephyr setup because of: ?
> Segmentation fault (core dumped) (from the create_firmware script)

Yeah.. I was noticing this as well, but never got any feedback if it is critical or not. The created firmware folder is working for me never the less the segmentation fault. Here is the output for my board: https://github.com/micro-ROS/micro-ros-build/issues/115#issuecomment-627480491 also the seg-fault at the end but everything works. I was not able to glimpse what causes this..

> So according to you, you have never used the debugger to flash, and maybe @pablogs9 will help me with the flashing process? Correct?

Yes. He is a maintainer of this repo. I'm just a friendly dude from the interwebs 🙂

kashishdhal, 2020-05-25:
Hello @gramss ,

> You mean failed zephyr setup because of: ?

I was referring to this: 
<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_zephyr_ws</b></font>$ ros2 run micro_ros_setup build_firmware.sh
Crosscompiled environment: cleaning path
Building firmware for zephyr platform olimex_stm32_e407
Selected app: ping_pong
Unrecognized board: olimex_stm32_e407
</pre>
Note: I changed the name of board to olimex_stm32_e407 while creating firmware folder.

But talking about segmentation fault, I don't know if it will create a problem but it did not occur in FreeRTOS version.

> Yes. He is a maintainer of this repo. I'm just a friendly dude from the interwebs slightly_smiling_face

The amount of time, you are spending here, I wish you were in their dev team. 

@pablogs9 would you help me flashing firmware to the board?

I am using ST-Link V2 and here is my output after succesful build of FreeRTOS version:
<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_freertos_ws</b></font>$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for freertos platform olimex-stm32-e407
Flashing firmware for freertos platform olimex-stm32-e407
Error. Unsuported OpenOCD USB programmer
</pre>

pablogs9, 2020-05-25:
Hello @kashishdhal, indeed the ST-Link is not covered by the flash script so it won't recognize it.

The ST-Link V2 is supported by OpenOCD, so it shouldn't' be difficult to integrate it: http://openocd.org/doc/html/Debug-Adapter-Hardware.html#Debug-Adapter-Hardware.

I have found that there is  an OpenOCD script for ST-Link V2 here: `/usr/share/openocd/scripts/interface/stlink-v2.cfg`. So, if you [check this script](https://github.com/micro-ROS/micro-ros-build/blob/dashing/micro_ros_setup/config/freertos/olimex-stm32-e407/flash.sh) maybe yo can add support for your programmer. Maybe something like:

```
...
      elif lsusb -d [USB ID ST-LINK]; then
        PROGRAMMER=interface/stlink-v2.cfg
...
```

Remember to rebuild the `micro-ros-build` package after the changes.

kashishdhal, 2020-05-26:
Hello @pablogs9 ,

Please see the output below:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_freertos_olimex</b></font>$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for freertos platform olimex-stm32-e407
Flashing firmware for freertos platform olimex-stm32-e407
Bus 001 Device 013: ID 0483:3748 STMicroelectronics ST-LINK/V2
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport &quot;hla_swd&quot;. To override use &apos;transport select &lt;transport&gt;&apos;.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 1.618385
Error: init mode failed (unable to connect to the target)
in procedure &apos;init&apos; 
in procedure &apos;ocd_bouncer&apos;
</pre>

Can you guess what's wrong here?

Regards,
Kashish Dhal

pablogs9, 2020-05-26:
Are you using a Olimex STM32-E407? Have you checked the power jumper? Please see `Step 4: Flashing the firmware` (here)[https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/] and ensure that your board is powered when the JTAG connector is plugged in.

kashishdhal, 2020-05-26:
Hello @pablogs9 ,

I have placed order after looking at your website so  I guess it should be the supported one only.

Regarding power, I already mentioned that red led is on. So power is coming from JTAG to the MCU.

kashishdhal, 2020-05-26:
Here is the link from my purchase history:
https://www.mouser.com/ProductDetail/Olimex-Ltd/STM32-E407?qs=UN6GZl1KCcit6Ye0xmPO4A%3D%3D

pablogs9, 2020-05-26:
Could you check the position of `B0_0/B0_1` and `B1_0/B1_1` jumpers?

kashishdhal, 2020-05-26:
Although, I didn't touch them but upon checking the location looks the same as given in picture in your tutorial.

pablogs9, 2020-05-26:
Well, it seems to be an issue related to your JTAG device. It is possible for you to verify in Olimex documentation how to use the STLink with this board?

I have seen that maybe it can be related to the reset: https://electronics.stackexchange.com/questions/204996/stm32-st-link-cannot-connect-to-mcu-after-successful-programming

Maybe the openocd command arguments must be modified when ST-Link is used. I don't have this JTAG here to test but, can you test the flashing procedure pushing the reset button on the board and releasing it just at the beginning of the flash procedure?

kashishdhal, 2020-05-26:
Okay, don't worry, I will the one which is supported.

So what kind of JTAG do you use for Crazyflie? I will buy them together

gramss, 2020-05-26:
> ST-Link is used. I don't have this JTAG here to test 
@pablogs9 if you did not sell your ST IOT Discovery board, then you have an ST-Link programmer available, embedded into it.. 😄 
Same for my nucleo-f401re board. I do not know which version you have but this here has some jumpers to disconnect the st-link from the rest of the board: https://docs.zephyrproject.org/1.13.0/boards/arm/stm32f4_disco/doc/stm32f4_disco.html (and use it to flash your OLIMEX board)
This here has also a ST-Programmer integrated but I think you cannot use it for external programming without sort of heavy modifications..: https://www.st.com/content/dam/technology-tour-2017/hands-on-training_stm32.pdf

pablogs9, 2020-05-26:
For Crazyflie we are using a Segger J-Link to debug, anyways CF has a DFU mode and it can be flashed by USB so I have not tested this J-Link for flashing.

I supposed that all of them were basically the same, but maybe we should review these flashing scripts... let me know if you do some achievements. Maybe it is a good idea to try to build a blinky hex and make some flashing test with openocd, leaving micro-ROS out of the equation.

kashishdhal, 2020-05-26:
Hello @pablogs9 ,

I have not done blinky hex stuff before, so if you can give me some links to follow then I will be glad to do so. Also, if you can explain a bit in detail it would be great!

I have taken basic courses in development of RTOS and bootloader hex files but I worked with TI board only and I have only used IDE (Code Composer Studio) to flash most of the time. We were also provided a Linux script (by the instructor) which was able to flash the code but for that we had to put the device in bootloader mode i.e. by pressing a push button while restarting . Actually, I didn't even used a JTAG before as we had a debugger on-board. So all this stuff is new to me and I have to go in small steps if I have to test something on my own.

pablogs9, 2020-05-27:
You should use `ros2 run micro_ros_setup create_firmware_ws.sh freertos crazyflie21`, then the steps are basically the same.

To flash we use the [DFU mode](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/building-and-flashing/dfu/) and the flash script of this build system.

kashishdhal, 2020-05-27:
And do we have to do ping_pong stuff? 

ros2 run micro_ros_setup configure_firmware.sh ping_pong | --transport serial --dev 3

pablogs9, 2020-05-27:
You can configure some crazyflie application, run `ros2 run micro_ros_setup configure_firmware.sh` to get a list of available apps. Since Crazyflie only has one transport the `--transport` flag is not necessary.

kashishdhal, 2020-05-27:
We still need --dev 3 flag?

Also do I need run   `ros2 run micro_ros_setup configure_firmware.sh [app name] [options]
`       for every app?

Available apps for FreeRTOS and crazyflie21:
+-- add_two_ints
+-- crazyflie_position_publisher
+-- crazyflie_position_publisher_reconnection
+-- crazyflie_uros_sample
+-- int32_publisher
+-- olimex_uros_sample
+-- ping_pong

Is it possible to do configure all apps at once?

pablogs9, 2020-05-27:
no, you only can configure one app per build.

kashishdhal, 2020-05-28:
So I was able to flash the firmware. But I don't know how to test or use it. I already have radio installed and udev permissions set as I was using it with crazyflie_ros stack. On the crazyflie demo page, the procedure is not displayed. If you can let me know then it would be great! I have configured the crazyflie_position_publisher.

Also, as I mentioned earlier, in [this](https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/) page, the sample app link for Nuttx is broken. Please see if you can correct that?

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_crazyflie_ws</b></font>$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for freertos platform crazyflie21
Flashing firmware for freertos platform crazyflie21
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: &quot;Internal Flash  &quot;
Downloading to address = 0x08000000, size = 283904
Download	[=========================] 100%       283904 bytes
Download done.
File downloaded successfully
</pre>

pablogs9, 2020-05-28:
Well, once you have the firmware flashed you will need a couple of thighs:

- A Crazyflie client with a radio to serial bridge. You can find the steps in the demo Dockerfile: https://github.com/micro-ROS/micro-ROS_crazyflie_demo/blob/dashing/dockerfile/client/Dockerfile 
- A micro-ROS Agent listening the serial port. You can find how to install it here: https://github.com/micro-ROS/micro-ros-build/tree/dashing/micro_ros_setup#building-micro-ros-agent

The Crazyflie client, once executed, will tell you in the command line where it has created the virtual serial port to use with the agent.

pablogs9, 2020-05-28:
Our custom Crazyflie client (the official GUI app with the radio to serial bridge) can be found in this branch: https://github.com/eProsima/crazyflie-clients-python/tree/Micro-XRCE-DDS_Bridge

kashishdhal, 2020-05-28:
So I am not able to find virtual serial port, which I guess is required by `ros2 run micro_ros_agent micro_ros_agent serial --dev [device]` I tried using `radio://0/100/2M/E7E7E7E702` but that doesn't work.

After using the client specified by you, I get the following output:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ python3 crazy_micro/crazyflie-clients-python/bin/cfclient
INFO:cfclient.gui:Disabling STL printouts
INFO:cfclient.utils.input.inputreaders:Input readers: [&apos;linuxjsdev&apos;, &apos;pysdl2&apos;]
INFO:cfclient.utils.input.inputreaders:Successfully initialized [linuxjsdev]
INFO:cfclient.utils.input.inputreaders:Could not initialize [pysdl2]: No SDL2 support on Linux
INFO:cfclient.utils.input.inputinterfaces:Found interfaces: [&apos;leapmotion&apos;, &apos;wiimote&apos;, &apos;zmqpull&apos;]
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [leapmotion]: Leap Motion library probably not installed (No module named &apos;leapsdk&apos;)
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [wiimote]: Missing cwiid (wiimote) driver No module named &apos;cwiid&apos;
INFO:cfclient.utils.config:Dist config read from /home/gnclab/crazy_micro/crazyflie-clients-python/src/cfclient/configs/config.json
INFO:cfclient.utils.config:Config file read from [/home/gnclab/.config/cfclient/config.json]
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [zmqpull]: ZMQ input disabled in config file
INFO:cfclient.utils.zmq_param:Biding ZMQ for parameters at tcp://*:1213
INFO:cfclient.utils.zmq_led_driver:Biding ZMQ for LED driverat tcp://*:1214
INFO:cfclient.utils.input:Using device blacklist [(VirtualBox|VMware)]
INFO:cflib.crtp.radiodriver:v0.53 dongle with serial N/A found
INFO:cfclient.ui.tabs.QualisysTab:Switching Flight Mode to: FlightModeStates.DISCONNECTED
INFO:cflib.crtp.serialdriver:Initialized serial driver.
INFO:cflib.drivers.cfusb:Looking for devices....
INFO:cflib.crtp.prrtdriver:Initialized PRRT driver.
INFO:cflib.crtp.radiodriver:v0.53 dongle with serial N/A found
INFO:cflib.crtp.serialdriver:Initialized serial driver.
INFO:cflib.drivers.cfusb:Looking for devices....
INFO:cflib.crtp.prrtdriver:Initialized PRRT driver.
INFO:cflib.crazyflie:Callback-&gt;Connection initialized[radio://0/100/2M/E7E7E7E702]
INFO:cflib.crazyflie:We are connected[radio://0/100/2M/E7E7E7E702], request connection setup
INFO:cflib.crazyflie:Callback-&gt;Connected to [radio://0/100/2M/E7E7E7E702]
WARNING:cflib.crazyflie.log:Error no LogEntry to handle id=3
INFO:cflib.crazyflie.platformservice:Procotol version: 4
INFO:cflib.crazyflie.toc:TOC for port [5] found in cache
INFO:cflib.crazyflie:Log TOC finished updating
INFO:cflib.crazyflie.mem:8 memories found
INFO:cflib.crazyflie:Memories finished updating
INFO:cflib.crazyflie.toc:TOC for port [2] found in cache
INFO:cflib.crazyflie:Param TOC finished updating
INFO:cflib.crazyflie:Callback-&gt;Connection setup finished [radio://0/100/2M/E7E7E7E702]
INFO:cfclient.utils.logconfigreader:Parsing [stabilizer.json]
QObject::connect: Cannot queue arguments of type &apos;QList&lt;QPersistentModelIndex&gt;&apos;
(Make sure &apos;QList&lt;QPersistentModelIndex&gt;&apos; is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type &apos;QList&lt;QPersistentModelIndex&gt;&apos;
(Make sure &apos;QList&lt;QPersistentModelIndex&gt;&apos; is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type &apos;QList&lt;QPersistentModelIndex&gt;&apos;
(Make sure &apos;QList&lt;QPersistentModelIndex&gt;&apos; is registered using qRegisterMetaType().)
INFO:cfclient.ui.tabs.LEDTab:Memory: id=1, type=LED driver, size=24
INFO:cfclient.ui.tabs.locopositioning_tab:Crazyflie connected to radio://0/100/2M/E7E7E7E702
INFO:cfclient.ui.tabs.locopositioning_tab:Requesting loco deck parameter
INFO:cfclient.ui.main:LED write done callback
QObject::connect: Cannot queue arguments of type &apos;QList&lt;QPersistentModelIndex&gt;&apos;
(Make sure &apos;QList&lt;QPersistentModelIndex&gt;&apos; is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type &apos;QList&lt;QPersistentModelIndex&gt;&apos;
(Make sure &apos;QList&lt;QPersistentModelIndex&gt;&apos; is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type &apos;QList&lt;QPersistentModelIndex&gt;&apos;
(Make sure &apos;QList&lt;QPersistentModelIndex&gt;&apos; is registered using qRegisterMetaType().)
INFO:cfclient.ui.tabs.FlightTab:[imu_sensors.BMP388]: 0
INFO:cflib.crazyflie.log:Have successfully started logging for id=1
INFO:cfclient.ui.tabs.FlightTab:[imu_sensors.HMC5883L]: 0
INFO:cflib.crazyflie.log:Have successfully started logging for id=3
INFO:cflib.crazyflie.log:Have successfully started logging for id=4
INFO:cflib.crazyflie.log:Have successfully started logging for id=5
INFO:cfclient.ui.tabs.FlightTab:[imu_sensors.MS5611]: 0
INFO:cfclient.ui.tabs.locopositioning_tab:No Loco deck installed
INFO:cfclient.ui.tabs.FlightTab:Changed effect to 0
QObject::connect: Cannot queue arguments of type &apos;QVector&lt;int&gt;&apos;
(Make sure &apos;QVector&lt;int&gt;&apos; is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type &apos;QVector&lt;int&gt;&apos;
(Make sure &apos;QVector&lt;int&gt;&apos; is registered using qRegisterMetaType().)
INFO:cfclient.ui.tabs.FlightTab:Changed effect to 0
</pre>

Please have a look to see if you can help!

pablogs9, 2020-05-29:
Are you using `Micro-XRCE-DDS_Bridge` branch?

You should see this print:
https://github.com/eProsima/crazyflie-clients-python/blob/8c8913f3878e82b913e40828255b927555f134a8/src/cfclient/ui/main.py#L871

kashishdhal, 2020-05-29:
I cloned your given link in crazy_micro folder and from there only I am running. But I will check it tomorrow again. Maybe I will redownload and run again.

pablogs9, 2020-05-29:
but have you cloned the correct branch?

gramss, 2020-05-29:
You need to check out a specific branch after cloning the repo.
Go into the repo folder and there enter `git checkout -b <tab tab>`. Then you’ll see the branch you need for this. Type in the specific branch and hit enter.
Rebuild your ws after this.

kashishdhal, 2020-05-29:
Thanks @gramss , I didn't check out that branch, will do that!

kashishdhal, 2020-05-31:
Hello @pablogs9 .

Regarding the Olimex and FreeRTOS, I got the supported debug adapter and I was able to flash the firmware. However, I don't have the USB to Serial Wire instead I have FTDI cable. I was wondering that if that should work? or do I need to stick to the one prescribed in the tutorials?

See the output below:
<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_freertos_olimex</b></font>$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for freertos platform olimex-stm32-e407
Flashing firmware for freertos platform olimex-stm32-e407
Bus 001 Device 018: ID 15ba:002a Olimex Ltd. ARM-USB-TINY-H JTAG interface
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport &quot;jtag&quot;. To override use &apos;transport select &lt;transport&gt;&apos;.
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
jtag_ntrst_delay: 100
none separate
cortex_m reset_config sysresetreq
Info : clock speed 2000 kHz
Info : JTAG tap: stm32f4x.cpu tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x4)
Info : JTAG tap: stm32f4x.bs tap/device found: 0x06413041 (mfg: 0x020 (STMicroelectronics), part: 0x6413, ver: 0x0)
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
adapter speed: 2000 kHz
Info : JTAG tap: stm32f4x.cpu tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x4)
Info : JTAG tap: stm32f4x.bs tap/device found: 0x06413041 (mfg: 0x020 (STMicroelectronics), part: 0x6413, ver: 0x0)
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0802ccf0 msp: 0x20020000
auto erase enabled
Info : device id = 0x10076413
Info : flash size = 1024kbytes
wrote 262144 bytes from file build/micro-ROS.bin in 7.099174s (36.061 KiB/s)
adapter speed: 2000 kHz
Info : JTAG tap: stm32f4x.cpu tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x4)
Info : JTAG tap: stm32f4x.bs tap/device found: 0x06413041 (mfg: 0x020 (STMicroelectronics), part: 0x6413, ver: 0x0)
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/uros_freertos_olimex</b></font>$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-FTDI_USB__-__Serial-if00-port0
Press CTRL+C to exit
<font color="#75507B">[1590901852.785730]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">TermiosAgentLinux.cpp</font> | <font color="#D3D7CF">init                    </font> | <font color="#4E9A06">running...</font>             | fd: 3
</pre>

However, I am not able to see the ros2 topics and I can't echo /microROS/ping. Could this be related to wire? Please let me know.

Regards,
Kashish Dhal

pablogs9, 2020-06-01:
Regarding the first comment, as you can read it is saying that a file can't be found. The reason is that this client bridge is designed to work with the Docker system. You can create this file to avoid this error or you can comment [these lines](https://github.com/eProsima/crazyflie-clients-python/blob/8c8913f3878e82b913e40828255b927555f134a8/src/cfclient/ui/main.py#L872).

Regarding the FTDI cable, I don't know which type or version is it, so I can't know. As far as it has a UART interface with a Tx, Rx and GND connections and it mounts a serial device over the USB it should work.

kashishdhal, 2020-06-01:
Okay,

I have this cable:
https://www.amazon.com/gp/aw/d/B07B5TP67V/ref=ppx_yo_mob_b_track_package_o1_img?ie=UTF8&psc=1

pablogs9, 2020-06-01:
How are you connecting the cable to the Olimex board?

Regarding the agent, I would recommend using also the `-v6` flag to increase verbosity and check if any serial communication is happening.

kashishdhal, 2020-06-01:
I am using only 3 wires as is given in tutorial. But I guess the color codes are different here, I will retry that with -v flag.

[5 more replies]

closed 2020-07-29 by kashishdhal