micro-ROS/micro_ros_setup #128: Error building First micro-ROS Application on Linux on Foxy

raymond-devries, 2020-06-03:
I know foxy is not released for another couple of days but I saw here https://github.com/ros2/ros2/issues/830 that foxy support has been implemented. 

With a clean installation of Ubuntu 20.04 and ros2 foxy built from source I tried to build the micro-ROS app on Linux from the features/foxy_migration branch and ran into these errors. 
`--- stderr: rosidl_typesupport_microxrcedds_cpp_tests
/home/raymonddevries/microros_ws/src/uros/rosidl_typesupport_microxrcedds/test/cpp/src/test_typesupport.cpp:24:3: fatal error: rosidl_typesupport_microxrcedds_test_msg/msg/primitive__rosidl_typesupport_microxrcedds_cpp.hpp: No such file or directory
   24 |   <rosidl_typesupport_microxrcedds_test_msg/msg/primitive__rosidl_typesupport_microxrcedds_cpp.hpp>
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/test_typesupport.dir/build.make:63: CMakeFiles/test_typesupport.dir/src/test_typesupport.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/test_typesupport.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
`
After this I decided to delete the test folder from rosidl_typesupport_microxrcedds just to see what happens and I didn't receive the same error message and it seems like the package compiled without the test folder. I tried compiling again and this error came up. 

`--- stderr: rclc                                                                                                          
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CXX_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CXX_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CXX_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CXX_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CXX_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CXX_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CXX_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CXX_LINK_EXECUTABLE
CMake Generate step failed.  Build files cannot be regenerated correctly.
---
`
I cannot figure out why this is coming up but to fix it I ended up having to update the rclc package to the latest master branch as it seems the foxy_migration branch of micro-ros-build still is cloning the dashing branch of the rclc package (https://github.com/micro-ROS/micro-ros-build/blob/feature/foxy_migration/micro_ros_setup/config/host/generic/client_host_packages.repos).
`uros/rclc:
    type: git
    url: https://github.com/micro-ROS/rclc.git
    version: dashing`

Rclc successfully compiled however the rclc examples still failed. After deleting those as well everything compiled, and I was successfully able to run the ping_pong example.

I'm still getting acquainted with the project, so I definitely don't know the best solution to these issues but hopefully this information is helpful.

Thank you.

pablogs9, 2020-06-03:
We are currently under active development these days. If you want to test a stable micro-ROS foxy port, please wait some days until the migration is complete.

raymond-devries, 2020-06-03:
That’s what I figured. I just put out the errors I was experiencing in case they were helpful for the migration. If micro-ROS team was already aware of these issues or you see it fit to close this issue feel free to do so.

pablogs9, 2020-06-03:
Thanks, stay tuned the next few days for the release!

closed 2020-06-03 by pablogs9