micro-ROS/micro_ros_setup #129: Issues building with Nuttx and Zephyr - First micro-ROS Application

CLansdaleBEC, 2020-06-03:
Hello! I am trying to run an example on the Olimex-stm32-e407 by following the First micro-ROS application tutorial for RTOS and am running into some errors while building.

Setup: Ubuntu 18.04, Dashing Diameta 

Steps Taken:
Install micro-ROS build system as described in the [first tutorial
](https://micro-ros.github.io/docs/tutorials/core/first_application_linux/)

    # Source the ROS 2 installation
    source /opt/ros/$ROS_DISTRO/setup.bash

    # Create a workspace and download the micro-ROS tools
    mkdir microros_ws 
    cd microros_ws
    git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro-ros-build.git src/micro-ros-build

    # Update dependencies using rosdep
    sudo apt update && rosdep update
    rosdep install --from-path src --ignore-src -y

    # Build micro-ROS tools and source them 
    colcon build
    source install/setup.bash

From here I first tried to build the ws with nuttx:

    ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407

and ran into the same output as Kashish in [this post](https://github.com/micro-ROS/micro-ros-build/issues/121)

    Starting >>> nuttx_kconfig_vendor
    --- stderr: nuttx_kconfig_vendor                                       
    ar: `u' modifier ignored since `D' is the default (see `U')
    ar: `u' modifier ignored since `D' is the default (see `U')
    --- 
    Finished <<< nuttx_kconfig_vendor [17.9s]

    Summary: 55 packages finished [1min 28s]
      1 package had stderr output: nuttx_kconfig_vendor
    #All required rosdeps installed successfully
    Installing uClibc++ in the NuttX source tree
    Installation succeeded

    #All required rosdeps installed successfully
    502
    504
    #All required rosdeps installed successfully

These seemed passable so I continued to the configure_firmware.sh step where it appears the app uros_pingpong doesn't exist, I looked through the firmware directory and found that a "pingpong" example does exist, so I used that.

    ros2 run micro_ros_setup configure_firmware.sh pingpong

Here I got an errr in the build 

    cd /home/connor/nuttx_microros_ws/firmware/apps/; colcon build --cmake-target clean --cmake-    target-skip-unavailable
    Starting >>> firmware
    --- stderr: firmware                         
    CMake Error: The source directory "/home/connor/nuttx_microros_ws/firmware/apps" does not appear to contain CMakeLists.txt.
    Specify --help for usage, or press the help button on the CMake GUI.
    ---
    Failed   <<< firmware	[ Exited with code 1 ]

    Summary: 0 packages finished [0.25s]
      1 package failed: firmware
      1 package had stderr output: firmware

My question here: is it necessary to build a whole new app? I really just want to flash the board and run an example, there is uros_pong_server and pingpong, and then the tutorial suggests creating uros_pingpong, which one do you recommend I use?

Following @gramss advice in the [previous post](https://github.com/micro-ROS/micro-ros-build/issues/121), I then tried Zephyr instead of Nuttx. First I set up the microROS build workspace, then when I build with 

[... 75 lines cut ...]
 * [new branch]            v1.7-branch                    -> eprosima/v1.7-branch
 * [new branch]            v1.8-branch                    -> eprosima/v1.8-branch
 * [new branch]            v1.9-branch                    -> eprosima/v1.9-branch
 * [new branch]            v2.0-branch                    -> eprosima/v2.0-branch
 * [new branch]            v2.1-branch                    -> eprosima/v2.1-branch
Note: checking out 'remotes/eprosima/feature/micro_ros'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 38b77be6c8 Working sensor
#All required rosdeps installed successfully
Segmentation fault (core dumped)
```
(I clipped the output here as it's very long)

Do you guys have any ideas here? Again I'm running a fresh ROS2 setup on a new Ubuntu VM. I'm not really sure how best to continue. I'm happy to provide any logs or additional info as needed need. Thank you very much.

Connor

gramss, 2020-06-03:
Hey,

The segfault is very irritating, but I have not experienced any downsides from it..
The firmware folder should be already complete..
@pablogs9 , I still do not know where it comes from. It is definitely irritating..

Continue on your pursuit! :) But you‘ll need to alter the scripts (build / flash) to work for your board! Look at my various issues/PRs also on zephyr_apps and you might figure out where you need to add your board! :)

CLansdaleBEC, 2020-06-03:
Good to know about the seg fault. Also, I'm using the Olimex-STM32-E407 which I believe is the board recommended by the tutorial. 

After creating the firmware I configure with
```
 ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport serial --dev /dev/ttyACM0
```

And then while building I get a lot of warnings and then it fails on microxrcedds_client 

```
Summary: 8 packages finished [16.7s]
  1 package failed: microxrcedds_client
  9 packages had stderr output: microcdr microxrcedds_client rcutils rosidl_adapter rosidl_cmake rosidl_generator_c rosidl_parser rosidl_typesupport_interface tinydir_vendor
  40 packages not processed
Makefile:57: recipe for target 'colcon_compile' failed
make[3]: *** [colcon_compile] Error 2
make[3]: Leaving directory '/home/connor/zephyr_microros_ws/firmware/zephyr_apps/microros_extensions'
CMakeFiles/microroslib_project.dir/build.make:134: recipe for target 'microros_extensions/src/microroslib_project-stamp/microroslib_project-build' failed
make[2]: *** [microros_extensions/src/microroslib_project-stamp/microroslib_project-build] Error 2
make[2]: Leaving directory '/home/connor/zephyr_microros_ws/firmware/build'
CMakeFiles/Makefile2:2026: recipe for target 'CMakeFiles/microroslib_project.dir/all' failed
make[1]: *** [CMakeFiles/microroslib_project.dir/all] Error 2
make[1]: Leaving directory '/home/connor/zephyr_microros_ws/firmware/build'
Makefile:106: recipe for target 'all' failed
make: *** [all] Error 2
FATAL ERROR: command exited with status 2: /usr/bin/cmake --build /home/connor/zephyr_microros_ws/firmware/build
```

I have the most recent version of Cmake, and I tried to build twice as the readme mentions that sometimes it fails the first time. 

The error message doesn't seem particularly helpful, any idea what Error code 2 is?

CLansdaleBEC, 2020-06-04:
To give you guys an update, I tried again using FreeRTOS this time and was able to get a build working error free (!). Now to flash the board, I have a Segger JLink Pro that doesn't look to be supported, so I'll go into openocd land to figure that out :). 

Thanks for the help, should I go ahead and close out this ticket?

pablogs9, 2020-06-05:
Hey @CLansdaleBEC, I was investigating your issue with Zephyr. But we are currently working on the Foxy port so I had no much time. 

**Segmentation fault**

In order to replicate the seg fault error, I have done the following steps in a clean `ros:dashing` docker (the one [here](https://hub.docker.com/layers/ros/library/ros/dashing/images/sha256-d666360ccd234dddcbfc91cee9788f8468dec621e71a48f497b556f04d097414?context=explore)):

```bash
cd /
source /opt/ros/$ROS_DISTRO/setup.bash
mkdir uros_ws && cd uros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro-ros-build.git src/micro-ros-build
rosdep update && rosdep install --from-path src --ignore-src -y
colcon build
source install/local_setup.bash

ros2 run micro_ros_setup create_firmware_ws.sh zephyr olimex-stm32-e407
```

This steps does not gives me any seg. fault. Is it possible for you to provide me a known environment and a step by step guide for reproducing this issue?

**Configuration step**

In your configuration step you are using the option `--dev`. This flag is intended for the serial port used on the board. As long as the board will be using `UART_0`, you should set the flag to `--dev 0`

**Build error**

There are a bug in the `zephyr_apps` repo, you should be able to build it checking out this branch: https://github.com/micro-ROS/zephyr_apps/tree/hotfix/remove_wifi_demo

CLansdaleBEC, 2020-06-05:
I am running an Ubuntu 18.04 virtual machine with no special configuration. To reproduce the seg fault I followed the guide starting from reinstalling ros2. Is there any more information I can provide to help?

The commands I followed:
```
sudo locale-gen en_US en_US.UTF-8
sudo update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
export LANG=en_US.UTF-8

sudo apt update && sudo apt install curl gnupg2 lsb-release
curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -

sudo sh -c 'echo "deb http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'
sudo apt update
sudo apt install ros-dashing-desktop
# Source the ROS 2 installation
source /opt/ros/dashing/setup.bash

# Create a workspace and download the micro-ROS tools
mkdir microros_ws 
cd microros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro-ros-build.git src/micro-ros-build

# Update dependencies using rosdep
sudo apt update && rosdep update
rosdep install --from-path src --ignore-src -y

# Build micro-ROS tools and source them
colcon build
source install/local_setup.bash
```

I went ahead and built on the zephyr hotfix branch you linked and that was able to build fine. Still working on flashing the board.

CLansdaleBEC, 2020-06-08:
For anyone in the future: I was able to get it working using freeRTOS with a serial connection on USART3 and Jlink Pro JTAG interface using opencd. Thank you @pablogs9 and @gramss for the help!

kashishdhal, 2020-06-08:
Same here @CLansdaleBEC, I was able to get it work only for FreeRTOS. But the problem I am facing is after the MCU is rebooted, it doesn't work anymore. Did you experience the same issue? I wrote to @pablogs9 and was waiting for the reply. But since you flashed it, you might have also experienced this. After reboot, the Green LED doesn't flash anymore neither do ROS work.

CLansdaleBEC, 2020-06-11:
@kashishdhal yes, I saw the same behavior. Although my board was having difficulties already with the green led and a sticky JTAG error, so I have moved on to a different board for development. Good luck!

kashishdhal, 2020-06-15:
Thanks @CLansdaleBEC

closed 2020-07-14 by CLansdaleBEC