micro-ROS/micro_ros_setup #131: Foxy migration

pablogs9, 2020-06-05:
This issue reports all repos involved in foxy migration and details the future requirements for each of them.

micro-ROS Foxy migration is compound of multiple repos with `feature/foxy_migration` branches that must be merged into a final `foxy` branch on each of them:

| Repo | Link | Comments | PR |
|-|-|-|-|
| micro-ROS/micro-ros-build | [feature/foxy_migration](https://github.com/micro-ROS/micro-ros-build/tree/feature/foxy_migration) | | |
| micro-ROS/rcl | [feature/foxy_migration](https://github.com/micro-ROS/rcl/tree/feature/foxy_migration) | Should be merged into ros2/rcl |
| micro-ROS/rcutils | [feature/foxy_migration](https://github.com/micro-ROS/rcutils/tree/feature/foxy_migration) | Should be merged into ros2/rcutils |
| micro-ROS-demos| [feature/foxy_migration](https://github.com/micro-ROS/micro-ROS-demos/tree/feature/foxy_migration) | | https://github.com/micro-ROS/micro-ROS-demos/pull/25 |
| micro-ROS/rosidl_typesupport_microxrcedds | [feature/foxy_migration](https://github.com/micro-ROS/rosidl_typesupport_microxrcedds/tree/feature/foxy_migration) | |
| micro-ROS/rmw-microxrcedds | [feature/foxy_migration](https://github.com/micro-ROS/rmw-microxrcedds/tree/feature/foxy_migration) | |
| micro-ROS/apps | [feature/foxy_migration](https://github.com/micro-ROS/apps/tree/feature/foxy_migration) | |
| micro-ROS/apps | [feature/foxy_migration](https://github.com/micro-ROS/apps/tree/feature/foxy_migration) | |
| micro-ROS/freertos_apps | [WIP](https://github.com/micro-ROS/freertos_apps) | |
| micro-ROS/raspbian_apps | [WIP](https://github.com/micro-ROS/raspbian_apps) | |
| micro-ROS/micro-ROS-Agent | [Not started](https://github.com/micro-ROS/micro-ROS-Agent) | Check required changes|
| micro-ROS/zephyr_apps | [feature/foxy_migration](https://github.com/micro-ROS/zephyr_apps/tree/feature/foxy_migration) | |

_Before release every row must have a PR merged from `feature/foxy_migration` to `foxy`_

Other required PR before release are:

- [ ] https://github.com/micro-ROS/rclc/pull/33
- [ ] https://github.com/micro-ROS/rclc/pull/32
- [ ] https://github.com/micro-ROS/zephyr_apps/pull/10
- [ ] https://github.com/micro-ROS/zephyr_apps/pull/9
- [ ] https://github.com/micro-ROS/zephyr_apps/pull/8
- [ ] https://github.com/micro-ROS/zephyr_apps/pull/10
- [ ] https://github.com/micro-ROS/micro-ros-build/pull/130
- [ ] https://github.com/micro-ROS/apps/pull/23
- [ ] https://github.com/micro-ROS/rmw-microxrcedds/pull/64
- [ ] https://github.com/micro-ROS/rmw-microxrcedds/pull/65
- [ ] https://github.com/micro-ROS/rmw-microxrcedds/pull/66
- [ ] https://github.com/micro-ROS/rmw-microxrcedds/pull/67

MillerLuca, 2020-06-22:
Hello,

You are doing a great job with the development of Micro-ROS, 
We are wondering when we should expect a Foxy release of Micro ROS ? 

Thank you for your hard work,
Luca Miller

pablogs9, 2020-06-22:
Hi thank!
We are currently working on it. I hope it is ready for the next week.

MillerLuca, 2020-06-22:
Great !!
Looking forward to play around with it,
Thank you for your hard work ;-)

russkel, 2020-07-27:
This looks great. Is this complete now @pablogs9 ?

pablogs9, 2020-07-27:
Yes, Foxy release is out!

closed 2020-07-27 by pablogs9