micro-ROS/micro_ros_setup #137: Migrating MicroROS to existing application

krm-int, 2020-06-19:
Dear MicroROS

I've been working with MicroROS on the Olimex-stm32-e470 running the different OS's.
I would now like to implement MicroROS in another application running on a different board.
**System info:**
- MCU: STM32F407ZGTx
- Memory: RAM = 192k, CCMRAM = 64k, FLASH = 896K
- OS: FreeRTOS
- COMPILER: arm-none-eabi-gcc 4.9.3

As i see it, the most intuitive approach would be to build a `libmicroros.a` like you do in the FreeRTOS apps. I have a few issues tracing the setup/build pipe you provide, as many scripts run & call each other.
Would you be able to provide the steps to do so?
**A few thoughts:**
- I would like to avoid using ROS-CLI.
- Is it possible to freeze the build pipe in a stable state?
- When a library is created, is it just a matter of adding it the the `OBJECTS ` in Makefile?
- The build could probably be done inside a container?

ps. I have tried to do this on my own, but I never got it to compile.

Thanks
Regards KRM

pablogs9, 2020-06-19:
Hello @krm-int, we are stabilizing the build system for the foxy release. But as you are saying, in FreeRTOS is a matter of generating the `libmicroros.a` and linking your app with it. 

Your hw setup seems feasible to port, so using the Olimex STM32E407 scripts as a reference you should be able to achieve it.

**Some questions:**
-  Could you share a repo with your approach?
-  Could you detail where is the building process failing?
- In which board is your STM32F407ZGTx? Is it a development board or a custom one?

krm-int, 2020-06-19:
I tried to mimic your setup with dev_ws, mcu_ws etc.
The build failed when building the mcu_ws. Whether it was due to the toolchain, environment or something else, I don't know - but then again, I'm not 100% how the micro_ros_build works..
My board is custom, so I only know those Memory-specs from the linker script.

Essentially I want to end up with something like this:
- Single setup step which pulls all source to build microROS without any autogenerated files (those are very coinfusing when looking from the outside)
- A single command to only build the `libmicroros.a` (maybge the same Makefile rules as your freertos-extensions?
If you can point me in the right that would be awesome.

pablogs9, 2020-06-22:
Could you paste your building error?

krm-int, 2020-06-23:
Hi @pablogs9 - sorry for the late answer, I've been trying to replicate my original build pipe, but w/o success.

In the meantime I've noticed something that could be a fundamental issue in this context.
My current FreeRTOS application is NOT POSIX - i can see it's has been [an issue before](https://github.com/micro-ROS/micro-ROS.github.io/issues/20).
Is it even possible to build microROS/microXRCEDDS without POSIX?

I am using LWIP though.

### A little about my current approach:
The mcu/dev_ws contain the same repos as your build pipe (foxy).

**My compile commands (in Makefile):**
```
compile_uros: arm_toolchain
	cd $(MCU_WS); \
	colcon build \
		--packages-ignore-regex=.*_cpp \
		--metas $(METAS_DIR)/colcon.meta \
		--cmake-args\
		-DCMAKE_TOOLCHAIN_FILE=$(TOOLCHAIN_FILE) \
		-DCMAKE_POSITION_INDEPENDENT_CODE=OFF \
		-DTHIRDPARTY=ON \
		-DBUILD_SHARED_LIBS=OFF \
		-DBUILD_TESTING=OFF \
		-DCMAKE_BUILD_TYPE=$(BUILD_TYPE) \
		-DCMAKE_VERBOSE_MAKEFILE=ON;
```

Error occurs then building _Micro-XRCE-DDS-Client_:
```
In file included from /workspaces/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/serial/serial_transport_posix.c:1:
/workspaces/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/serial/serial_transport_posix.h:23:10: fatal error: poll.h: No such file or directory
   23 | #include <poll.h>
      |          ^~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:404: CMakeFiles/microxrcedds_client.dir/src/c/profile/transport/serial/serial_transport_posix.c.o] Error 1
make[3]: *** Waiting for unfinished jobs....
In file included from /workspaces/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/udp/udp_transport_posix.c:1:
/workspaces/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/ip/udp/udp_transport_posix.h:25:10: fatal error: poll.h: No such file or directory
   25 | #include <poll.h>
      |          ^~~~~~~~
compilation terminated.
In file included from /workspaces/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/tcp/tcp_transport_posix.c:1:
/workspaces/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/ip/tcp/tcp_transport_posix.h:25:10: fatal error: poll.h: No such file or directory
   25 | #include <poll.h>
      |          ^~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:339: CMakeFiles/microxrcedds_client.dir/src/c/profile/transport/ip/udp/udp_transport_posix.c.o] Error 1
/workspaces/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/ip_posix.c:19:10: fatal error: arpa/inet.h: No such file or directory
   19 | #include <arpa/inet.h>
      |  
```

**Solution?**
I have added `"-DUCLIENT_PLATFORM_POSIX_NOPOLL=ON"` to `"microxrcedds_client"` in `colcon.meta` which removed some include errors, but not all.

Maybe this is a eProsima related question?

Regards

pablogs9, 2020-06-23:
Well, you should be able to build a Micro XRCE-DDS app using LWIP and the  `-DUCLIENT_PLATFORM_POSIX_NOPOLL=ON` flag. The library would need also a implementation of  POSIX `clock_gettime()` here: https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/abcef44b1cf166445f923ce7e49ef524350d9842/src/c/util/time.c

Here you have a sample Micro XRCE-DDS app (no micro-ROS, only the middleware) for the Olimex STM32 E407 board using STM32CubeMX: https://github.com/eProsima/Micro-XRCE-DDS-Apps/tree/feature/refactor_multiple_rtos_v2/FreeRTOS

Regarding Micro-ROS stack, we are targeting POSIX API so I guess that you will need some compliance. For the micro-ROS + FreeRTOS port we are using the FreeRTOS POSIX extension. If you can add it to your project, the port to micro-ROS will be easier. You can check this addon here: https://github.com/micro-ROS/freertos_apps/tree/dashing/microros_olimex_e407_extensions/FreeRTOS-Plus-POSIX

krm-int, 2020-06-23:
Hi @pablogs9 - sorry for the late answer, I've been trying to replicate my original build pipe, but w/o success.

In the meantime I've noticed something that could be a fundamental issue in this context.
My current FreeRTOS application is NOT POSIX - i can see it's has been an [issue before](https://github.com/micro-ROS/micro-ROS.github.io/issues/20).
Is it even possible to build microROS/microXRCEDDS without POSIX?

I am using LWIP.

pablogs9, 2020-06-23:
> Is this more a eProsima micro-xrce-dds related question?

It depends on if you are targeting micro-ROS or Micro XRCE-DDS.

krm-int, 2020-06-23:
I definitely want microROS as well

pablogs9, 2020-06-23:
So, I guess that you would need a POSIX extension to your FreeRTOS environment.

Integrating it should not be so difficult, try to use [these](https://github.com/micro-ROS/micro-ros-build/issues/137#issuecomment-648007212) links as starting point. 

They have recently upload the POSIX extension to this repo recently: https://github.com/FreeRTOS/Lab-Project-FreeRTOS-POSIX

krm-int, 2020-06-23:
> So, I guess that you would need a POSIX extension to your FreeRTOS environment.
>
> Integrating it should not be so difficult, try to use [...]

Sorry - I didn't see your previous post before

krm-int, 2020-06-24:
Hi @pablogs9 

I've Successfully adde POSIX to my application.
It seems `microxrcedds_client` still includes `udp_transport_posix_nopoll`.

I get the error(s):
```
/workspaces/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/udp/udp_transport_posix_nopoll.c: In function 'uxr_init_udp_platform':
/workspaces/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/udp/udp_transport_posix_nopoll.c:31:25: error: storage size of 'hints' isn't known
         struct addrinfo hints;
                         ^~~~~
...
/workspaces/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/udp/udp_transport_posix_nopoll.c:49:54: error: dereferencing pointer to incomplete type 'struct addrinfo'
             for (ptr = result; ptr != NULL; ptr = ptr->ai_next)
                                                      ^~
```

Maybe its my toolchain.cmake? Or do I need some specific arguments in the .meta?
**Toolchain.cmake:**
```
include(CMakeForceCompiler)

set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR arm)
set(CMAKE_CROSSCOMPILING TRUE)
set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
set(PLATFORM_NAME "LwIP")

set(CMAKE_SYSROOT /workspaces/embedded/dds/../App)

set(tools /usr/bin/arm-none-eabi-)
set(CMAKE_C_COMPILER ${tools}gcc)
set(CMAKE_CXX_COMPILER ${tools}g++)

set(ARCH_CPU_FLAGS "-mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DSTM32F427xx -D_REENT_SMALL -DRXPDO_CALLBACK -DSDO_MAX_SIMULTANEOUS_TRANSFERS=12 -DHAL_IWDG_MODULE_ENABLED  --specs=nano.specs --specs=nosys.specs -Og -fno-inline-small-functions -funwind-tables -Wall -fdata-sections -ffunction-sections -g -gdwarf-2")

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_C_FLAGS "${ARCH_CPU_FLAGS}" CACHE STRING "" FORCE)
#set(CMAKE_EXE_LINKER_FLAGS "-mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -specs=nano.specs -TSTM32F407ZGTx_FLASH.ld  -lc -lm  -Wl,-Map=build/App.map,--cref -Wl,--gc-sections,--no-wchar-size-warning -u _printf_float -Wl,--undefined=uxTopUsedPriority")
set(CMAKE_C_STANDARD_LIBRARIES "-lc -lm -lnosys") #

set(CMAKE_CXX_FLAGS_INIT "-std=c++14 ${ARCH_CPU_FLAGS} " CACHE STRING "" FORCE)

include_directories(SYSTEM 
	    /workspaces/embedded/dds/../App/../dds/Lab-Project-FreeRTOS-POSIX/include
 /workspaces/embedded/dds/../App/../dds/Lab-Project-FreeRTOS-POSIX/include/private
 /workspaces/embedded/dds/../App/../dds/Lab-Project-FreeRTOS-POSIX/FreeRTOS-Plus-POSIX/include
 /workspaces/embedded/dds/../App/../dds/Lab-Project-FreeRTOS-POSIX//FreeRTOS-Plus-POSIX/include/portable
 /workspaces/embedded/dds/../App/../dds/Lab-Project-FreeRTOS-POSIX//FreeRTOS-Plus-POSIX/include/portable/st/stm32l475_discovery
 /workspaces/embedded/dds/../App/../dds/Lab-Project-FreeRTOS-POSIX/include/FreeRTOS-POSIX
 /workspaces/embedded/dds/../App/../dds/Lab-Project-FreeRTOS-POSIX/include/FreeRTOS-POSIX/sys
 /workspaces/embedded/dds/../App/Middlewares/Third_Party/LwIP/src/include/posix
 /workspaces/embedded/dds/../App/Middlewares/Third_Party/LwIP/src/include
 /workspaces/embedded/dds/../App/Middlewares/Third_Party/LwIP/src/include/lwip/priv
 /workspaces/embedded/dds/../App/Middlewares/Third_Party/LwIP/src/include/lwip/prot
 /workspaces/embedded/dds/../App/Middlewares/Third_Party/LwIP/src/include/netif
 /workspaces/embedded/dds/../App/Middlewares/Third_Party/LwIP/src/include/posix
 /workspaces/embedded/dds/../App/Middlewares/Third_Party/LwIP/src/include/posix/sys
 /workspaces/embedded/dds/../App/Middlewares/Third_Party/LwIP/system/arch
 /workspaces/embedded/dds/../App/Inc
 /workspaces/embedded/dds/../App/Drivers/STM32F4xx_HAL_Driver/Inc
 /workspaces/embedded/dds/../App/Drivers/CMSIS/Device/ST/STM32F4xx/Include
 /workspaces/embedded/dds/../App/Drivers/CMSIS/Include
 /workspaces/embedded/dds/../App/Middlewares/Third_Party/LwIP/system

    )
    
set(__BIG_ENDIAN__ 0)
```

pablogs9, 2020-06-24:
Does your LwIP has this: https://github.com/micro-ROS/freertos_apps/blob/8e172698ad9b29eb40270e6d83c8b650d12b63aa/microros_olimex_e407_extensions/Middlewares/Third_Party/LwIP/src/include/lwip/netdb.h#L103?

krm-int, 2020-06-24:
Yes.
and `LWIP_DNS_API_DECLARE_STRUCTS` is defined in the same file, so I don't see why it complains.

Are the includes in my toolchain correct?

pablogs9, 2020-06-24:
This is my toolchain for FreeRTOS + Olimex STM32 E407 + micro-ROS:

```
include(CMakeForceCompiler)

set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR arm)
set(CMAKE_CROSSCOMPILING 1)
set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
set(PLATFORM_NAME "LwIP")

set(CMAKE_SYSROOT /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/../..)

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

# External transports
# set(EXTERNAL_TRANSPORT_HEADER "@EXTERNAL_TRANSPORT_HEADER@" CACHE PATH "" FORCE)
# set(EXTERNAL_TRANSPORT_SRC "@EXTERNAL_TRANSPORT@" CACHE PATH "" FORCE) 

# Makefile flags
set(CROSSDEV /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/../../toolchain/bin/arm-none-eabi-)
set(ARCH_CPU_FLAGS "-mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DSTM32F407xx -D_TIMEVAL_DEFINED -O0 -Wall -fdata-sections -ffunction-sections")
set(ARCH_OPT_FLAGS "")

# Compiler tools
foreach(tool gcc ld ar)
	string(TOUPPER ${tool} TOOL)
    find_program(${TOOL} ${CROSSDEV}${tool})
	if(NOT ${TOOL})
		message(FATAL_ERROR "could not find ${CROSSDEV}${tool}")
	endif()
endforeach()

set(CMAKE_C_COMPILER ${CROSSDEV}gcc)
set(CMAKE_CXX_COMPILER ${CROSSDEV}g++)

set(CMAKE_C_FLAGS_INIT "-std=c11 ${ARCH_CPU_FLAGS} ${ARCH_OPT_FLAGS}" CACHE STRING "" FORCE)
set(CMAKE_CXX_FLAGS_INIT "-std=c++14 ${ARCH_CPU_FLAGS} ${ARCH_OPT_FLAGS} " CACHE STRING "" FORCE)

include_directories(SYSTEM 
	/workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/FreeRTOS-Plus-POSIX/include
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/include
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/include/private
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/include/FreeRTOS_POSIX
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/include/FreeRTOS_POSIX/sys
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/src/hal/interface
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/src/modules/interface
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/src/utils/interface
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/src/config
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/src/drivers/interface
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/Middlewares/Third_Party/LwIP/src/include/posix
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/Middlewares/Third_Party/LwIP/src/include
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/Inc
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/Drivers/STM32F4xx_HAL_Driver/Inc
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/Drivers/CMSIS/Device/ST/STM32F4xx/Include
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/Drivers/CMSIS/Include
 /workspaces/foxy_migration/uros_foxy_freertos_olimex/firmware/freertos_apps/microros_olimex_e407_extensions/Middlewares/Third_Party/LwIP/system

    )
    
set(__BIG_ENDIAN__ 0)
```

krm-int, 2020-06-24:
I'm using the dashing branch on all packages at the moment. Should I switch to the foxy/foxy_migration?

pablogs9, 2020-06-24:
I should work on dashing, foxy is currently under acive development

krm-int, 2020-06-25:
After some debugging at compile time, I've found out that `netdb.h` was not included correctly.
I now get the error:
```
/workspaces/mcu_ws/uros/rcutils/src/time_unix.c: In function 'rcutils_steady_time_now':
/workspaces/mcu_ws/uros/rcutils/src/time_unix.c:103:17: error: 'CLOCK_MONOTONIC' undeclared (first use in this function)
   clock_gettime(CLOCK_MONOTONIC, &timespec_now);
```

Same debugging process told me that colcon/cmake includes the `/usr/local/arm-none-eabi/include/time.h` and NOT `FreeRTOS_POSIX/time.h` in `mcu_ws/uros/rcutils/src/time_unix.c`

Could this have something to with the environment? I don't run any sort of `clean_env.sh` as I'm not sure what this script actually does.

pablogs9, 2020-06-25:
I have `CLOCK_MONOTONIC` defined here: 

https://github.com/micro-ROS/freertos_apps/blob/e0e857942449b786829c20e423a6fa0ab4600221/microros_olimex_e407_extensions/include/FreeRTOS_POSIX/time.h#L56

and here:

https://github.com/micro-ROS/freertos_apps/blob/e0e857942449b786829c20e423a6fa0ab4600221/microros_crazyflie21_extensions/include/FreeRTOS_POSIX/time.h#L56

I'm not sure if it has been added, because I remember adding it at some point...

krm-int, 2020-06-30:
@pablogs9 Friendly ping & update

After some days of struggling I have finally made microROS run on my application.

**My observations/steps:**
- I ditched my own implementation of the library build. I run the 'original' microROS build targeting freeRTOS/olimex_e407 and the Int32_publisher app INSIDE A CONTAINER. To configure the DDS settings I overwrtite the app-colcon.meta at image build time. The toolchain is arm-none-eabi v8.3.1.
- I change the optimization flags in the [Makefile](https://github.com/micro-ROS/freertos_apps/blob/dashing/microros_olimex_e407_extensions/Makefile) to match my own app - THIS WAS VERY IMPORTANT AS I HAD LINKING/ALIGNMENT PROBLEMS!
- I extract/copy both `libmicroros.a` and the `mcu_ws` includes from the docker container and place them in my project.
- I copied the [`libatomic.c`](https://github.com/micro-ROS/freertos_apps/blob/dashing/microros_olimex_e407_extensions/Src/libatomic.c) into my project.
- I added FreeRTOS + POSIX from [The suggested repo](https://github.com/FreeRTOS/Lab-Project-FreeRTOS-POSIX), and included the same directories/sources as the [olimex_e407-extension](https://github.com/micro-ROS/freertos_apps/tree/dashing/microros_olimex_e407_extensions).
- I needed to tell the toolchain to use GNU_SOURCE with: `CFLAGS += -D_GNU_SOURCE` (this might be a total hack/patch!)
- `libmicroros.a`is included in the app build like so: `OBJECTS +=  <path>/libmicroros.a`
- I build my application with arm-none-eabi v6.3.1

I hope this last comment can help others facing similar issues :)
Closing issue.

closed 2020-06-30 by krm-int

pablogs9, 2020-06-30:
Thank you so much for the comments!