micro-ROS/micro_ros_setup #138: Issues building with Nuttx

jayanthendo, 2020-06-22:
Hello! I am trying to run an example on the Olimex-stm32-e407 by following the micro-ROS application tutorial for RTOS and getting some errors while building.
i am doing on ubuntu 18
setup is Dashing Diademata and crystal also
when creating the framework with the following command
ros2 run micro_ros_setup create_firmware_ws.sh nuttx 
the following error appears

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
micro_ros_agent: Cannot locate rosdep definition for [microxrcedds_agent]

i have installed the following package
ros-crystal-micro-xrce-dds-agent
ros-crystal-microxrcedds-agent-cmake-module
ros-crystal-microxrcedds-agent

and installed Micro-XRCE-DDS-Agent using https://micro-xrce-dds.docs.eprosima.com/en/latest/installation.html
but still i am getting the same error

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
micro_ros_agent: Cannot locate rosdep definition for [microxrcedds_agent]

is there any package i am missing? or any other steps to follow?
Thank you very much.

Jayanth kandregula

pablogs9, 2020-06-22:
Hello, try to continue with the tutorial. These rosdep errors can be taken as warning for now. Please check if you micro-ROS workspace builds correctly.

We are working on fixing them.

jayanthendo, 2020-06-23:
yes its build correctly for the linux example and worked properly. But when creating this for stm32f4 nuttx rtos this rosdep dependency error is coming and can't flash to Development board

pablogs9, 2020-06-23:
Could you please copy here the error you are having?

jayanthendo, 2020-06-23:
the same error as mentioned above
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
micro_ros_agent: Cannot locate rosdep definition for [microxrcedds_agent]

pablogs9, 2020-06-23:
This error seems to be in the micro-ROS Agent installation. 

Are you running `ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407` or are you running `ros2 run micro_ros_setup create_agent_ws.sh` ?

jayanthendo, 2020-06-23:
ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407

pablogs9, 2020-06-23:
can you continue with the configuration step ignoring this dependencies errors?

jayanthendo, 2020-06-23:
is it going to work like that? without first creating firmware?

pablogs9, 2020-06-23:
do you have a `firmware` folder with other folders (`dev_ws`, `mcu_ws`, `NuttX`, etc) inside of it?

jayanthendo, 2020-06-23:
no just firmware folder

jayanthendo, 2020-06-23:
when i have created firmware for the host machine then the firmware folder containing dev_ws 
but for the nuttx i have removed firmware folder and creating a new firmware creating the error

pablogs9, 2020-06-23:
Well, on this [docker](https://hub.docker.com/layers/osrf/ros/dashing-desktop/images/sha256-8d8a38945b93e3ba1105e2588f72313ea3effe35d83add6f80a34d69681a748a?context=explore) I have done:

```
sudo apt update
source /opt/ros/dashing/setup.bash
mkdir uros_ws && cd uros_ws
git clone -b dashing https://github.com/micro-ROS/micro-ros-build.git src/micro-ros-build
rosdep update && rosdep install --from-path src --ignore-src -y
colcon build
source install/local_setup.bash 

ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407
```

And I obtain:
```
root@pgarrido:/uros_ws# tree -L 2 .
.
├── build
│   ├── COLCON_IGNORE
│   └── micro_ros_setup
├── firmware
│   ├── COLCON_IGNORE
│   ├── NuttX
│   ├── PLATFORM
│   ├── apps
│   ├── dev_ws
│   ├── drive_base
│   ├── mcu_ws
│   └── uclibc
├── install
│   ├── COLCON_IGNORE
│   ├── _local_setup_util_ps1.py
│   ├── _local_setup_util_sh.py
│   ├── local_setup.bash
│   ├── local_setup.ps1
│   ├── local_setup.sh
│   ├── local_setup.zsh
│   ├── micro_ros_setup
│   ├── setup.bash
│   ├── setup.ps1
│   ├── setup.sh
│   └── setup.zsh
├── log
│   ├── COLCON_IGNORE
│   ├── build_2020-06-23_10-27-01
│   ├── latest -> latest_build
│   └── latest_build -> build_2020-06-23_10-27-01
└── src
    └── micro-ros-build

```

Could you please replicate these steps in a clean ROS 2 Dashing environment?

jayanthendo, 2020-06-23:
ok with clean ROS2 error microxrcedds_agent is gone now
Thank you

but configuring with 
`ros2 run micro_ros_setup configure_firmware.sh uros_pingpong `
gives 
Configuration configs/olimex-stm32-e407/uros_pingpong (expanded from uros_pingpong) not found
and with 
`ros2 run micro_ros_setup configure_firmware.sh pingpong --transport serial --dev 3`
gives 
--- stderr: firmware
CMake Error: The source directory "/home/jayanth/uros_ws/firmware/apps" does not appear to contain CMakeLists.txt.

pablogs9, 2020-06-23:
try with `pingpong-eth` configuration

jayanthendo, 2020-06-23:
with pingpong-eth configuration the same error
```
Starting >>> firmware
--- stderr: firmware                         
CMake Error: The source directory "/home/jayanth/uros_ws/firmware/apps" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
---
Failed   <<< firmware	[ Exited with code 1 ]
```
with drive_base configuration
```
Failed   <<< rclc	[ Exited with code 1 ]
Aborted  <<< shape_msgs                                 
Aborted  <<< trajectory_msgs                            
Aborted  <<< visualization_msgs

Summary: 44 packages finished [1min 27s]
  1 package failed: rclc
  3 packages aborted: shape_msgs trajectory_msgs visualization_msgs
  3 packages had stderr output: rclc rcutils rmw_microxrcedds
  8 packages not processed
```

pablogs9, 2020-06-23:
I'm not sure if the RCLC layer is working currently on dashing. Could you create a file named `COLCON_IGNORE` in `firmware/mcu_ws/uros/rclc` ?

pablogs9, 2020-06-23:
We are fixing all these issues on foxy release, it would be released in a week hopefully

jayanthendo, 2020-06-24:
in crystal also i am getting dependency issues
`ros2 run micro_ros_setup create_agent_ws.sh`
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
microxrcedds_agent_cmake_module: Cannot locate rosdep definition for [microxrcedds_agent]
rosidl_typesupport_fastrtps_cpp: Cannot locate rosdep definition for [fastrtps]
rcl_logging_noop: Cannot locate rosdep definition for [launch_testing]
rmw_implementation: Cannot locate rosdep definition for [poco_vendor]
rosidl_typesupport_fastrtps_c: Cannot locate rosdep definition for [fastrtps]
rcl_action: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
rcl_yaml_param_parser: Cannot locate rosdep definition for [launch_testing]
rcl_logging_log4cxx: Cannot locate rosdep definition for [launch_testing]
rcl_lifecycle: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
rcl: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
rosidl_typesupport_cpp: Cannot locate rosdep definition for [poco_vendor]
rmw_fastrtps_dynamic_cpp: Cannot locate rosdep definition for [fastrtps]
rcutils: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
rmw_fastrtps_shared_cpp: Cannot locate rosdep definition for [fastrtps]
micro_ros_agent: Cannot locate rosdep definition for [fastrtps]
rmw_fastrtps_cpp: Cannot locate rosdep definition for [fastrtps]
rosidl_typesupport_c: Cannot locate rosdep definition for [poco_vendor]

and 
`colcon build`
Summary: 1 package finished [2.52s]
  1 package failed: rcutils
  3 packages aborted: fastrtps_cmake_module rmw_implementation_cmake rosidl_adapter
  1 package had stderr output: rcutils
  52 packages not processed

so is there any working version of micro-ROS present now so i can try it.

pablogs9, 2020-06-24:
please target ROS 2 dashing installation in order to use micro-ROS

jayanthendo, 2020-06-24:
yes i have done that before with ROS 2 dashing and just trying with crystal checking if it is working or not.

can i manually configure the nuttx for dashing and flash the micro-ROS to development board?
so is there any working version of micro-ROS present now, so i can try it?
is there any alternative to micro-ROS that i can use ROS2 on development board?

jayanthendo, 2020-06-24:
can i use "freeRTPS" or "Embedded_nuttx" or "riot_ROS2" to use on development board and ROS2 on linux to communicate ?
which is better?

jayanthendo, 2020-06-25:
@pablogs9 please tell the solution as soon as possible, as i was struggling to make it work from past few days.

pablogs9, 2020-06-25:
Well, on this [docker](https://hub.docker.com/layers/osrf/ros/dashing-desktop/images/sha256-8d8a38945b93e3ba1105e2588f72313ea3effe35d83add6f80a34d69681a748a?context=explore) I have done:

```
sudo apt update
source /opt/ros/dashing/setup.bash
mkdir uros_ws && cd uros_ws
git clone -b dashing https://github.com/micro-ROS/micro-ros-build.git src/micro-ros-build
rosdep update && rosdep install --from-path src --ignore-src -y
colcon build
source install/local_setup.bash 

ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407
ros2 run micro_ros_setup configure_firmware.sh pingpong-eth

# Make sure that you have kconfig-frontends
git clone https://bitbucket.org/nuttx/tools.git firmware/tools
pushd firmware/tools/kconfig-frontends
./configure --enable-mconf --disable-nconf --disable-gconf --disable-qconf 
LD_RUN_PATH=/usr/local/lib && make && sudo make install && sudo ldconfig
popd

# Enable exception support (we will fix this soon)
pushd firmware/Nuttx
make menuconfig
# Go to Library Routines -> Enable Exception Support (under LLVM C++ Library)
# Hit save and exit
popd

ros2 run micro_ros_setup build_firmware.sh

# Now you have a .bin ready to be flashed
```

acassis, 2020-07-01:
Hi Pablo, instead opening a new issue I will post my log here since it appears a similar issue:

$ ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407
Firmware already created. Please delete /home/alan/microros/firmware folder if you want a fresh installation.
alan@dev:~/microros$ rm -rf firmware
alan@dev:~/microros$ ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407
Creating firmware for nuttx platform olimex-stm32-e407
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Add distro "dashing"
Add distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /home/alan/.ros/rosdep/sources.cache
#All required rosdeps installed successfully
/home/alan/microros/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:10: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_info = yaml.load(sys.stdin)
/home/alan/microros/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:11: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_keep = yaml.load(open(sys.argv[1]))['keep'].split()
#All required rosdeps installed successfully
/home/alan/microros/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:10: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_info = yaml.load(sys.stdin)
/home/alan/microros/install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py:11: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  repos_keep = yaml.load(open(sys.argv[1]))['keep'].split()
Crosscompiled environment: cleaning path
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> ament_cppcheck
Starting >>> gtest_vendor
Starting >>> ament_pep8
Finished <<< gtest_vendor [1.10s]                                                
Starting >>> gmock_vendor
Finished <<< gmock_vendor [0.16s]                                                
Finished <<< ament_cppcheck [1.36s]                                              
Finished <<< ament_pep8 [1.36s]
Starting >>> ament_pyflakes
Finished <<< ament_lint [1.64s]                                                  
Finished <<< ament_package [1.66s]
Starting >>> ament_cmake_core
Starting >>> ament_flake8
Finished <<< ament_pyflakes [0.62s]                                                           
Finished <<< ament_cmake_core [0.43s]
Starting >>> ament_cmake_libraries
Starting >>> ament_cmake_export_libraries
Starting >>> ament_cmake_include_directories                                               
Starting >>> ament_cmake_python
Starting >>> ament_cmake_export_definitions
Starting >>> ament_cmake_export_include_directories
Starting >>> ament_cmake_export_link_flags
Finished <<< ament_flake8 [0.73s]                                                                
Starting >>> ament_pep257
Finished <<< ament_cmake_include_directories [0.85s]                       
Finished <<< ament_cmake_python [0.87s]
Finished <<< ament_cmake_export_include_directories [0.86s]
[... 95 lines cut ...]
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
---
Finished <<< nuttx_kconfig_vendor [14.3s]

Summary: 55 packages finished [23.9s]
  1 package had stderr output: nuttx_kconfig_vendor
#All required rosdeps installed successfully

Installing uClibc++ in the NuttX source tree
Installation succeeded

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rcl_logging_noop: No definition of [launch_testing] for OS version [focal]
rcl: No definition of [osrf_testing_tools_cpp] for OS version [focal]
rosidl_default_runtime: No definition of [rosidl_generator_py] for OS version [focal]
rosidl_typesupport_c: No definition of [poco_vendor] for OS version [focal]
rclc: No definition of [osrf_testing_tools_cpp] for OS version [focal]
rcutils: No definition of [osrf_testing_tools_cpp] for OS version [focal]
rosidl_typesupport_cpp: No definition of [poco_vendor] for OS version [focal]
rcl_lifecycle: No definition of [osrf_testing_tools_cpp] for OS version [focal]
rmw_implementation: No definition of [poco_vendor] for OS version [focal]

Do you have some idea why these dependence are reported?

pablogs9, 2020-07-02:
Are you using Ubuntu 18.04?

jayanthendo, 2020-07-02:
@pablogs9  do we have an stable update for ROS2 foxy released?

pablogs9, 2020-07-02:
not yet, the stable version is dashing until next week or so.

jayanthendo, 2020-07-02:
> Well, on this [docker](https://hub.docker.com/layers/osrf/ros/dashing-desktop/images/sha256-8d8a38945b93e3ba1105e2588f72313ea3effe35d83add6f80a34d69681a748a?context=explore) I have done:
>
> ```
> sudo [...]

i have downloaded this docker and yet end up with errors, not able to create firmware in dashing

acassis, 2020-07-02:
> Are you using Ubuntu 18.04?

Hi Pablo,
No, I'm using Ubuntu 20.04. BTW, the compilation of ROS2 happened correctly.
I think there is only some minor issues to get it working correctly. Do you have some idea how to fix it or debug it?

acassis, 2020-07-05:
> i have downloaded this docker and yet end up with errors, not able to create firmware in dashing

Hi @pablogs9 in fact the issue appears related to Ubuntu 20.04 and CMake version. Also there are tons of warning messages during the compilation. I submitted a patch to rmw_microxrcedds_c fixing some of these warnings.

My complete compilation log is here:
https://pastebin.com/0tgV6ZvS

pablogs9, 2020-07-06:
Let us check this. We are not in 20.04 yet.

FranFin, 2020-07-20:
@jayanthendo @acassis Hi! Thank you for your interest in micro-ROS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460). You’re invited to bring your micro-ROS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.

jayanthendo, 2020-07-24:
Thank you @pablogs9  i think this issue is in my local configuration as it was working on other system.

closed 2020-07-24 by jayanthendo

zivy90, 2020-07-24:
Hi, @jayanthendo and @acassis have you solved these issues? I also got similar errors. I am using Dashing to bulid micro_ROS, it appers error as @jayanthendo showed.
When I use Foxy, these errors shows like @acassis's.
@pablogs9 do you have any suggestions? I am using Ubuntu 18.04, but updating to 20.04.

jayanthendo, 2020-07-24:
@zivy90 i still don't know why this error's are coming on my system. i am using lenovo with ubuntu 18 but i have tested on dell system with same ubuntu and its working with same libraries as mine.

pablogs9, 2020-07-24:
We are merging the new Ubuntu 20.04 + micro-ROS Foxy tutorials in this PR: https://github.com/micro-ROS/micro-ROS.github.io/pull/187

I would suggest that you try the new tutorials + Foxy release when they are merged.

Sorry for the inconvenience.

pablogs9, 2020-07-24:
Hello again! Please check this new tutorials for micro-ROS Foxy + Ubuntu 20.04 + Nuttx + Olimex: https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/nuttx/

I would also suggest you use a docker container using ros:foxy image.

zivy90, 2020-07-24:
@pablogs9 Thanks for the help!

zivy90, 2020-07-24:
@jayanthendo, @pablogs9 Thanks for the reply, I am going to try docker container. Thanks again for the help!

zivy90, 2020-07-25:
Hi, @pablogs9, one more question, I am using container, but after` ros2 run micro_ros_setup configure_firmware.sh pingpong-eth`
it got an error:

```
cd /microros_ws/firmware/apps/; colcon build --cmake-target clean --cmake-target-skip-unavailable
Starting >>> firmware
--- stderr: firmware
CMake Error: The source directory "/microros_ws/firmware/apps" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
---
Failed   <<< firmware [0.07s, exited with code 1]
```
Do you have any idea? Thank you!

jayanthendo, 2020-07-27:
@zivy90  which system are you using?
its better to switch system and try because i also got the same problem using normal **ubuntu and docker** also, when i switched to dell system its not showing these errors
i don't know why these error are causing

zivy90, 2020-07-27:
> @zivy90 which system are you using?
> its better to switch system and try because i also got the same problem using normal **ubuntu and docker** also, when i switched to dell system its not showing these errors
> i don't know why these error are causing

@jamoralp I am using Ubuntu 20.04, my computer is Dell XPS. It is weird, I used Ubuntu 18.04 before, and showed these errors, while I upgraded the system to 20.04, same errors showed again.
BTW, dell system means using dell computer? Thank you!

jayanthendo, 2020-07-27:
i am using lenovo, its weird it's working on some system's and but not on other systems