micro-ROS/micro_ros_setup #141: 6lowPAN issue: likely to be some fragmentation issue

amx-piap, 2020-07-14:
Hi,
I am running in an issue when I am trying to make a micro-ROS agent udpv6 and micro-ROS client  communicate with each other.

# Hardware setup

The set-up is as follow:
 * The micro-ROS udpv6 is running on Linux within a docker. Git version is pointing to branch **dashing**
 * The micro-ROS client is running on NuttX on Olimex-E430. Git version was point to branch **dashing** and after some talk with the eProsima the version was updated to version tag **v1.2.5** .
 * The PC is using a atusb
 * The Olimex is connected with a mrf24 from diligent PMODRF2.

# The configuration for Olimex in Docker

The MCU code is compiled and download after the documentation here https://github.com/micro-ROS/docker#micro-ros-olimex-nuttx.
Once in the docker it will be necessary to append **"-DRMW_UXRCE_IPV=ipv6"** to the /uros_ws/firmware/apps/uros/rmw_config.meta.in . 

Then following the tutorial https://micro-ros.github.io/docs/tutorials/advanced/nuttx/6lowpan/ up to the **Software set-up** section.

 # The configuration on PC
Command to type on the pc are the following.

```
    sudo iwpan dev wpan0 set pan_id 0xabcd
    sudo iwpan phy phy0 set channel 0 26

    sudo ip link add link wpan0 name lowpan0 type lowpan
    sudo ip link set wpan0 up
    sudo ip link set lowpan0 up
    sudo ip -6 route delete fe80::/64 dev wlan0 proto kernel metric 256 pref medium # Choose deleve all but the lowpan0 if
    sudo ip -6 route delete fe80::/64 dev eth0 proto kernel metric 256 pref medium # Choose deleve all but the lowpan0 if
    sudo ip neigh add fe80::2be:adde : de:fa00 dev all lladdr 00:be:ad: de:00: de:fa:00
    docker run -it --net=host microros/micro-ros-agent:dashing udp6 --port 9999
```

# Running on PC

The following shows the network configuration:
```
$ ip addr
lowpan0@wpan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
    link/6lowpan 10:e2:d5:ff:ff:00:02:a3 brd ff:ff:ff:ff:ff:ff:ff:ff
    inet6 fe80::12e2:d5ff:ff00:2a3/64 scope link 
wpan0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 127 qdisc fq_codel state UNKNOWN group default qlen 300
    link/ieee802.15.4 10:e2:d5:ff:ff:00:02:a3 brd ff:ff:ff:ff:ff:ff:ff:ff

```
Running the agent:
```
 docker run -it --net=host microros/micro-ros-agent:dashing udp6 --port 9999
```
# On the Olimex via Serial

```
uros_6lowpan fe80::12e2:d5ff:ff00:2a3 9999 pub
```

 Then type "y" to configure it, and type any key to start it.

```
uros_6lowpan fe80::12e2:d5ff:ff00:2a3 9999 pub
Do you want to configure the 6lowpan network? (y/n)
ifdown wpan0...OK
i8sak: resetting MAC layer
i8sak: starting PAN
i8sak: accepting all assoc requests
i8sak: daemon started
ifup wpan0...OK
Connection data
wpan0   Link encap:6LoWPAN HWaddr 00:be:ad: de:00:de:fa:00 at UP
        inet6 addr: fe80::2be:adde : de :fa00/64
        inet6 DRaddr: ::/64

        RX: Received Fragment Errors
            00000000 00000000 00000000
            IPv6     Dropped
            00000000 00000000
        TX: Queued   Sent     Errors   Timeouts
            00000000 00000000 00000000 00000000
        Total Errors: 00000000

Press any key to continue
UDP mode => ip: fe80::12e2:d5ff:ff00:2a3 - port: 9999
micro-ROS Publisher
```

# Attached files

Attached to this ticket the wireshark capture:
 * One from the wpan0
 * One from the lowpan0
 * One from the all interfaces (we can see malformed packet).

And the output of the agent on the PC.

[LOGS.zip](https://github.com/micro-ROS/micro-ros-build/files/4916817/LOGS.zip)

walkiry, 2020-07-16:
After investigating the issues, it looks like the NuttX is not correctly managing packet fragmentation. But still there is a remaining issue related to the memory footprint of the software. Need to investigate.

amx-piap, 2020-07-22:
Pull request was done in this regards.
However, still the issue persisted. To finally solve the issue the CONFIG_UDP_TRANSPORT_MTU had to be reduced to the value 450B instead of 512B

closed 2020-09-08 by amx-piap