micro-ROS/micro_ros_setup #142: nuttx olimex-stm32-h407 Build using serial protocol

thihakyawjob, 2020-07-14:
Hi,

I cannot build it for olimex-stm32-h407. 

https://micro-ros.github.io/docs/tutorials/advanced/nuttx/add_microros_config/

Error is shown below;

[image]

The error is due to following settings.
Application Configuration > micro-ROS > Transport > Serial Transport
Application Configuration > Examples > microROS Publisher

[image]

How can I solve the issue?

Please advise me.

Thank you.

Regards,
Thiha Kyaw

thihakyawjob, 2020-07-15:
Dear Pablo,
Thank you assigning person to help me. If you want me to test, let me know. I will quickly test it and give you results.
If you want to know how I test. I can give you the detailed test procedure also.

I would like to make the system work for my project as well.

Regards,
Thiha Kyaw

thihakyawjob, 2020-07-15:
Dear All,

I retest using olimex-stm32-e407 board with serial protocol. I got the same kind of error. Therefore I assume the the serial protocol setting inside Nuttx  does have problem. Thank you.

[image]

In olimex-stm32-e407, the serial port is used for testing while transport setting is UDP protocol. If I want to use Ethernet connection for testing, please advise me how to set Ethernet for communication.

Thanks.

Regards,
Thiha Kyaw

FranFin, 2020-07-20:
@thihakyawjob Hi! Thank you for your interest in micro-ROS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460). You’re invited to bring your micro-ROS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.

thihakyawjob, 2020-07-20:
> @thihakyawjob Hi! Thank you for your interest in micro-ROS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG [...]

Dear FranFin,

I will join the meeting.

Thanks.

Regards,
Thiha Kyaw

jamoralp, 2020-07-22:
Good morning @thihakyawjob,
Good news! We fixed this issue during the release process of micro-ROS Foxy. You can find further information about the actual fix in [this commit](https://github.com/micro-ROS/rmw-microxrcedds/commit/4182dcfba5c20adca0ee2b12b3e396fec28da180).
Remember to switch to branch `foxy` before testing, as it is not the default branch yet.
At the precise moment of writing this message, the [micro-ros-build](https://github.com/micro-ROS/micro-ros-build) repository migration process to Foxy has not been completed yet, due to some errors for Raspbian build system which we are still fixing. Please check [this pull-request](https://github.com/micro-ROS/micro-ros-build/pull/148) and, if it has not been merged yet, switch to branch [feature/foxy_migration](https://github.com/micro-ROS/micro-ros-build/tree/feature/foxy_migration) instead.
Thanks!

closed 2020-08-02 by thihakyawjob