micro-ROS/micro_ros_setup #144: MicroROS with Nuttx Shell?

CLansdaleBEC, 2020-07-20:
Hello there. I would like to run micro-ROS on the STM32H743zi, and as a first step I am trying to build with the micro-ros-build scripts and access NSH, with the idea that from there I will try to get the uros examples working.

I have a working build **without** microROS enabled from official nuttx repo to confirm my serial transport is working. I then confirmed the same settings in ```make menuconfig``` on the new build. My issue is when I build with microROS scripts I cannot access the nuttx shell, is there a step I am missing? 

Environment: 
Ubuntu 18.04
ROS2 Dashing

Steps taken:
1. Followed create/configure/build steps from [First application on RTOS  ](https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/) for nucleo-h743zi and nuttx. no issues here
2. Flashed with openocd.
3. Read output with picocom

However minicom can't seem to open up nsh. Am I missing anything here? Again my idea is to get nsh up and running and then move on to enabling uros through the tutorials on the main page. Any help or advice would be greatly appreciated. 

Thanks
Connor

Aiven92, 2020-07-22:
If you want to use NSH on serial simultaneous with XRCE, you need extra serial(/dev/ttyX).
For my cases(odrive board) I use NSH over USART( cause it allow to print asserts) and usb serial for XRCE.
Can you attach you .config file ?

CLansdaleBEC, 2020-07-22:
I have XRCE disabled right now to test _only_ the nuttx shell while using micro-ROS build tools, so I should be able to access NSH with /dev/ttyACM0 correct? Do I need to have a separate serial connection even if I'm not using XRCE? 

.config file here:
[config.txt](https://github.com/micro-ROS/micro-ros-build/files/4960328/config.txt)

pablogs9, 2022-01-28:
Any update on this?

gseqBE, 2022-06-15:
This can be closed. We have internally moved to Zephyr.