micro-ROS/micro_ros_setup #154: Query on need for MPC57xx

rpcme, 2020-07-22:
In the RTOS comparison, there is a lineitem for MPC57xx support where none of the RTOS are supported.

It is my understanding that NXP provides FreeRTOS but the port is not upstream.  Is this adequate to change the status or no?  Does the port need to be upstream?

Also, would be interested to know the background on the requirement.

pablogs9, 2020-07-28:
Hello @rpcme, after our conversation yesterday I'm going to answers your questions here:
- The lineitem on our webpage I guess that should be removed because this table is inherited from another documentation, maybe this MCU was a possible requirement at the beginning of the project.
- I'm not sure what is the minimum version of  FreeRTOS in order to work with micro-ROS. As I told you, we are using the version provided by STM32CubeMX in our reference platform. But, for example, we also support Crazyflie 2.1 Drone, which uses a newer version (v10+).
- In order to run the micro-ROS middleware (Micro XRCE-DDS) it is only required to have `clock_gettime` available. Defaults transports (UDP/TCP/Serial) are written on top of POSIX, but we have successfully implemented "custom transports" on top of other APIs. 
- In order to run the whole micro-ROS stack, some level of POSIX compliance is required. In the case of FreeRTOS, we are using the FreeRTOS Labs POSIX extension.

pablogs9, 2021-12-17:
Closing due to inactivity. Reopen if needed.

closed 2021-12-17 by pablogs9