micro-ROS/micro_ros_setup #160: Cannot make micro-ROS with 'stderr: nuttx_kconfig_vendor'

zivy90, 2020-07-23:
Hi guys,

When I build micro-ROS using the [tutorial ](https://github.com/micro-ROS/micro-ros-build/blob/foxy/micro_ros_setup/README.md) , the branch is foxy.

After ` ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407` ，
terminal shows
`--- stderr: nuttx_kconfig_vendor                                 
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
---
` 

And run `ros2 run micro_ros_setup configure_firmware.sh drive_base`,
terminal shows: 
`Configuring firmware for nuttx platform olimex-stm32-e407
Configuration configs/olimex-stm32-e407/drive_base (expanded from drive_base) not found
`
I checked, there is no 'configs/'

BTW, my system is Ubuntu 18.04, the kernel is 4.15.0

Ang help is appreciated.

pablogs9, 2020-07-23:
Sorry, we are updating tutorials since the Foxy release was yesterday. Please check the website when this PR is merged: https://github.com/micro-ROS/micro-ROS.github.io/pull/187

Sorry for the inconvenience.

@FranFin  please make sure we remember to ping this issue when Foxy documentation is merged

zivy90, 2020-07-23:
> Sorry, we are updating tutorials since the Foxy release was yesterday. Please check the website when this PR is merged: [micro-ROS/micro-ROS.github.io#187](https://github.com/micro-ROS/micro-ROS.github.io/pull/187)
> 
> Sorry for the inconvenience.
> 
> @FranFin please make sure we remember to ping this issue when Foxy documentation is merged

Thanks for the reply, I will try that!

pablogs9, 2020-07-24:
Hello again! Please check this new tutorials for micro-ROS Foxy + Ubuntu 20.04 + Nuttx + Olimex: https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/nuttx/

closed 2022-01-28 by pablogs9