micro-ROS/micro_ros_setup #161: Using build scripts a different board

CLansdaleBEC, 2020-07-28:
Beginner question: to what extent (if any) can the micro_ros_build scripts be used to configure the firmware for a different board? 

I noticed that the Zephyr dir has a 'generic' directory that I would think is for this purpose. What is the high level strategy for porting Zephyr to a new board? 

Thank you

pablogs9, 2020-07-28:
Which board are you planning to use? Which transport do you want to use?

The high level strategy is something like:
 - First of all you should add your board name  [here](https://github.com/micro-ROS/micro-ros-build/blob/foxy/micro_ros_setup/config/zephyr/generic/supported_platforms). 
- Make sure you add your platform [here](https://github.com/micro-ROS/micro-ros-build/blob/722d0361f9289323d107fd788952c783c9f2e2ba/micro_ros_setup/config/zephyr/generic/build.sh#L33) too. This translates from the "micro-ROS board name" to the "Zephyr board name".
- Then, do the `colcon build` for the build system.
- You should be able now to run `ros2 run micro_ros_setup create_firmware_ws.sh zephyr [your platform].
- For example for porting the int32 publisher, you should add your board name (the one that uses Zephyr) to the app CMake file [here](https://github.com/micro-ROS/zephyr_apps/blob/d361d4c5dd8207a5b6875acc2a75811582259486/apps/int32_publisher/CMakeLists.txt#L5).
- Then you have to be sure that the serial transport will work on your platform. This means that [this configuration](https://github.com/micro-ROS/zephyr_apps/blob/foxy/apps/int32_publisher/serial.conf) should work in your platform.
- As you can see, the Zephyr serial transport is pretty generic: https://github.com/micro-ROS/zephyr_apps/blob/foxy/microros_extensions/zephyr_serial_transport.c so it should be so difficult to port it to a new platform.

If you give me more details about your setup, we can go thought this issue step by step. And if you success with the porting do not hesitate to pull request your platform support to the micro-ROS community.

CLansdaleBEC, 2020-07-29:
I am using the nucleo-h743zi with serial transport.

It looks like the nucleo_h743zi board directory is not included in the current uros Zephyr folder, but it **is** included in the latest version of Zephyr as of [last month](https://github.com/zephyrproject-rtos/zephyr/tree/master/boards/arm). 

Any tips for bringing this board over to the microROS world? 

Also, yes when I get this board up and running will certainly submit a PR.

pablogs9, 2020-07-29:
Please try:

```
source /opt/ros/$ROS_DISTRO/setup.bash
mkdir uros_ws && cd uros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro-ros-build.git src/micro-ros-build
rosdep update && rosdep install --from-path src --ignore-src -y
colcon build
source install/local_setup.bash
ros2 run micro_ros_setup create_firmware_ws.sh zephyr nucleo_h743zi
```

At this point make sure that `firmware/zephyr_apps/apps/int32_publisher/CMakeLists.txt` has your board name in line 5, for example: `set(COMPATIBLE_BOARDS disco_l475_iot1 olimex_stm32_e407 native_posix nucleo_h743zi)`.

```
ros2 run micro_ros_setup configure_firmware.sh int32_publisher -t serial -d 3
ros2 run micro_ros_setup build_firmware.sh
```

I have tested this until this point. I can't guarantee that the flashing step is going to work but you can try:

```
ros2 run micro_ros_setup flash_firmware.sh

# Agent installation
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
source install/local_setup.sh
ros2 run micro_ros_agent micro_ros_agent serial --dev [serial device] -v6
```

CLansdaleBEC, 2020-07-29:
That worked! I flashed the board with 
```~/.platformio/packages/tool-openocd/bin/openocd -f ~/.platformio/packages/tool-openocd/scripts/interface/stlink.cfg -f ~/.platformio/packages/tool-openocd/scripts/target/stm32h7x.cfg -c init -c "reset halt" -c "flash write_image erase zephyr.bin 0x08000000" -c "exit" ```
Thanks Pablo!

pablogs9, 2020-07-30:
Hello @CLansdaleBEC, does the flash_firmware.sh script works?

Have you tested the micro-ROS communication between the client and the agent? Is your board publishing?

If everything is ok, please close the issue.

Thanks!

CLansdaleBEC, 2020-07-30:
flash_firmware.sh didn't work for me. The h743zi uses stlink, and it doesn't look like zephyr/generic/flash.sh supports it currently.

I tested with the agent and could see the board publishing. 

Thanks again

closed 2020-07-30 by CLansdaleBEC

pablogs9, 2020-07-30:
If you make some progress with the flash script to support your board we would be happy to include it! 

Enjoy micro-ROS!