micro-ROS/micro_ros_setup #167: Error in Flashing firmware on STM32 Discovery Board

kashishdhal, 2020-08-03:
Hello @pablogs9 ,

I followed your instructions as in #121 and I was able to build the firmware. However, while flashing it gave me an error. 

These are my connections:
[image]
[image]

And here is the output, please help:

<pre><font color="#4E9A06"><b>gnclab@gnclab</b></font>:<font color="#3465A4"><b>~/uros_ws</b></font>$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for zephyr platform discovery_l475_iot1
<font color="#8AE234">-- west flash: rebuilding</font>
/usr/bin/cmake -S/home/gnclab/uros_ws/firmware/zephyr_apps/apps/int32_publisher -B/home/gnclab/uros_ws/firmware/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/make -f CMakeFiles/Makefile2 west_flash_depends
make[1]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
/usr/bin/cmake -S/home/gnclab/uros_ws/firmware/zephyr_apps/apps/int32_publisher -B/home/gnclab/uros_ws/firmware/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/gnclab/uros_ws/firmware/build/CMakeFiles 97
/usr/bin/make -f CMakeFiles/Makefile2 zephyr/cmake/flash/CMakeFiles/west_flash_depends.dir/all
make[2]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
/usr/bin/make -f zephyr/CMakeFiles/parse_syscalls_target.dir/build.make zephyr/CMakeFiles/parse_syscalls_target.dir/depend
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
cd /home/gnclab/uros_ws/firmware/build &amp;&amp; /usr/bin/cmake -E cmake_depends &quot;Unix Makefiles&quot; /home/gnclab/uros_ws/firmware/zephyr_apps/apps/int32_publisher /home/gnclab/uros_ws/firmware/zephyrproject/zephyr /home/gnclab/uros_ws/firmware/build /home/gnclab/uros_ws/firmware/build/zephyr /home/gnclab/uros_ws/firmware/build/zephyr/CMakeFiles/parse_syscalls_target.dir/DependInfo.cmake --color=
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
/usr/bin/make -f zephyr/CMakeFiles/parse_syscalls_target.dir/build.make zephyr/CMakeFiles/parse_syscalls_target.dir/build
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
make[3]: Nothing to be done for &apos;zephyr/CMakeFiles/parse_syscalls_target.dir/build&apos;.
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
[  2%] Built target parse_syscalls_target
/usr/bin/make -f zephyr/CMakeFiles/kobj_types_h_target.dir/build.make zephyr/CMakeFiles/kobj_types_h_target.dir/depend
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
cd /home/gnclab/uros_ws/firmware/build &amp;&amp; /usr/bin/cmake -E cmake_depends &quot;Unix Makefiles&quot; /home/gnclab/uros_ws/firmware/zephyr_apps/apps/int32_publisher /home/gnclab/uros_ws/firmware/zephyrproject/zephyr /home/gnclab/uros_ws/firmware/build /home/gnclab/uros_ws/firmware/build/zephyr /home/gnclab/uros_ws/firmware/build/zephyr/CMakeFiles/kobj_types_h_target.dir/DependInfo.cmake --color=
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
/usr/bin/make -f zephyr/CMakeFiles/kobj_types_h_target.dir/build.make zephyr/CMakeFiles/kobj_types_h_target.dir/build
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
make[3]: Nothing to be done for &apos;zephyr/CMakeFiles/kobj_types_h_target.dir/build&apos;.
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
[  3%] Built target kobj_types_h_target
/usr/bin/make -f zephyr/CMakeFiles/syscall_list_h_target.dir/build.make zephyr/CMakeFiles/syscall_list_h_target.dir/depend
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
cd /home/gnclab/uros_ws/firmware/build &amp;&amp; /usr/bin/cmake -E cmake_depends &quot;Unix Makefiles&quot; /home/gnclab/uros_ws/firmware/zephyr_apps/apps/int32_publisher /home/gnclab/uros_ws/firmware/zephyrproject/zephyr /home/gnclab/uros_ws/firmware/build /home/gnclab/uros_ws/firmware/build/zephyr /home/gnclab/uros_ws/firmware/build/zephyr/CMakeFiles/syscall_list_h_target.dir/DependInfo.cmake --color=
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
/usr/bin/make -f zephyr/CMakeFiles/syscall_list_h_target.dir/build.make zephyr/CMakeFiles/syscall_list_h_target.dir/build
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
make[3]: Nothing to be done for &apos;zephyr/CMakeFiles/syscall_list_h_target.dir/build&apos;.
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
[  5%] Built target syscall_list_h_target
/usr/bin/make -f zephyr/CMakeFiles/driver_validation_h_target.dir/build.make zephyr/CMakeFiles/driver_validation_h_target.dir/depend
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
cd /home/gnclab/uros_ws/firmware/build &amp;&amp; /usr/bin/cmake -E cmake_depends &quot;Unix Makefiles&quot; /home/gnclab/uros_ws/firmware/zephyr_apps/apps/int32_publisher /home/gnclab/uros_ws/firmware/zephyrproject/zephyr /home/gnclab/uros_ws/firmware/build /home/gnclab/uros_ws/firmware/build/zephyr /home/gnclab/uros_ws/firmware/build/zephyr/CMakeFiles/driver_validation_h_target.dir/DependInfo.cmake --color=
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
/usr/bin/make -f zephyr/CMakeFiles/driver_validation_h_target.dir/build.make zephyr/CMakeFiles/driver_validation_h_target.dir/build
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
make[3]: Nothing to be done for &apos;zephyr/CMakeFiles/driver_validation_h_target.dir/build&apos;.
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
[  7%] Built target driver_validation_h_target
/usr/bin/make -f zephyr/CMakeFiles/offsets.dir/build.make zephyr/CMakeFiles/offsets.dir/depend
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
cd /home/gnclab/uros_ws/firmware/build &amp;&amp; /usr/bin/cmake -E cmake_depends &quot;Unix Makefiles&quot; /home/gnclab/uros_ws/firmware/zephyr_apps/apps/int32_publisher /home/gnclab/uros_ws/firmware/zephyrproject/zephyr /home/gnclab/uros_ws/firmware/build /home/gnclab/uros_ws/firmware/build/zephyr /home/gnclab/uros_ws/firmware/build/zephyr/CMakeFiles/offsets.dir/DependInfo.cmake --color=
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
/usr/bin/make -f zephyr/CMakeFiles/offsets.dir/build.make zephyr/CMakeFiles/offsets.dir/build
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
make[3]: Nothing to be done for &apos;zephyr/CMakeFiles/offsets.dir/build&apos;.
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
[  7%] Built target offsets
/usr/bin/make -f zephyr/CMakeFiles/offsets_h.dir/build.make zephyr/CMakeFiles/offsets_h.dir/depend
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
cd /home/gnclab/uros_ws/firmware/build &amp;&amp; /usr/bin/cmake -E cmake_depends &quot;Unix Makefiles&quot; /home/gnclab/uros_ws/firmware/zephyr_apps/apps/int32_publisher /home/gnclab/uros_ws/firmware/zephyrproject/zephyr /home/gnclab/uros_ws/firmware/build /home/gnclab/uros_ws/firmware/build/zephyr /home/gnclab/uros_ws/firmware/build/zephyr/CMakeFiles/offsets_h.dir/DependInfo.cmake --color=
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
/usr/bin/make -f zephyr/CMakeFiles/offsets_h.dir/build.make zephyr/CMakeFiles/offsets_h.dir/build
[... 186 lines cut ...]
cd /home/gnclab/uros_ws/firmware/build &amp;&amp; /usr/bin/cmake -E cmake_depends &quot;Unix Makefiles&quot; /home/gnclab/uros_ws/firmware/zephyr_apps/apps/int32_publisher /home/gnclab/uros_ws/firmware/zephyrproject/zephyr/cmake/flash /home/gnclab/uros_ws/firmware/build /home/gnclab/uros_ws/firmware/build/zephyr/cmake/flash /home/gnclab/uros_ws/firmware/build/zephyr/cmake/flash/CMakeFiles/west_flash_depends.dir/DependInfo.cmake --color=
make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
/usr/bin/make -f zephyr/cmake/flash/CMakeFiles/west_flash_depends.dir/build.make zephyr/cmake/flash/CMakeFiles/west_flash_depends.dir/build
make[3]: Entering directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
cd /home/gnclab/uros_ws/firmware/build/zephyr/cmake/flash &amp;&amp; /usr/bin/cmake -E echo 

make[3]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
[100%] Built target west_flash_depends
make[2]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
/usr/bin/cmake -E cmake_progress_start /home/gnclab/uros_ws/firmware/build/CMakeFiles 0
[... cut ...] rt -s /home/gnclab/uros_ws/firmware/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/gnclab/uros_ws/firmware/zephyrproject/zephyr/boards/arm/disco_l475_iot1/support/openocd.cfg -c init -c targets -c &apos;reset halt&apos; -c &apos;flash write_image erase /home/gnclab/uros_ws/firmware/build/zephyr/zephyr.hex&apos; -c &apos;reset halt&apos; -c &apos;verify_image /home/gnclab/uros_ws/firmware/build/zephyr/zephyr.hex&apos; -c &apos;reset run&apos; -c shutdown</font>
</pre>

kashishdhal, 2020-08-03:
Well I guess, it could be because I am running Ubuntu in a virtual box and I am not able to access usb port on it.

pablogs9, 2020-08-04:
Yes, it seems like your USB port is not been found. Try using Ubuntu in your local machine or using Docker.

kashishdhal, 2020-08-04:
I have Ubuntu 18.04 in my local machine and Foxy needs 20.04, that is why I was using a virtual box. Is it possible to flash the firmware from Ubuntu 18.04 which I already built-in 20.04, if I copy the firmware folder? or it has to be the same version?

kashishdhal, 2020-08-04:
Also, I tried using one of my old computers and installed Ubuntu 20 on it. However, I am getting an error at the step below which I was not getting in the other computer. I tried searching this issue online and implemented all possible solutions presented but I had no success. Afterwards, I formatted the computer and installed Ubuntu again and got the same error. So, is this something related to your scripts or there is a problem in my computer? Note that everything else works fine, I installed ROS2, pip, colon, etc. I am not sure how to get rid of this error. Please see if you can help. Otherwise I will have to do it on a different computer as it was working before on other computer.

<pre><font color="#4E9A06"><b>kashish@kashish</b></font>:<font color="#3465A4"><b>~/uros_ws</b></font>$ ros2 run micro_ros_setup create_firmware_ws.sh zephyr discovery_l475_iot1
Creating firmware for zephyr platform discovery_l475_iot1
Reading package lists... Done
<font color="#CC0000"><b>E: </b></font>Could not open lock file /var/lib/apt/lists/lock - open (13: Permission denied)
<font color="#CC0000"><b>E: </b></font>Unable to lock directory /var/lib/apt/lists/
<font color="#C4A000"><b>W: </b></font>Problem unlinking the file /var/cache/apt/pkgcache.bin - RemoveCaches (13: Permission denied)
<font color="#C4A000"><b>W: </b></font>Problem unlinking the file /var/cache/apt/srcpkgcache.bin - RemoveCaches (13: Permission denied)
</pre>

kashishdhal, 2020-08-05:
So I was able to fix the USB problem with the Virtual Box and was able to flash the firmware to the board.

Here is the output of the micro_ros_agent:
<pre><font color="#4E9A06"><b>gnclab@gnclab</b></font>:<font color="#3465A4"><b>~/uros_ws</b></font>$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_0671FF3134354D5043075629-if02 -v6
Press CTRL+C to exit
<font color="#75507B">[1596589044.582841]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">TermiosAgentLinux.cpp</font> | <font color="#D3D7CF">init                    </font> | <font color="#4E9A06">running...</font>             | fd: 3
<font color="#75507B">[1596589044.582998]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">Root.cpp          </font> | <font color="#D3D7CF">set_verbose_level       </font> | <font color="#4E9A06">logger setup</font>           | verbose_level: 6
</pre>

Now, how should I test the micro-ros app installed on the board? Do I need to change the jumper connection on the back of the board after the flash process? When I type ros2 topic list, I don't see anything any topics.

pablogs9, 2020-08-05:
Which transport did you choose in the configuration step? Can you paste here the commands you follow to create, configure, build and flash you board?

kashishdhal, 2020-08-05:
I used the same commands you pasted in #121 in the last comment.

kashishdhal, 2020-08-05:
source /opt/ros/$ROS_DISTRO/setup.bash
mkdir uros_ws && cd uros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro-ros-build.git src/micro-ros-build
rosdep update && rosdep install --from-path src --ignore-src -y
colcon build
source install/local_setup.bash
ros2 run micro_ros_setup create_firmware_ws.sh zephyr discovery_l475_iot1
ros2 run micro_ros_setup configure_firmware.sh int32_publisher -t serial-usb
ros2 run micro_ros_setup build_firmware.sh
# Now set the power supply jumper to the ST-link selector, and connect the board using a micro-USB to the ST Link port
ros2 run micro_ros_setup flash_firmware.sh
# Now set the power supply jumper to the device USB selector, and connect the board using a micro-USB to the device USB

# Agent installation
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
source install/local_setup.sh
ros2 run micro_ros_agent micro_ros_agent serial --dev [serial device] -v6

pablogs9, 2020-08-05:
ok, so you have to connect your board using the USB OTG (the unconnected in your images) and you have to change the power jumper to `5V_USB_FS`.

Now, when you do a `ls /dev/serial/by-id/`, you should see a serial device with a zephyr label and this is the serial device that you have to choose in the agent.

kashishdhal, 2020-08-05:
Ok, I don't have a USB OTG right now, I will have to order it. So a serial connection like what we did with Olimex Board (a wire connected to tx rx of board) will not work?

pablogs9, 2020-08-05:
You dont need to have an USB OTG cable, use the same cable as you used for flashing. It works for me.

kashishdhal, 2020-08-05:
Well that is a micro-usb cable, they have different size.

kashishdhal, 2020-08-05:
I will try if that works and update you tomorrow, thanks!

pablogs9, 2020-08-05:
For us, it works with a micro-USB cable. 
Please tell me tomorrow if everything works!

kashishdhal, 2020-08-05:
Thanks @pablogs9 , it worked!

So, now how do I run Wifi example? I mean how will the board connect to the wifi?

closed 2020-08-05 by kashishdhal

pablogs9, 2020-08-06:
Wifi is experimental in this board, I have not completely polish the Wifi transport so it works but not as smoothly as expected. You can try with this configuration step: `configure_firmware.sh int32_wifi_publisher -t udp -i [your agent IP] -p [your agent port]`

Remember to set your SSID and PSK in the source code before building: https://github.com/micro-ROS/zephyr_apps/blob/b37c48ff2f0d8a937b9c27a66962c977baaad67a/apps/int32_wifi_publisher/src/main.c#L58

In any case, if you want to use wireless transports I strongly recommend you to buy the new supported ESP32. This super cheap board works really well with wifi, serial and serial-usb transports and FreeRTOS: https://www.mouser.es/ProductDetail/Espressif-Systems/ESP32-DevKitC-32E?qs=GedFDFLaBXFpgD0kAZWDrQ%3D%3D

kashishdhal, 2020-08-06:
Thanks for the reply!

In the configure_firmware step you posted above, IP should be the IP address of the computer in which the agent is running, and what should be the agent port?

I can certainly buy another board but I don't think the given board is as powerful as ST Discovery board. In the end, I want to put MCU on a robot and want to work it wirelessly. I saw that ST board has other sensors as well so I guess I will have to connect the two MCU together in the end. But I worry about that later, the first goal is to make it work wirelessly.

But in any case, have you also tried working with the RF module (sub-GHz module) installed on the ST board.  I guess there are also NRF24L01+(2.5 GHz) modules available very cheap these days, have you tried integrating those?

pablogs9, 2020-08-07:
The IP is the IP of the computer, right. The port is the port where the agent is listening, remember that the command for running the agent in UDP mode is: `ros2 run micro_ros_agent micro_ros_agent udp4 --port [port]`

kashishdhal, 2020-08-09:
@pablogs9 Do I need additional CMakeLists.txt or is there any other error?

<pre><font color="#4E9A06"><b>gnclab@gnclab</b></font>:<font color="#3465A4"><b>~/uros_ws</b></font>$ ros2 run micro_ros_setup configure_firmware.sh int32_wifi_publisher -t udp -i 10.0.2.15 -p 8888
Configuring firmware for zephyr platform discovery_l475_iot1
Configured udp mode with agent at 10.0.2.15:8888
<font color="#4E9A06"><b>gnclab@gnclab</b></font>:<font color="#3465A4"><b>~/uros_ws</b></font>$ ros2 run micro_ros_setup build_firmware.sh
Crosscompiled environment: cleaning path
Building firmware for zephyr platform discovery_l475_iot1
Selected app: int32_wifi_publisher
Specific config for transport not found. Using prj.conf.
<font color="#EF2929">ERROR: /home/gnclab/uros_ws/firmware/zephyr_apps/apps/int32_wifi_publisher doesn&apos;t contain a CMakeLists.txt</font>
<font color="#EF2929">FATAL ERROR: refusing to proceed without --force due to above error</font>
</pre>

pablogs9, 2020-08-24:
Hello @kashishdhal, sorry for the delay.

Indeed, `CMakeLists.txt` and `app-colcon.meta` are missing. I guess that there is an error in this app folder because it is not fully tested and working.

Is it possible for you to add both missing files (take [this](https://github.com/micro-ROS/zephyr_apps/tree/foxy/apps/int32_publisher) as reference) in order to test it and if it works PR them?

Thanks

kashishdhal, 2020-08-25:
sure, I will try to add the files and update you, thanks!

kashishdhal, 2020-08-31:
Hello @pablogs9 ,

Sorry for the delay in response. I tried to build firmware and got an error. I will explain what I did:

I looked at the `CMakeLists.txt` for different apps and they looked exactly similar so I just copied the `CMakeLists.txt` from int32_publisher without making any changes.

However, I was not sure about the `app-colcon.meta`, this is how it looks:

`{
    "names": {
        "microxrcedds_client":{
            "cmake-args":[
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF"
            ]
        },`

`        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=1",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=0",
                "-DRMW_UXRCE_MAX_SERVICES=0",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=1"
            ]
        }
    }
}`

Here is the output: 

<pre>In file included from <b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.h:20</b>,
                 from <b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_client.c:15</b>:
<b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:63:3:</b> <font color="#CC0000"><b>error: </b></font>unknown type name &apos;<b>uxrUDPTransport</b>&apos;
   63 |   <font color="#CC0000"><b>uxrUDPTransport</b></font> transport;
      |   <font color="#CC0000"><b>^~~~~~~~~~~~~~~</b></font>
<b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:64:3:</b> <font color="#CC0000"><b>error: </b></font>unknown type name &apos;<b>uxrUDPPlatform</b>&apos;
   64 |   <font color="#CC0000"><b>uxrUDPPlatform</b></font> udp_platform;
      |   <font color="#CC0000"><b>^~~~~~~~~~~~~~</b></font>
In file included from <b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:31</b>,
                 from <b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.h:20</b>,
                 from <b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_client.c:15</b>:
<b>/home/gnclab/uros_ws/firmware/mcu_ws/build/rmw_microxrcedds/include/rmw_microxrcedds_c/config.h:15:41:</b> <font color="#CC0000"><b>error: </b></font>&apos;<b>UXR_CONFIG_UDP_TRANSPORT_MTU</b>&apos; undeclared here (not in a function); did you mean &apos;<b>UXR_CONFIG_SERIAL_TRANSPORT_MTU</b>&apos;?
   15 |     #define RMW_UXRCE_MAX_TRANSPORT_MTU <font color="#CC0000"><b>UXR_CONFIG_UDP_TRANSPORT_MTU</b></font>
      |                                         <font color="#CC0000"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/gnclab/uros_ws/firmware/mcu_ws/build/rmw_microxrcedds/include/rmw_microxrcedds_c/config.h:15:41:</b> <font color="#06989A"><b>note: </b></font>in definition of macro &apos;<b>RMW_UXRCE_MAX_TRANSPORT_MTU</b>&apos;
   15 |     #define RMW_UXRCE_MAX_TRANSPORT_MTU <font color="#06989A"><b>UXR_CONFIG_UDP_TRANSPORT_MTU</b></font>
      |                                         <font color="#06989A"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:73:40:</b> <font color="#06989A"><b>note: </b></font>in expansion of macro &apos;<b>RMW_UXRCE_MAX_BUFFER_SIZE</b>&apos;
   73 |   uint8_t input_reliable_stream_buffer[<font color="#06989A"><b>RMW_UXRCE_MAX_BUFFER_SIZE</b></font>];
      |                                        <font color="#06989A"><b>^~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
In file included from <b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.h:20</b>,
                 from <b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_client.c:15</b>:
<b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:223:33:</b> <font color="#75507B"><b>warning: </b></font>ISO C forbids zero-size array &apos;<b>custom_subscriptions</b>&apos; [<font color="#75507B"><b>-Wpedantic</b></font>]
  223 | extern rmw_uxrce_subscription_t <font color="#75507B"><b>custom_subscriptions</b></font>[RMW_UXRCE_MAX_SUBSCRIPTIONS];
      |                                 <font color="#75507B"><b>^~~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:226:28:</b> <font color="#75507B"><b>warning: </b></font>ISO C forbids zero-size array &apos;<b>custom_services</b>&apos; [<font color="#75507B"><b>-Wpedantic</b></font>]
  226 | extern rmw_uxrce_service_t <font color="#75507B"><b>custom_services</b></font>[RMW_UXRCE_MAX_SERVICES];
      |                            <font color="#75507B"><b>^~~~~~~~~~~~~~~</b></font>
<b>/home/gnclab/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:229:27:</b> <font color="#75507B"><b>warning: </b></font>ISO C forbids zero-size array &apos;<b>custom_clients</b>&apos; [<font color="#75507B"><b>-Wpedantic</b></font>]
  229 | extern rmw_uxrce_client_t <font color="#75507B"><b>custom_clients</b></font>[RMW_UXRCE_MAX_CLIENTS];
      |                           <font color="#75507B"><b>^~~~~~~~~~~~~~</b></font>
[... cut ...]
make[2]: *** [CMakeFiles/app.dir/build.make:79: CMakeFiles/app.dir/src/main.c.obj] Error 1
make[2]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
make[1]: *** [CMakeFiles/Makefile2:2313: CMakeFiles/app.dir/all] Error 2
make[1]: Leaving directory &apos;/home/gnclab/uros_ws/firmware/build&apos;
make: *** [Makefile:87: all] Error 2
<font color="#EF2929">FATAL ERROR: command exited with status 2: /usr/bin/cmake --build /home/gnclab/uros_ws/firmware/build</font>
</pre>

It would be great if you can guide me! Thanks!

pablogs9, 2020-09-01:
We have updated this app for you, check here: https://github.com/micro-ROS/zephyr_apps/pull/19

I can build it successfully, please remember to do this in the configuration step:

`ros2 run micro_ros_setup configure_firmware.sh int32_wifi_publisher -t udp -i [AGENT IP] -p [AGENT PORT]`

The WiFi SSID and PSK can be set up in `main.c` line 59;

Then you can do the build step as usual. Please give me feedback about your progress.

Thanks!

**Please remember that WiFi support for this board is experimental and I had some troubles with connection stability when I implemented it.**

kashishdhal, 2020-09-01:
Hello @pablogs9 ,

Thanks a ton for the update! I will check this out and update you! If wifi doesn't work as expected then I can use an onboard computer for serial connection. However, I will surely try to make this work!

Although I had a quick question: Can I run the micro-ros agent in ROS2 Dashing or it is mandatory to use ROS2 Foxy for micro-ros agent? I will flash the Foxy micro-ros code in the MCU. Will there be any communication issue or it would be fine?

Secondly, I want to learn how to create custom apps for my board. What would be recommended way of going through that process? Should I look at all the apps, how they have been created? or if you recommend any other way of doing that? Is there any document available which I can have a look?

Please let me know about it!  Thanks!

pablogs9, 2020-09-02:
Hello @kashishdhal:

- In order to test this update, please start with a clean environment.
- Regarding the micro-ROS Agent, if you don't have ROS 2 Foxy installed you can use our micro-ROS Foxy Agent Docker: https://hub.docker.com/r/microros/micro-ros-agent I guess that this would be super useful in order to not install Foxy locally on your computer.
- If you want to create your own app you can check the apps inside `firmware/zephyr_apps/apps` to see the basic structure of an app whitin the examples. You will find that RCLC is being used as API for micro-ROS apps, you can find further information in the [RCLC repo](https://github.com/micro-ROS/rclc) and in [this tutorial](https://micro-ros.github.io/docs/tutorials/core/programming_rcl_rclc/).

Please let us know if you have any other doubt.

kashishdhal, 2020-09-02:
Hello @pablogs9,

Thanks for the reply and sharing the tutorials! I will go through them as time allows!

I can build wifi app successfully now with the clean environment.

Secondly, I have ROS2 Foxy installed, but it is installed on a Virtual Box and I am not able to ping that computer from my local computer. However, I have installed ROS2 Dashing (and micro-ros Dashing) on my local computer. So that is what I asked, if I can run the micro-ros agent in ROS2 Dashing not to have communication problems.

Another question, Can I run Docker for ROS2 Foxy in Ubuntu 18.04? I am not familiar with docker and don't know how much time it will take to learn the functionality of docker.

pablogs9, 2020-09-03:
Regarding compatibility, I'm not sure about it because they have changed major version for FastDDS between Dashing and Foxy and we use it in the agent. Probably XRCE-DDS (middleware of micro-ROS between client and agent) is not problematic because there is no significant changes. So you can try.

Regarding Docker, of course, you can run a Docker image of Ubuntu 20.04 and ROS 2 Foxy on top of 18.04 (in fact, this is my setup until I have time to update my workstation to 20.04). I encourage you to try Docker because it can ease a lot some parts of the development and the learning curve is not so hard.

Let me know your progress. Thanks!

kashishdhal, 2020-09-03:
Okay thanks @pablogs9 , I will try learning Docker as well as running micro-ros on Dashing and let you know how it goes!

kashishdhal, 2020-12-27:
Hello @pablogs9 ,

Thanks for all the help and sorry for the delayed response as I was a bit involved with my course work. 

I learnt the Docker and was able to flash the STM32 board using your micro-ros Docker image, it was super easy as compared to the Virtual Box. I also looked at the apps build by your team and rclc tutorials as suggested by you.

Now, I want to create an app to send PWM commands to the motor. Can you please guide me on how to configure the PWM on this board? If there is any manual available which I can refer to, please provide me a link to that! or else if you have created any similar app, please provide a link to that. I will be thankful to you!

pablogs9, 2021-01-11:
Hello, @kashishdhal I guess that this is off-topic from micro-ROS.

I guess that Zephyr documentation will be super useful for you: https://docs.zephyrproject.org/latest/reference/peripherals/pwm.html

Hope that your project goes well, and sorry for the inconvenience.

Remember to keep us updated about your project with micro-ROS!

kashishdhal, 2021-01-11:
Hello @pablogs9 ,

Thanks for the reply! For one of my robots currently, I am using Arduino+ROS-SERIAL. I was thinking to replace it with STM32 Disco+Micro-ROS. For that, I would need to send PWM command to control the motor + capability of reading the encoders in the board.

I will read the Zephyr documentation for PWM and how about reading the encoders? I am reading the encoder channels in hardware interrupt in arduino. What would be my best bet here?

Also, do you recommend exploring the Arduino+Micro-ROS repository for that? Please let me know.

Regards,
Kashish Dhal