micro-ROS/micro_ros_setup #169: apt update breaks firmware create script

eden-desta, 2020-08-05:
The apt update placed on line 45 breaks the script if you are running the create firmware step. 
Upon removing it, the system worked. 

This should likely be removed because a user will manually run the update prior to running the script anyways.

pablogs9, 2020-08-07:
Hello @eden-desta, can you try if this changes https://github.com/micro-ROS/micro_ros_setup/pull/172 fixes this issue?

I guess that you had troubles with `apt update` permissions right?

eden-desta, 2020-08-11:
@pablogs9 you should not have to run it with sudo. If you remove that, the build works.

pablogs9, 2020-08-24:
Yes, but have to be sure that apt is up-to-date. Do you propose any solution for automating this in the creation script?

closed 2020-09-07 by pablogs9