micro-ROS/micro_ros_setup #174: ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 Inside Docker

thihakyawjob, 2020-08-09:
Dear,
The following command does not work inside docker system.
**ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0**

How can I access ttyUSB0 from docker image?

Please guide me.

[image]

Regards,
Thiha Kyaw

DorBenHarush, 2020-08-11:
Hi,
You need to share the dev directory with the docker container in the creation like this:
docker run -it --net=host -v /dev:/dev --privileged  ros:foxy
Then type in the container:
ls /dev/serial/by-id/*
to find the name of the device, use it for --dev.

Regards,
Dor Ben Harush.

pablogs9, 2020-08-24:
Thanks @DorBenHarush for the contribution.

This should resolve the Docker issues.

closed 2020-08-24 by pablogs9