micro-ROS/micro_ros_setup #175: uros_pingpong in stm32f429i-disco

thihakyawjob, 2020-08-09:
Dear,

I use stm32f429i-disco board.

I got the following error when I build uros_pingpong example.

> arm-none-eabi-ld: /uros_ws/firmware/NuttX/staging/libapps.a(libmicroxrcedds_client_serial_transport_posix.c.obj): in function `uxr_read_serial_data_platform':
> /uros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/serial/serial_transport_posix.c:45: undefined reference to `poll'
> make[1]: *** [Makefile:185: nuttx] Error 1

Please guide me how I can solve the problem.

[image]

Regards,
Thiha

MARLARCHAN, 2020-08-19:
For this issue, please check **defconfig** file in **board/nsh **folder. 

Edit to be enable poll

CONFIG_DISABLE_POLL=y
into 
#CONFIG_DISABLE_POLL is not set

And firmware build again. 

Best Regards,
Marlar

pablogs9, 2020-08-24:
Hello, thanks @MARLARCHAN for the answer. @thihakyawjob does it solve your problem?

MARLARCHAN, 2020-08-24:
Please suggest my current error.

[image]Currently I'm trying ping_pong example with STM32F429-disc board on ubuntu 20.04version.

firstly, i call nsh uros_pingpong using gtkterm.

Then, i run microros agent. It shows in figure.

And with another new terminal, i put micrors topic.
When i put topic, it shows [image]the error Could not determine the type for the passed topic.
This shows in below figure also.

Please advise me how to solve this problem.

Thank you all!!!
Best Regards,
Marlar

pablogs9, 2020-08-24:
Can you run the micro-ROS Agent with the `-v6` flag in order to see if the communication is happening between both parts?

luminize, 2021-01-21:
I have the same situation (I think) the pin_pong does not work on the discovery board. I'm running zephyr and ubuntu 20.04
starting the agent it seems to be stuck at this:
```
bdb@bdb-ThinkPad-P51:~/projects/microros_ws_discovery-l475-iot1$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_0669FF323338424E43162338-if02 
[1611245992.922297] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
```
when running the micro-ros agent with the `-v6` flag I get this output (waiting forever):
```
bdb@bdb-ThinkPad-P51:~/projects/microros_ws_discovery-l475-iot1$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_0669FF323338424E43162338-if02 -v6
[1611246551.507044] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1611246551.507255] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```
(un)plugging the board shows the popup of the USB disk. So I guess I might have something wrong with flashing? I plugged in the board straight out of the box.
flashing routine output:
```
bdb@bdb-ThinkPad-P51:~/projects/microros_ws_discovery-l475-iot1$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for zephyr platform discovery_l475_iot1
-- west flash: rebuilding
[  2%] Built target parse_syscalls_target
[  3%] Built target kobj_types_h_target
[  4%] Built target syscall_list_h_target
[  4%] Built target driver_validation_h_target
[  5%] Built target offsets
[  5%] Built target offsets_h
[  5%] Built target zephyr_generated_headers
[ 22%] Built target kernel
[ 27%] Built target microroslib_project
[ 28%] Built target app
[ 59%] Built target zephyr
[ 60%] Built target linker_script_target
[ 62%] Built target isr_tables
[ 63%] Built target arch__common
[ 71%] Built target arch__arm__core__aarch32
[ 77%] Built target arch__arm__core__aarch32__cortex_m
[ 79%] Built target arch__arm__core__aarch32__cortex_m__mpu
[ 80%] Built target lib__libc__newlib
[ 85%] Built target lib__posix
[ 86%] Built target soc__arm__common__cortex_m
[ 88%] Built target drivers__gpio
[ 90%] Built target drivers__serial
[ 95%] Built target ..__modules__hal__stm32__stm32cube
[ 97%] Built target zephyr_prebuilt
[ 97%] Built target linker_pass_final_script_target
[100%] Built target zephyr_final
Scanning dependencies of target west_flash_depends

[100%] Built target west_flash_depends
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/bdb/projects/microros_ws_discovery-l475-iot1/firmware/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-01340-ga0e8edc4e-dirty (2020-02-14-05:38)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : clock speed 500 kHz
Info : STLINK V2J28M18 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.221551
Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32l4x.cpu       hla_target little stm32l4x.cpu       running

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800738c msp: 0x2000eae8
Info : device idcode = 0x10076415 (STM32L47/L48xx - Rev: 4)
Info : flash size = 1024kbytes
Info : flash mode : dual-bank
Info : Padding image section 0 at 0x0802600c with 4 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x08026010 .. 0x080267ff
auto erase enabled
wrote 155664 bytes from file /home/bdb/projects/microros_ws_discovery-l475-iot1/firmware/build/zephyr/zephyr.hex in 6.799408s (22.357 KiB/s)

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08006df8 msp: 0x2000e308
verified 155660 bytes in 4.574846s (33.228 KiB/s)

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
shutdown command invoked
```

closed 2022-01-28 by pablogs9