micro-ROS/micro_ros_setup #178: Suggestion: Using PlatformIO Core internally

russkel, 2020-08-18:
Hello,

Have you considered using platformio for a lot of the firmware/buildchain setup stages? I think there's quite a large overlap with a lot of the scripts inside microros and piocore.

JanStaschulat, 2020-08-18:
Hi @russkel, 
thank you very much for your comment. Currently the colleagues who maintain the build toolchain are on vacation until 21.08. They will come back on this issue as soon as they are back.

russkel, 2020-08-18:
Thanks for your reply @JanStaschulat.

pablogs9, 2020-08-24:
Hello @russkel, we don't have planned the integration with PlatformIO. But of course, we will be happy to see community Pull Requests with these kinds of improvements.

Let us know if you are planning to do this integration.

russkel, 2020-08-24:
> Let us know if you are planning to do this integration.

Hi @pablogs9, okay no worries. I will seriously consider it when I get a bit more experience with microROS (and PlatformIO I guess!). I believe this support would open up microROS to a much wider range of boards (FreeRTOS support not-withstanding..)

pablogs9, 2020-08-24:
Ok thanks, I'm closing this. Please feel free to reopen if you need to discuss something more about this PlatformIO support.

closed 2020-08-24 by pablogs9