micro-ROS/micro_ros_setup #179: Issue: Could not determine the type for the passed topic

MARLARCHAN, 2020-08-19:
Currently I'm trying ping_pong example with STM32F429-disc board on ubuntu 20.04version.

firstly, i call nsh uros_pingpong using gtkterm.

Then, i run microros agent. It shows in figure.

And with another new terminal, i put micrors topic.
When i put topic, it shows the error Could not determine the type for the passed topic.
This shows in below figure also.

Please advise me how to solve this problem.

Thank you all!!!
Best Regards, 
Marlar[image][image]

pablogs9, 2020-08-24:
Please run the agent with `-v6` flag (verbose mode) in order to see if there is communication between agent and client.

Could you tell us which tutorial/instructions are you following?

MARLARCHAN, 2020-08-24:
Hi!

  Thank you for your support. I follow First microROS Application on Nuttx.

I'm trying my project for microROS using stm32f429-disc with serial communication. I use Nuttx on ubuntu 20.04version. 

I'm trying now using with v6 but still have issue like this. Please suggest me using v6 command correct or not.

Thank you
Marlar[image]

pablogs9, 2020-08-24:
sorry `-v 6` I forgot a space, and please add it at the end of the of the line

MARLARCHAN, 2020-08-24:
Thank you for your support. [image]
But I still have problem when testing app.
I use with gym term, in there, when I test uros_pingpong, it's activated but cannot read any information like this.
[image]

After that, I test Ping topic, but always show topic does not appear to be published yet.[image]

Please suggest for these issues. 

I would like to know when is microROS discourse seminar meeting? If there is any seminar meeting, I would like to join. 

And I understand this tutorials are still upgrading but nobody answer when I face issue. please advise me which email can I contact directly because  now my project is delay. 

Thank you 
MARLAR

pablogs9, 2020-08-24:
Well, your micro-ROS application is working BUT your configuration is using the same serial port for the NSH and micro-ROS.

The unknown characters `???XRCEY???`  are the data that should be arriving to the agent. As I'm seeing you are opening the serial port twice: one in GKTerm and other in the micro-ROS Agent.

Do you have any other serial port available on the board? You can change the serial port used by micro-ROS in the Nuttx menuconfig.

MARLARCHAN, 2020-08-24:
I open serial one by one. After i check with GTKTerm, I run microROS Agent. 

Yes. I have changed in Nuttx menuconfig , 
UDP  transport -----> to Serial Transport

But it shows the issue topic not found.

pablogs9, 2020-08-24:
The topic is not published because there is no communication between micro-ROS client and micro-ROS agent so the topic is not been published.

The communication that is supposed to arrive at the agent is been printed in the GTKTerm. I not sure about having two serial ports opened at the same time.

Do you have any idea of launching the Nuttx app directly? It would ease the client-agent connection...

Another possibility is to use a different port, in the same `menuconfig` where you have selected the serial port you can also select which serial port will be used by the micro-ROS middleware.

MARLARCHAN, 2020-08-25:
Hi,
I open GTKTerm for usb port initiate using nsh by typing uros_pingpong.

After that I cloae GTKTetm and agent start run.

But still have the same issue. 

Can you explain what is  Nuttx app directly?

Best Regards, 
Marlar

pablogs9, 2020-08-25:
This open and close approach is not likely to work with serial communication because the micro-ROS client starts its communication as soon as the app is launched.  Please find if your board support for Nuttx allows you to use a second serial port and select it in the Nuttx `menuconfig` micro-ROS configuration section.

Regarding the Zephyr support: https://docs.zephyrproject.org/latest/boards/index.html
Some successful Zephyr + micro-ROS ports by the community: https://github.com/micro-ROS/micro_ros_setup/pull/182

MARLARCHAN, 2020-08-25:
For STM32F429-disc, USART1 is used. 

So,I use USART1 in menuconfig, and also in board.h also. 

But still have the problem.

Best Regards, 
Marlar

pablogs9, 2020-08-25:
Can you enable and use `USART2` for example?

Do you know how to configure `USART2` for being used by Nuttx as a mounted device?

Once the device is ready to be used by Nuttx it should have a descriptor like `/dev/ttyS0` or something like this.

This is the descriptor that you should set in the `menuconfig` micro-ROS section.

MARLARCHAN, 2020-08-26:
Hi 
[Board.txt](https://github.com/micro-ROS/micro_ros_setup/files/5128532/Board.txt)

  Could you suggest me how to enable USART2  and driver support also. 

In board.h , only USART1 is included. So it means USART2 is not supported.
Also UART1 is working with nsh.

I attach my board.h in here.

pablogs9, 2020-08-26:
I'm not an expert in Nuttx, maybe you can find help in some Nuttx specific forum...

Anyways, have you tried to enable USART2 in `menuconfig`?

JackFrost67, 2021-04-20:
my 2 cents on this issue: I was having the same problem yesterday, I triggered the RST button on my nucleo_f767zi and it works fine.

closed 2022-01-28 by pablogs9