micro-ROS/micro_ros_setup #180: Mutiple nodes on one board

allstarry, 2020-08-24:
Hi, I want to develop something with micro-ros, and I am learning this demo for now. I want to know if micro-ros support mutiple nodes working together, because I didn't see anyting about that. Or it do support it now but need users to some extra work?

bjv-capra, 2021-05-04:
I'd like to add a +1 on interest for support for multithreading.
We run multiple nodes, each one has different needs and we have a design that lets us easily hook different nodes to different threads, but not multiple nodes to a single thread.
I'll see if I can work around it anyways. This topic was quite useful

pablogs9, 2021-05-05:
Closing this since [multithread support](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/c292903e6c388011520c03ccb39872a6705f72c6/CMakeLists.txt#L65) has been added to micro-ROS middleware. Please reopen if you have any other questions/suggestions.

closed 2021-05-05 by pablogs9