micro-ROS/micro_ros_setup #184: micro-ros  host build problem

DorBenHarush, 2020-08-27:
Hi, 
I`m trying to create firmware for host and some unexpected errors occurred during the build process.
These are the errors:

[image]

I followed the tutorial [here](https://micro-ros.github.io/docs/tutorials/core/first_application_linux/), and I`m using a docker container as shown. 

Regards,
Dor.

pablogs9, 2020-08-27:
Hello @DorBenHarush we have found some issues related to the build system thanks to your issue.

We have working here: https://github.com/micro-ROS/micro_ros_setup/pull/185, now it passing the CI test and once it is merged you can try again the tutorial without (hopefully) any issue.

Thanks!!

pablogs9, 2020-08-28:
The PR has been merged and the issue solved. Please @DorBenHarush, reopen this issue if this does not solve your problem.

Sorry for the inconvenience.

DorBenHarush, 2020-08-28:
Thanks, it worked. But I think the tutorial is outdated.
I typed this:
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
as shown [here](https://micro-ros.github.io//docs/tutorials/core/first_application_linux/)
and recived:
`Usage: 'MicroXRCEAgent <udp4|udp6|tcp4|tpc6|serial|pseudoterminal> <<args>>'
For a more detailed description about all the available arguments, please execute the agent with '-h/--help' option.`
Can you please guide me. 

Regards, 
Dor.

pablogs9, 2020-08-31:
@jamoralp can you check this? 

Related: https://answers.ros.org/question/360494/micro-ros-agent-displays-usage-and-wont-start-with-valid-arguments/

pablogs9, 2020-08-31:
Hello @DorBenHarush  we have solved it here: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/183

Can you check if the Agent is working now and close the issue?

Thanks!!

DorBenHarush, 2020-08-31:
Hi, the Agent runs but when I try to run the micro-ROS node with `ros2 run micro_ros_demos_rclc ping_pong` this response pops:
`Package 'micro_ros_demos_rclc' not found`. I checked the pkg list and the package is not there.

pablogs9, 2020-09-01:
Hello @DorBenHarush , `micro_ros_demos_rclc` package should be found inside [this folder](https://github.com/micro-ROS/micro-ROS-demos/tree/foxy/rclc/ping_pong) can you check if you have this folder inside `src`?

Have you done the `source install/local_setup.bash`? Anyways I'm doing checks for this tutorial step by step, I will tell you in a while.

DorBenHarush, 2020-09-01:
Hi @pablogs9 , I typed `ros2 pkg list | grep micro` to check if the package `micro_ros_demos_rclc` exists. furthermore I checked in the `src` folder and it only contained  `micro_ros_build`.

pablogs9, 2020-09-01:
Can you paste here a step by step procedure with all commands that you are executing for reproducing this issue? I would like to check your workflow 

Thanks!

DorBenHarush, 2020-09-01:
I have solved the problem, since I'm working from docker container I needed to create another container in roder to run the `micro_ros_deomos_rclc`. Then I repeated the first stages in the tutorial to recreate the enviorment. My mistake was **this**: I didnt proceed to the `create_firmware_ws.sh` step in the new container, Therefore I didnt download `micro_ros_demos_rclc` package. 

I have tried it again but this time, instead of creating another container, I created another session to the same container and it worked. I think this tutorial needs some rewriting in order to be more clear.
Thank you for your support.
Regards,
Dor.

pablogs9, 2020-09-01:
Thanks for the feedback @DorBenHarush.

Can we close the issue or you have some more questions?

Thanks!

closed 2020-09-01 by DorBenHarush