micro-ROS/micro_ros_setup #193: Cannot create a custom micro-ROS type following the tutorial.

Aetherbase, 2020-09-15:
I followed the tutorial as given in the link [here](https://micro-ros.github.io/docs/tutorials/core/create_new_type/).
but cannot include the header in a custom app.
Following is the steps i took:
```
ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32
mkdir -p firmware/freertos_apps/apps/custom_publisher

#creeated appropropriate app.c and app-colcon.meta

ros2 run micro_ros_setup configure_firmware.sh custom_publisher -t udp -i <localip> -p 8888
cd firmware/mcu_ws
ros2 pkg create --build-type ament_cmake my_custom_message
cd my_custom_message
mkdir msg
touch msg/MyCustomMessage.msg

#added few fields of type int16 in MyCustomMessage.msg
#added appropriate dependencies in CMakeLists.txt and package.xml.
#added #include <my_custom_message/msg/my_custom_message.h> to my app.c 

ros2 run micro_ros_setup build_firmware.sh
```
upon following th above steps, build_firmware terminates with output below:
```
fatal error: my_custom_message/msg/my_custom_message.h: No such file or directory
 #include <my_custom_message/msg/my_custom_message.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
Please help me to get past this issue.I am not understanding what wrong i did. I followed the tutorial as it is but the package my_custom_message is not even compiled yet.

Aetherbase, 2020-09-15:
Hello @pablogs9, Please help if possible.

pablogs9, 2020-09-16:
Hello @Aetherbase, check the first block:

```bash
cd firmware/mcu_ws
ros2 pkg create --build-type ament_cmake my_custom_message
cd my_custom_message
mkdir msg
touch msg/MyCustomMessage.msg
```

The folder is created inside `firmware/mcu_ws` which is the folder that colcon crosscompile, you are creating it inside `firmware/freertos_apps/apps/custom_publisher` which is the app folder (not directly built by colcon as `libmicroros.a`). So try to put your custom message inside the correct folder.

Anyways I'm thinking that makes a lot of sense to have custom messages inside the micro-ROS app that is going to be built. Maybe we can modify the build system in order to allow this. @jamoralp @FranFin

Aetherbase, 2020-09-16:
@pablogs9 thank you for your reply. If u read the steps a bit carefully, I actually had created the message package inside `firmware/mcu_ws` itself. The package is not being cross compiled at all. The folder `firmware/freertos_apps/apps/custom_publisher` is being used to create the custom app only. I don't understand why the package `my_custom_message` isn't being compiled. do i have to add it somewhere in dependencies?

pablogs9, 2020-09-16:
@Aetherbase so sorry for this, I did not read it well... Let me share my test session in a ROS 2 Foxy clean environment:

https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz

The session is quite long but you can find this points:
- Message creation [5:40](https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz?t=5:40)
- Header check [11:28](https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz?t=11:28)
- Library .a check [12:08](https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz?t=12:08)

How does your .msg looks like?

Aetherbase, 2020-09-16:
@pablogs9 the session was really helpful. Thanks a lot for all the time you put for this, and also Im very sorry. The mistake was mine, I did a typo while i created the package. Big mistake. 

> 
> Anyways I'm thinking that makes a lot of sense to have custom messages inside the micro-ROS app that is going to be built. Maybe we can modify the build system in order to allow this. @jamoralp @FranFin

About this i think it will be better for creating a package accessible to host packages in the same workspace because most of the usecases will require the message type to be accessible to host packages for creating a ros2 - micro-ROS communication. Ill keep this issue open for this, or should i open a PR?

pablogs9, 2020-09-16:
If you contribute with a PR it would be awesome. IMO we can close this issue and then start discussing about this feature on a PR.

One of my first ideas is having a `create_message_package.sh` script integrated into the build system which creates the package in a certain folder and then "links" it to the host workspace in order to have it compiled and cross-compiled.  Which approach would you take?

Aetherbase, 2020-09-16:
>
>One of my first ideas is having a create_message_package.sh script integrated into the build system which creates the package in a certain folder and then "links" it to the host workspace in order to have it compiled and cross-compiled. Which approach would you take?

I was thinking of something similar to this. Will close this issue and open a PR. Let's discuss about this in detail there.

closed 2020-09-16 by Aetherbase