micro-ROS/micro_ros_setup #199: build_firmware.sh breaks with Zephyr

CLansdaleBEC, 2020-09-30:
Hello, following up on the question from the micro-ROS slack group originally posted by @edesta-be . As of yesterday (29Sept2020) I'm not able to build firmware while using Zephyr. I created a fresh workspace and followed the create/configure/build steps that I have run successfully many times before. 

When using an old microROS workspace, one with the firmware directory already created, I can still build just fine. It's only when I create a new firmware workspace that it fails. 

**Steps Taken**

```
mkdir uros_ws && cd uros_ws

git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup

rosdep update && rosdep install --from-path src --ignore-src -y

colcon build

source install/local_setup.bash

ros2 run micro_ros_setup create_firmware_ws.sh zephyr nucleo_h743zi

ros2 run micro_ros_setup configure_firmware.sh int32_publisher --transport serial --dev 3

ros2 run micro_ros_setup build_firmware.sh
```

The build script fails on rmw_microxrcedds

[image]

Thank you!

edesta-be, 2020-09-30:
@pablogs9 This is the ticket.
Thank you!

pablogs9, 2020-10-01:
Hello @edesta-be @CLansdaleBEC sorry for that... A bug in a recent implementation of the UDP client-agent discovery.

Fixed with https://github.com/micro-ROS/rmw-microxrcedds/pull/80

pablogs9, 2020-10-01:
Closing since merged. Please reopen if any other related issue.

closed 2020-10-01 by pablogs9