micro-ROS/micro_ros_setup #200: Debugging Nuttx Application

MARLARCHAN, 2020-10-01:
Dear MicroROS Team,
I'm trying to add customize message for new application. so Debugging for Nuttx is needed.

So, I'm trying firstly Debugging NuttX Application Tutorial with Ubuntu 20.04  But, I don't understand the tutorial. 
After downloading  "git clone --depth 1 https://github.com/sony/openocd-nuttx", I cannot find .gdbinit: file. Do I need to create by myself? 

Also although i installed arm-none-eabi-gdb, I cannot do this command:  "arm-none-eabi-gdb nuttx"

I can only use with this command "gdb nuttx", and it shows the error as follow:
arm-none-eabi-gdb nuttx
(gdb) print-offset

Best Regards.
Marlar

pablogs9, 2020-10-01:
Hello @MARLARCHAN, I'm not the maintainer of this tutorial but I have been reading it and maybe you should create this `.gdbinit` in the directory with the content detailed in the tutorial:

```
define print-offset
 printf "#define PID  %p\n",&((struct tcb_s *)(0))->pid
 printf "#define XCPREG  %p\n",&((struct tcb_s *)(0))->xcp.regs
 printf "#define STATE %p\n",&((struct tcb_s *)(0))->task_state
 printf "#define NAME %p\n",&((struct tcb_s *)(0))->name
 printf "#define NAME_SIZE %d\n",sizeof(((struct tcb_s *)(0))->name)
end
```

---

@ralph-lange @FranFin What should we do with these Nuttx tutorials? Is there any maintainer for them?

MARLARCHAN, 2020-10-01:
Dear @pablogs9 ,

    I have created .gdbinit file. After that, I can only use the command " gdb nuttx " instead of " arm-none-eabi-gdb nuttx".  And the error is still as in above attachment. Please advise me something.

Thank you 
Best Regards,
Marlar

MARLARCHAN, 2020-10-05:
Dear @ralph-lange @FranFin @pablogs9 
           Please suggest me this issue also! Thank you.

Best Regards,
Marlar

closed 2021-03-12 by pablogs9