micro-ROS/micro_ros_setup #201: Create New Custom Message

MARLARCHAN, 2020-10-05:
Dear @pablogs9 @ralph-lange @FranFin 
 
   I'm trying to add custom message into microros right now. I follow according to the link.
https://micro-ros.github.io/docs/tutorials/core/create_new_type/

In that link, I update the following variables as in eg.
my_custom_message__msg__MyCustomMessage msg;

msg.byte_test = 3;
msg.uint32_test = 42;
msg.point32_test.x = 1.23;
msg.point32_test.y = 2.31;
msg.point32_test.z = 3.12;

I follwed according to the instructure procedure. 

(my_custom_message__msg__MyCustomMessage ) structure is declared in (my_custom_message__struct.h ) as the following:
// Struct defined in msg/MyCustomMessage in the package my_custom_message.
typedef struct my_custom_message__msg__MyCustomMessage
{
  bool bool_test;
  uint8_t byte_test;
  uint8_t char_test;
  float float32_test;
  double double_test;
  int8_t int8_test;
  uint8_t uint8_test;
  int16_t int16_test;
  uint16_t uint16_test;
  int32_t int32_test;
  uint32_t uint32_test;
  int64_t int64_test;
  uint64_t uint64_test;
  geometry_msgs__msg__Point32 point32_test;
} my_custom_message__msg__MyCustomMessage;

However, I'm facing the issue as shown in following:
[image]

 I've no idea to trace this issue.Please check and guide for me what is the issue related.

Best Regards,
Marlar

pablogs9, 2020-10-05:
Provide a comprehensive list of commands and code of your application in order to reproduce this issue.

pablogs9, 2020-10-05:
For more info check:

https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz

The session is quite long but you can find:
- Message creation [5:40](https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz?t=5:40)
- Header check [11:28](https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz?t=11:28)
- Library .a check [12:08](https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz?t=12:08)

MARLARCHAN, 2020-10-08:
Dear @pablogs9 

     Thank you for sharing your video. I just want to ask some questions.
I follow according to your video. However, when agent run and application start run, the nsh become aborting. I suspect I need to update int into microROS agent also. 
Please suggest me do I need to update microROS agent side?

Best Regards,
Marlar
[image]

pablogs9, 2020-10-08:
Hello @MARLARCHAN I'm closing this issue since it seems that you have solved the problem of creation and cross-compilation of a micro-ROS new type.

Regarding the other part: open a new issue targeting this new problem, and please take into account:
- Be **extremely descriptive** about the procedure that produces the error
- Consider opening the issue in a repository related to the nature of the problems, not always in the build system repository.
- **Always** provide a list of commands so we can reproduce your issue.
- If possible, **always** provide the code or a link to the code so we can reproduce your issue.
- **Avoid** screenshots and try to copy-paste full logs or upload the log file (if the log is too long for copy-pasting)

closed 2020-10-08 by pablogs9