micro-ROS/micro_ros_setup #202: Does STM32CubeIDE support Olimex STM32-E407

vibnwis, 2020-10-05:
Hi there,
Very appreciate if someone able to comment of the above title, please. If not, what IDE platform is recommended. The screen-shot shows, STM32CubeMX is being associated with. However, I couldn't find any communities claim using it. Sincerely, hope someone can shed some lights here, please.

Thank you
Vibwis
[image]

pablogs9, 2020-10-05:
Hello @vibnwis, this row in the compatibility table shows that the version of FreeRTOS used for Olimex STM32-E407 in micro-ROS is the lastest used in CubeMX. In that sense, we provide a build system that uses a [CubeMX generated project](https://github.com/micro-ROS/freertos_apps/tree/foxy/microros_olimex_e407_extensions) for Make in order to build a FreeRTOS-based system for this platform.

You have all the available information for using this board with micro-ROS and FreeRTOS in our web page: https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/freertos/

Please tell us if this solves your question.

vibnwis, 2020-10-05:
Hi pablogs9  and mRos team, thank you so much for looking into my questions. Your answers truly shed some lights for me. I will use it, the CubeMX generated project, by importing into STM32CubeIDE, next thing in the morning. BTW, in the web page: https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/freertos/, is it correct to say, for the Olimex board, it  has to use JTag, however, for STM32 Nucleo, it would be ST3-Link ? Sorry if this sound ignorant.

pablogs9, 2020-10-05:
You are welcome, this tutorial is for our reference hardware platform, so in this case, you will need to JTAG to flash the Olimex STM32-E407. In the case you have another not-supported board, some scripts may differ a little.

In the case of an STM32 Nucleo maybe you will need to modify flashing files in order to use micro-ROS in it. We can keep this issue open and we can talk about the procedure here. First, let us know if you are able to build the micro-ROS library in your toolchain.

vibnwis, 2020-10-06:
Hi, first and foremost, I managed to compile the micro-ROS firmware without any issue. Proceeded to downloading the freertos Olimex STM32e407 and imported in STM32CubeIDE. The IDE complained that .ico filename should be same as the project filename. Hence, changed that accordingly. After that, the compilation was successful. I have also downloaded the ST Nucleo F446ZE, the same issues with the .ico. Having changed that, everything was sweet.
However, after click the "clean" project, all kind of errors came out. I thought those were due to the include paths have been deleted, and changed the relative paths to absolute path. It did resolve some errors, but, some errors like the following. Can you help, please. However, I think if in worst case, I can re-load the code. However, it would be good to know what the settings are. Can you shed some more lights on that. I was thinking to port it to STM32G431xxx eval board. The board is without Ethernet. I chose that for the reason below.

On a separate issues. Actually, I was interested in micro-ROS and FreeRTOS for my master degree project. However, it was meant to use CAN-bus instead of Ethernet. Would you able to comment if the existing micro-ROS can be easily added the CANopen? Any comments will be appreciated.

Thanks in advance. 

In file included from /home/wiki-ros/Downloads/uros_ws/firmware/mcu_ws/uros/rcutils/include/rcutils/macros.h:23:0,
                 from /home/wiki-ros/Downloads/uros_ws/firmware/mcu_ws/uros/rcutils/include/rcutils/allocator.h:26,
                 from /home/wiki-ros/Downloads/uros_ws/firmware/mcu_ws/uros/rcl/rcl/include/rcl/allocator.h:23,
                 from /home/wiki-ros/Downloads/uros_ws/firmware/mcu_ws/uros/rcl/rcl/include/rcl/init.h:23,
                 from /home/wiki-ros/Downloads/uros_ws/firmware/mcu_ws/uros/rcl/rcl/include/rcl/rcl.h:76,
                 from ../Src/main.c:32:
/home/wiki-ros/Downloads/uros_ws/firmware/mcu_ws/uros/rcutils/include/rcutils/configuration_flags.h:10:2: error: invalid preprocessing directive #cmakedefine; did you mean #define?
 #cmakedefine RCUTILS_NO_FILESYSTEM
  ^~~~~~~~~~~
  define
/home/wiki-ros/Downloads/uros_ws/firmware/mcu_ws/uros/rcutils/include/rcutils/configuration_flags.h:11:2: error: invalid preprocessing directive #cmakedefine; did you mean #define?
 #cmakedefine RCUTILS_AVOID_DYNAMIC_ALLOCATION
  ^~~~~~~~~~~
  define
/home/wiki-ros/Downloads/uros_ws/firmware/mcu_ws/uros/rcutils/include/rcutils/configuration_flags.h:12:2: error: invalid preprocessing directive #cmakedefine; did you mean #define?
 #cmakedefine RCUTILS_NO_THREAD_SUPPORT
  ^~~~~~~~~~~
  define
In file included from ../Src/main.c:33:0:
/home/wiki-ros/Downloads/uros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/client.h:18:10: fatal error: uxr/client/config.h: No such file or directory
 #include <uxr/client/config.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Src/subdir.mk:50: Src/main.o] Error 1
"make -j8 all" terminated with exit code 2. Build might be incomplete.

20:42:33 Build Failed. 8 errors, 9 warnings. (took 1s.175ms)

pablogs9, 2020-10-06:
Hello again @vibnwis, could you clarify which IDE are you using? I guess that you are using a ST proprietary IDE and I'm not sure about how does it works, so I feel that I can't help so much in this IDE-related issues.

When talking about FreeRTOS + micro-ROS + STM32 based targets, we generate the code with STM32CubeMX (the folder that you have seen) and then we build using CMake and Makefiles in a command line. Is compulsory for you to use this ST IDE? Which are the benefits? 

Regarding your errors, there are some `config.h.in` file along the micro-ROS environment that are configured in the CMake configuration time. So for example, in your errors I can see that `uxr/client/config.h` is not being created. This file should be created using [this one](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/foxy/include/uxr/client/config.h.in) in [this line](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/356a1e6d6181134f7d45c9ce7fe5bdff27d7c1ec/CMakeLists.txt#L406). 

This "file configuration process" allow us to pass CMake variables to a .h file and build conditionally some features, so for example if `rcutils` package is configured with the CMake flag `RCUTILS_NO_THREAD_SUPPORT` in the file `include/rcutils/configuration_flags.h` you will see a `#define RCUTILS_NO_THREAD_SUPPORT`. If not, this define will be commented.

In your case it seems that this is not working because your `.h` files have the same CMake flags (`#cmakedefine`) as the `.h.in`.
So maybe you need to investigate further in your IDE build process and understand how it manages the CMake configure/build process.

---

Regarding the CAN bus integration: the micro-ROS middleware is [Micro XRCE-DDS](https://micro-xrce-dds.docs.eprosima.com/en/latest/) and this middleware allows customizable transports. 

Usually, we work with two kinds of transports: stream-oriented and message-oriented. This way you have UDP or TCP transports (message-oriented) or UART (stream-oriented), there are some internal differences but the only thing that you have to know is that you can write a custom transport if you are able to fill functions (`uxr_open`, `uxr_close`, `uxr_write_` and `uxr_read`) with your own CAN logic. [Here](https://github.com/micro-ROS/freertos_apps/blob/foxy/microros_crazyflie21_extensions/src/crazyflie_transport.c) you have an example of a custom transport used in the Crazyflie Drone radio. The external transport can be configured via CMake arguments, so we can talk here about this if you have doubts.

vibnwis, 2020-10-06:
Sure!
The IDE is STM32CubeIDE.
It is not compulsory. The reasons are I am familiar with STM32CubeIDE, and
the STM32 community is large, and their support, especially from ST are
satisfactory so far. I could get a lot of information for their forum. I
have a limited time frame to complete my thesis. Hence, I was thinking to
spend as little time as possible on the development environment, allowing
me to spend more time on the development. However, I will use another IDE
if I need. For that reason, I have purchased 3 Olimex STM32E407, but I also
investigated the corresponding IDE. However, seeing the forum about the
IDE, it has been no activity for quite some time. I am a bit confused, in
the website, where STM32CubeMX ( is it not the same as STM32CubeIDE? ), if
it is the same, it should be able to compile it even after "clean" up the
project. I will give a try for a few hours. Did you (your IDE) run cmake
and then invoking the IDE to complete the compilation? I will investigate
how STM32CubeIDE does it.

The decision at the moment of time is using Olimex STM32, get the suggested
IDE, and work on the adding CAN onto the Micro-XRCE-DDS. As such, can I ask
what IDE do you use? Eclipse with STM32 plugin? Do you mind sharing?

Thank you.

V

vibnwis, 2020-10-06:
Okie, found out STM32CuteMX is not the same STM32CubeIDE. I have downloaded that and to be installed tomorrow. However, I can't do anything as I board is using ST-LInk/V3.

vibnwis, 2020-10-06:
I think I was mistaken the code generation as compilation. The compilation does not work in STM32Cube IDE. I will try on STM32CuteMX tomorrow.

vibnwis, 2020-10-07:
Hi pablogs9, I managed to install the STM32CubeMX and generated the code. In the end of that, it jumped to STM32CubeIDE. The same list of errors took place. I am still curious, what IDE you use, do you mind sharing, please? STM32CubeMX is only a code generator. I am thinking I could recreate a STM32CubeIDE project manually for micro-ROS based on STM32-E407 equivalent board. I would then recreate or import the files from  micro-ROS, one by one. I might missed out some of the files, would you lent a helping hand please? I see that micro-ROS uses a few middleware, FreeRTOS, and LwIP, may I ask where are those Micro XRCE-DDS reside, and other files allowing me to compose a project properly. I am open for other suggestion ? If I can setup other IDE, so long it is not commercial version, or too expensive, I think I would it.

Please advise.

Thank you
V!

closed 2020-10-07 by vibnwis

pablogs9, 2020-10-07:
Hello @vibnwis, let me explain a bit about the micro-ROS build system. micro-ROS build system is mainly based on CMake and it usually takes two steps (which are already integrated in the four step procedure described [here](https://micro-ros.github.io/docs/tutorials/core/first_application_linux/)):
 - **Cross-compiling libmicroros for a certain hardware platform**: this is done using the ROS 2 meta-building tool `colcon` which analyze the ROS 2 packages interdependencies and compiles (or crosscompile) them in a certain order in order to obtain a set of libraries that finally are merged in a single `libmicroros.a`. This step need headers of the RTOS and some POSIX compliance given by FreeRTOS+POSIX plugging.
 - **Building your application**: the application is linked against the RTOS and the `libmicroros.a`.

So, in general, we use the tool known as `micro_ros_setup`. This tool automatically installs all the dependencies and is able to cross-compile micro-ROS apps for a certain set of platforms. To edit the app code you can use your favorite editor (I use Visual Studio Code).

When talking about micro-ROS + FreeRTOS, this `micro_ros_setup` uses a CubeMX generated project in order to ease the RTOS+hardware integration. So, we have generated a CubeMX project, we have slightly modified the Makefile generated and everything is integrated into the  `micro_ros_setup` tool. 

As we know that users may want to modify the hardware configuration using CubeMX, we provide the CubeMX project in order to let the user modify the project and regenerate the code (always keeping in mind the micro-ROS changes). Once you have modified the configuration the `micro_ros_setup` will build the project.

Have you followed the [getting started tutorial with FreeRTOS](https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/freertos/)?

vibnwis, 2020-10-07:
Very well explained. I am actually reading the Makefile and got stuck in
the lines (libmicroros: colcon_compile) of generating libmicroros.a, trying
to figure out why the make failed to create libmicroros.a.