micro-ROS/micro_ros_setup #203: Networked Transport with Zephyr Guide

CLansdaleBEC, 2020-10-05:
Hello, I was wondering if there is any guide for connecting a board to microROS over UDP. My hardware setup is a STM32F7 board connected to usb power and connected to a PC running Ubuntu with Ethernet. 

A couple initial questions:

1. How do I configure the right port for the agent?
2. I have my PC assign an IP address to the board, is that the agent IP I should use in the configure and micro_ros_agent steps?

Thank you

pablogs9, 2020-10-05:
Hello @CLansdaleBEC. You have to configure a couple of things: the client and the agent. 

First, the agent will listen for incoming connections from clients using a transport (UDP in this case) at a certain port (any available port in your system, I usually use UDP/8888). It is common to set the verbosity level to a high value in order to check if the system is working, so in the agent part you should run something like:

```
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
```

This machine running the agent will have an interface with an IP address, let's say for example `192.168.1.100`. So, you have to use this agent IP and the previously selected port in the configuration step of the micro-ROS client crosscompilation. This way the micro-ROS client will know where it can find an agent:

```
ros2 run micro_ros_setup configure_firmware.sh [your app name] -t udp -i  192.168.1.100 -p 8888
```

This way you will configure your micro-ROS client at build time. You have a couple of more advanced options to configure where the micro-ROS client is looking for the micro-ROS agent: runtime configuration and automated agent discovery (we can talk about them later).

Answering your question 2): no, you have to configure the agent IP in the client's configure step.

Let me know if this is useful for you.

CLansdaleBEC, 2020-10-05:
This is definitely helpful.I think I have it configured right but I'm having some difficulty getting it to build. 

The int32_publisher app doesn't have a specific transport config nor a prj.conf file out of the box, so I tried to make my own (see below) based on the int32_wifi_pubisher. 

This fails at the build step on rmw_microxrcedds with this error

```
firmware/mcu_ws/build/rmw_microxrcedds/include/rmw_microxrcedds_c/config.h:15:41: error: 

UXR_CONFIG_UDP_TRANSPORT_MTU' undeclared here (not in a function);
 did you mean 'UXR_CONFIG_SERIAL_TRANSPORT_MTU'?
   15 |     #define RMW_UXRCE_MAX_TRANSPORT_MTU UXR_CONFIG_UDP_TRANSPORT_MTU
```
Do you know where this is defined? It seems like I'm missing something in prj.conf, but I couldn't it in the Zephyr/eProsima docs, any ideas?

proj.conf:
```
CONFIG_MAIN_STACK_SIZE=25000
CONFIG_MAIN_THREAD_PRIORITY=3

CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_NANO=n
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_PTHREAD_IPC=n

CONFIG_POSIX_API=y
CONFIG_NATIVE_POSIX_TIMER=y
CONFIG_APP_LINK_WITH_POSIX_SUBSYS=y
CONFIG_POSIX_CLOCK=y

CONFIG_HEAP_MEM_POOL_SIZE=8192
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
CONFIG_NET_IPV4=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_DHCPV4=y
CONFIG_NETWORKING=y
```

pablogs9, 2020-10-06:
The `proj.conf` is the Zephyr configuration so it does not configure any property of micro-ROS. Some of its options are required in order to build micro-ROS correctly (i.e `CONFIG_NEWLIB_LIBC`), so here you should keep the configuration that just enables the networking in your board.

Regarding your error, your build is still configured using serial transport. Two points:
- Can you share `firmware/mcu_ws/colcon.meta` and your app `app-colcon.meta`?
- When you change the transport layer a clean build is recommended, have you ensured that you have no file from a previous configuration with a serial transport?

CLansdaleBEC, 2020-10-06:
Looks like part of the issue was in app-colcon.meta, it sets UDP to off by default:

```
        "microxrcedds_client":{
            "cmake-args":[
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF"
            ]
        },
```

Changing DUCLIENT_PROFILE_UDP to ON gets me past building rmw_microxrcedds, but in the linking step these issues come up, among other undefined refrences in libmicroxrcedds_client-udp_transport_posix.c.obj
```
int32_publisher/../../../mcu_ws/install/libmicroros.a(libmicroxrcedds_client-udp_transport_posix.c.obj): in function `getaddrinfo':
firmware/zephyr_apps/microros_extensions/../../zephyrproject/zephyr/include/posix/netdb.h:21: undefined reference to `zsock_getaddrinfo'

zephyr_apps/apps/int32_publisher/../../../mcu_ws/install/libmicroros.a(libmicroxrcedds_client-udp_transport_posix.c.obj): in function `zsock_recvfrom':
eth_test_ws/firmware/build/zephyr/include/generated/syscalls/socket.h:176: undefined reference to `z_impl_zsock_recvfrom'

```

I'm pretty sure that workspace was new but I'm building again to be sure.

app-colcon.meta:
```
{
    "names": {
        "microxrcedds_client":{
            "cmake-args":[
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_TCP=OFF"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=1",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=0",
                "-DRMW_UXRCE_MAX_SERVICES=0",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=1"
            ]
        }
    }
}

```

colcon.meta:
```
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rosidl_typesupport": {
            "cmake-args": [
                "-DROSIDL_TYPESUPPORT_SINGLE_TYPESUPPORT=ON"
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
                "-DRCL_LOGGING_ENABLED=OFF"
            ]
        },
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_EXTERNAL_SERIAL=OFF"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=1",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=1",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=192.168.1.150",
                "-DRMW_UXRCE_DEFAULT_UDP_PORT=8888"
            ]
        }
    }
}

```

pablogs9, 2020-10-07:
Regarding the first point, the application meta should not have middleware specific configuration. These configurations should be in `mcu_ws/colcon.meta` which is the one modified by the `configure_firmware.sh` step. Both of them are used during the build process. To fix this I have prepared https://github.com/micro-ROS/zephyr_apps/pull/23.

---

Regarding the communication between client and agent, I would like to be sure if the client is sending the UDP packages. Could you check using Netcat (`nc -l 8888`) or Wireshark if the board is actually sending anything to the agent?

If you are able to debug using a JTAG or something similar, it would be interesting to set some breakpoints [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/356a1e6d6181134f7d45c9ce7fe5bdff27d7c1ec/src/c/profile/transport/ip/udp/udp_transport_posix.c#L76) to check if the middleware transport is sending packets to the agent.

Please tell me if you can see output packets being sent from the client to the network or if you can trace if the executing flow reaches the transport layer. 

---

Another thought that I just have, I'm not sure about this being related with your problem but the fact is that we haven't any hardware here that supports Zephyr and have Ethernet: 
  - We are supposing that Zephyr has a complete POSIX API for networking ([here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/356a1e6d6181134f7d45c9ce7fe5bdff27d7c1ec/CMakeLists.txt#L129-L130)).
  - This fact makes the middleware to use the default POSIX UDP transport which is [this](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/foxy/src/c/profile/transport/ip/udp/udp_transport_posix.c).
 - This default POSIX UDP transport [uses polling](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/356a1e6d6181134f7d45c9ce7fe5bdff27d7c1ec/src/c/profile/transport/ip/udp/udp_transport_posix.c#L97).
 - In other platforms based on FreeRTOS that uses the LwIP network stack we have found that polling is not supported, so we have created [another POSIX transport without polling](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/foxy/src/c/profile/transport/ip/udp/udp_transport_posix_nopoll.c)
 - Does Zephyr support this kind of polling? Is [this config](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_POLL.html#cmdoption-arg-config-poll) related?

CLansdaleBEC, 2020-10-09:
Wireshark shows repeating ARP messages on the interface but still no microROS messages

[image]

And thank you for the resources I'll take a deeper dive and get back to you

pablogs9, 2020-10-13:
Is the ARP being resolved? If not, maybe there is a networking problem.

Before trying micro-ROS, have you tried a simple UDP communication between the board and the PC in order to verify the behavior of all the UDP/IP stack?

pablogs9, 2020-12-15:
Related: https://github.com/micro-ROS/zephyr_apps/issues/31

I'm closing due to inactivity. Please do not hesitate to reopen if you need further discussion.

closed 2020-12-15 by pablogs9