micro-ROS/micro_ros_setup #204: Unable to locate ros-foxy-google-benchmark-vendor

CLansdaleBEC, 2020-10-06:
Hello, this issue just popped up when trying to run create_firmware_ws.sh. It doesn't look like it's board/rtos specific. 

I'm using Ubuntu 20.04 with Foxy, and ran an apt update before testing

[image]

vibnwis, 2020-10-07:
Having see your message, I took a close look of mine, me too experienced the same error.

vibnwis, 2020-10-07:
out of curiosity, what is that for? the ros-foxy-google-benchmark-vendor

pablogs9, 2020-10-07:
They have updated `ament_cmake` adding this new dependency but it seems not to be resolving well `ros-foxy-google-benchmark-vendor`. Check this: https://github.com/ament/ament_cmake/commits/foxy

I have opened an issue in this package: https://github.com/ament/ament_cmake/issues/285

I'm working on a fix here: https://github.com/micro-ROS/micro_ros_setup/pull/205, I will alert you when merged (this morning).

pablogs9, 2020-10-07:
Hello @CLansdaleBEC and @vibnwis, https://github.com/micro-ROS/micro_ros_setup/pull/205 merged, this problem should be fixed.

Thanks a lot for reporting!

closed 2020-10-07 by pablogs9