micro-ROS/micro_ros_setup #206: Integrating CAN transport to Micro XRCE-DDS

vibnwis, 2020-10-07:
Hi Pablo and teams,
Many thanks for the IDE stuff. 
I am now moving onto integrating or adding a CAN transport onto Micro XRCE_DDS. I have also spent sometime reading the code you suggested. I did not see the code uses any of those APIs, i.e. uxr_open, uxr_close, uxr_read and uxr_write.  Only the function naming has some wording of those API wordings. That confused me, how the integration work. Where are those new functions get called? I believe at some other files need to be modified to make the new functions visible. Would you mind elaborate, please?

Thank you
V!

pablogs9, 2020-10-07:
Hello @vibnwis, let me explain this using just XRCE-DDS examples:

- When you [start a transport](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/66df0a6df20063d246dd638ac3d33eb2e652fab2/examples/PublishHelloWorld/main.c#L43) the `transport->comm` [is being filled](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/46ee5bcd3fa1a07cdbe41ecb2b9a9d8c4b2a874d/src/c/profile/transport/ip/udp/udp_transport.c#L83) with this write/read functions.
- These functions directly uses [uxr_write_udp_data_platform()](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/46ee5bcd3fa1a07cdbe41ecb2b9a9d8c4b2a874d/src/c/profile/transport/ip/udp/udp_transport.c#L24) and [uxr_read_udp_data_platform()](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/46ee5bcd3fa1a07cdbe41ecb2b9a9d8c4b2a874d/src/c/profile/transport/ip/udp/udp_transport.c#L42)
- When you init an XRCE session you pass the `transport->comm` structure ([check here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/66df0a6df20063d246dd638ac3d33eb2e652fab2/examples/PublishHelloWorld/main.c#L51)) so the library knows how to send and receive data

Well, so you need to implement a transport using these kinds of [templates](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/foxy/src/c/profile/transport/ip/udp/udp_transport_external.c.template). But first you need to decide if your transport is UDP-like or serial-like. The difference is:
- In **UDP-like** transport your custom transport can send and receive complete messages and the start and the end of the data is delimited by the whole message. Let's say that your transport is message oriented.
- In other transport, **serial-like** ones, you only send a stream. In these cases, your transport can send or receive an arbitrary number of bytes depending on the internal buffers, the data available in the lower layers, etc. In this case, XRCE can't determine the start and the end of the messages, so the XRCE library implements a framing algorithm that delimits the start and the end of the message and adds a redundancy check CRC code.

Which one is the most similar to CAN BUS?

Remember that you also have to implement this transport in the Micro XRCE-DDS Agent or you have to create an external "brigde" that convert from CAN to UDP or Serial stream. Maybe @jamoralp can help you here.

vibnwis, 2020-10-08:
Hi Pablo,

Very good stuff. I will spend time reading through those stuff, and will
come back to you with more questions if you don't mind. Also, I will also
be sending @jamoralp when necessary.

CAN-bus is a frame-based, connectionless communication protocol very
similar to UDP but without IP. Hence, it is similar to UDP.

pablogs9, 2020-10-08:
Nice to know this about CAN-bus, I think that creating a UDP like transport is easier. Let me know when you need more help.

@jamoralp is now on holiday, and he'll come back to the office in a couple of weeks.

vibnwis, 2020-10-08:
Many thanks for the thorough explanation.

Having read your explanation and the code. Here is my understanding,
however, please feel free to comment. The interface to the existing
Micro_XRCE-DDS is uxrUDPTransport* transport. By filling up the interface,
i.e. the 'transport' variable, the existing Micro_XRCE-DDS will pick up not
only the uxr_read() and uxr_write() and other  information, and execute
accordingly. The template as it implies, is a template, a sample code,
which the function-names can also be "localised" if needed, so long the
calling program, for instance, the PublishHelloWorld/main.c, has declared
the  uxrUDPTransport* transport, initialised accordingly. It is needed so
the following uxr_init_session can initialise accordingly.

Feel free to comment.

Thank you
V!

pablogs9, 2020-10-08:
Yes. Try to fill this UDP external transport template and build the library using this transport instead of the UDP_POSIX. Check the CMakeLists.txt to see the variables involved.

vibnwis, 2020-10-08:
Will do. Many thanks.

vibnwis, 2020-10-12:
Hi Pablo,

I have "sort-of" completed the coding, as I find the coding method is kind
of "painful". I am using VS-Code, and made a project and copied all folders
under the micro-ros firmware. I was trying to set up in such a way that I
can browse the definition of the data-type, and jump back and forth. Sorry,
this is my first time using VS-code. How did you set your workspace from
compiling to flashing? If I may ask.

Many thanks in advance.

Cheers!
V!

pablogs9, 2020-10-13:
Well, you don't need any special setup of the workspace, just open a new folder, open a terminal, and follow the [instructions of the tutorial](https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/). It will create all the folders and download required files. Which data-type are you browsing?

vibnwis, 2020-10-13:
I am in the course of implementing it. very well-organised information, and well explained.

vibnwis, 2020-10-14:
Hi Pablo,
I think I am stuck with the compilation. I had run the command, ros2 run
micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407 of which
created the firmware folder. For my case,

I make changes on some files in the folder
firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/PublishHellowWorld/main.c
, and added CAN_circular_buffer.c, CAN_circular_buffer.h.

Also, a file I changed in the folder
Micro-XRCE-DDS-Client/src/c/profile/transport/ip/udp/udp_transport.c

I have not changed those variables in the CMakeList.txt relating to the
static memory allocation yet. I will leave at a later stage when I have
passed the compilation and debugging stages.

I followed almost to the completion of the step in
https://micro-ros.github.io/docs/tutorials/core/create_new_type/, but I
think I have got into a wrong pathway, as I am not creating a new message.
I am creating a transport, but still using existing uxr_xxx_read()/write()
functions, and using the main.c in the folder
firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/PublishHellowWorld.

Appreciate your advice, please.

pablogs9, 2020-10-15:
Hello @vibnwis, some points:

- `firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/PublishHellowWorld/main.c` is not being built with the micro-ROS build system since it is a Micro XRCE-DDS client example.
- Create a new type tutorial is not intended for creating new transports.

I think that it would be easier to write and test your transport just using the middleware library, once you have it working using it in micro-ROS will be straightforward.

I kindly suggest you trying the demos of the XRCE library and start working with it since, in fact, you will write a transport for XRCE. With these commands, you can build some standalone demos of XRCE, remember that this is detached from micro-ROS but the transports will be the same:

```bash
git clone -b v1.5.0 https://github.com/eProsima/Micro-XRCE-DDS-Agent
cd Micro-XRCE-DDS-Agent/
mkdir build; cd build;
cmake ..
make

git clone -b v1.2.5 https://github.com/eProsima/Micro-XRCE-DDS-Client
cd Micro-XRCE-DDS-Client/
mkdir build; cd build;
cmake .. -DUCLIENT_BUILD_EXAMPLES=ON
make
```

With this commands you have a working executable of the agent places in Micro-XRCE-DDS-Agent/build/MicroXRCEAgent and several examples under Micro-XRCE-DDS-Client/examples.

vibnwis, 2020-10-15:
> git clone -b v1.5.0 https://github.com/eProsima/Micro-XRCE-DDS-Agentcd Micro-XRCE-DDS-Agent/
> git clone -b v1.2.5 https://github.com/eProsima/Micro-XRCE-DDS-Clientcd Micro-XRCE-DDS-Client/

Many thanks for your email. I have figured that out from one the Issue in
https://github.com/eProsima/Micro-XRCE-DDS-Client. I have got it compiled
with the changes I made. However, it does not recognised the new files, I
presume the Makefile needs to be edited to incorporate that.

BTW, since the eProsima/Micro-XRCE-DDS-Client
<https://github.com/eProsima/Micro-XRCE-DDS-Client> is also exist in
firmware/mcu_sw/eProsima/Micro-XRCE-DDS-Client
<https://github.com/eProsima/Micro-XRCE-DDS-Client>, I would not need to
git that separately or I need for some reasons?

Of course, final question, how do I run or specify what to run those
examples ?

pablogs9, 2020-10-15:
Let's move to https://github.com/eProsima/Micro-XRCE-DDS-Client/issues/168