micro-ROS/micro_ros_setup #207: MicroROS Agent for new custom message

MARLARCHAN, 2020-10-08:
Dear @pablogs9 

      I'm trying to test with new custom message. I attach the app.c file in here. I can debug and compile firmware. According to your link, I use std_msg. But when I test agent running, nsh is aborting. Please guide me how can I solve.
[image]

[app.txt](https://github.com/micro-ROS/micro_ros_setup/files/5346583/app.txt)

pablogs9, 2020-10-08:
You are creating two nodes using the same `rcl_node_t` struct.

Check lines 53 and 54:

```c
RCCHECK(rclc_node_init_default(&node, "int32_publisher", "", &support));
RCCHECK(rclc_node_init_default(&node, "int32_subscriber", "", &support));
```

I recommend you using the same node for creating the subscriber and the publisher, just:

```c
RCCHECK(rclc_node_init_default(&node, "my_microros_node", "", &support));
```

pablogs9, 2020-10-13:
Close if solved, please.

MARLARCHAN, 2020-10-13:
Thank you. I can solve right now.

closed 2020-10-13 by MARLARCHAN